cartographer-project/cartographer_ros #978: message_generationConfig.cmake no found

Cass-zhu, 2018-08-11:
OS: Ubuntu 16.04
ROS kinetic

catkin_make_isolated --install --use-ninja \
  -DCMAKE_PREFIX_PATH="${PWD}/install_isolated;${PWD}/protobuf/install;${CMAKE_PREFIX_PATH}"
Base path: /home/castiel/cartographer_ws
Source space: /home/castiel/cartographer_ws/src
Build space: /home/castiel/cartographer_ws/build_isolated
Devel space: /home/castiel/cartographer_ws/devel_isolated
Install space: /home/castiel/cartographer_ws/install_isolated
Additional CMake Arguments: -DCMAKE_PREFIX_PATH=/home/castiel/cartographer_ws/install_isolated;/home/castiel/cartographer_ws/protobuf/install;/home/castiel/catkin_ws/devel:/opt/ros/kinetic:/home/castiel/cartographer_ws/protobuf/install
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 14 packages in topological order:
~~  - cartographer_ros_msgs
~~  - ceres-solver (plain cmake)
~~  - cartographer (plain cmake)
~~  - cartographer_ros
~~  - cartographer_rviz
~~  - jackal_control
~~  - jackal_description
~~  - jackal_desktop
~~  - jackal_gazebo
~~  - jackal_msgs
~~  - jackal_navigation
~~  - jackal_simulator
~~  - jackal_tutorials
~~  - jackal_viz
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
The packages or cmake arguments have changed, forcing cmake invocation

==> Processing catkin package: 'cartographer_ros_msgs'
==> Creating build directory: 'build_isolated/cartographer_ros_msgs'
==> cmake /home/castiel/cartographer_ws/src/cartographer_ros/cartographer_ros_msgs -DCATKIN_DEVEL_PREFIX=/home/castiel/cartographer_ws/devel_isolated/cartographer_ros_msgs -DCMAKE_INSTALL_PREFIX=/home/castiel/cartographer_ws/install_isolated -DCMAKE_PREFIX_PATH=/home/castiel/cartographer_ws/install_isolated;/home/castiel/cartographer_ws/protobuf/install;/home/castiel/catkin_ws/devel:/opt/ros/kinetic:/home/castiel/cartographer_ws/protobuf/install -G Ninja in '/home/castiel/cartographer_ws/build_isolated/cartographer_ros_msgs'
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/castiel/cartographer_ws/devel_isolated/cartographer_ros_msgs
-- Using CMAKE_PREFIX_PATH: /home/castiel/cartographer_ws/install_isolated;/home/castiel/cartographer_ws/protobuf/install;/home/castiel/catkin_ws/devel:/opt/ros/kinetic:/home/castiel/cartographer_ws/protobuf/install
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/castiel/cartographer_ws/build_isolated/cartographer_ros_msgs/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.11
CMake Warning at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:76 (find_package):
  Could not find a package configuration file provided by
  "message_generation" with any of the following names:

    message_generationConfig.cmake
    message_generation-config.cmake

  Add the installation prefix of "message_generation" to CMAKE_PREFIX_PATH or
  set "message_generation_DIR" to a directory containing one of the above
  files.  If "message_generation" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:23 (find_package)

-- Could not find the required component 'message_generation'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "message_generation" with any of the following names:

    message_generationConfig.cmake
    message_generation-config.cmake

  Add the installation prefix of "message_generation" to CMAKE_PREFIX_PATH or
  set "message_generation_DIR" to a directory containing one of the above
  files.  If "message_generation" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:23 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/castiel/cartographer_ws/build_isolated/cartographer_ros_msgs/CMakeFiles/CMakeOutput.log".
See also "/home/castiel/cartographer_ws/build_isolated/cartographer_ros_msgs/CMakeFiles/CMakeError.log".
<== Failed to process package 'cartographer_ros_msgs': 
  Command '['cmake', '/home/castiel/cartographer_ws/src/cartographer_ros/cartographer_ros_msgs', '-DCATKIN_DEVEL_PREFIX=/home/castiel/cartographer_ws/devel_isolated/cartographer_ros_msgs', '-DCMAKE_INSTALL_PREFIX=/home/castiel/cartographer_ws/install_isolated', '-DCMAKE_PREFIX_PATH=/home/castiel/cartographer_ws/install_isolated;/home/castiel/cartographer_ws/protobuf/install;/home/castiel/catkin_ws/devel:/opt/ros/kinetic:/home/castiel/cartographer_ws/protobuf/install', '-G', 'Ninja']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/castiel/cartographer_ws/build_isolated/cartographer_ros_msgs && cmake /home/castiel/cartographer_ws/src/cartographer_ros/cartographer_ros_msgs -DCATKIN_DEVEL_PREFIX=/home/castiel/cartographer_ws/devel_isolated/cartographer_ros_msgs -DCMAKE_INSTALL_PREFIX=/home/castiel/cartographer_ws/install_isolated '-DCMAKE_PREFIX_PATH=/home/castiel/cartographer_ws/install_isolated;/home/castiel/cartographer_ws/protobuf/install;/home/castiel/catkin_ws/devel:/opt/ros/kinetic:/home/castiel/cartographer_ws/protobuf/install' -G Ninja

I tried rospack find message_generation I can find it.
But still this error happened.
Can anyone tell me how to deal with it? THanks!

kdaun, 2018-08-20:
The the message_generation package is missing in your system. You can install it via apt-get:
`sudo apt-get install ros-kinetic-message-generation `

rubenanapu, 2019-05-30:
I have the same problem, but the package aforementioned is installed.
The command `sudo dpkg -l | grep generation` outputs:

```
ros-kinetic-message-generation                     0.4.0-0xenial-20180824-081816-0800  
```
and `rospack find message_generation`  outputs:

```
rospack find message_generation
/opt/ros/kinetic/share/message_generation
```

Sunshinehualong, 2019-08-09:
> I have the same problem, but the package aforementioned is installed.
> The command `sudo dpkg -l | grep generation` outputs:
>
> ```
> ros-kinetic-message-generation [...]

Hi, rubenanapu, have you solved this problem,
I really installed the package, but still have hte error

rubenanapu, 2019-08-09:
Hi @Sunshinehualong, I may have solved that problem but I'm afraid to say I don't remember how.

I even looked at my logs `git log | grep carto` but couldn't find any references to the solution.

I used `grep carto . -rn` on my repo as well to see any notes related to this but I couldn't find.

Next time I'll try to pay more attention to updating the issues.

whintch, 2019-10-06:
I have the some question and finally I solve it by checking other package code
Sorry I didn't figure out the concrete reason but I can share with you my CmakeLists.txt code and package.xml code. You can modify them and then use in your own package.
BTW,if any one figure out the reason please write an answer here. Thanks in advance.

CmakeLists.txt:
cmake_minimum_required(VERSION 2.8.3)
project(testmsg3)

find_package(catkin REQUIRED COMPONENTS
  message_generation
  roscpp
  rospy
  std_msgs
)

## Generate messages in the 'msg' folder
add_message_files(
   FILES
 # msg1.msg
)

## Generate added messages and services with any dependencies listed here
generate_messages(
   DEPENDENCIES
   std_msgs
)

###################################
## catkin specific configuration ##
###################################
catkin_package(
#  INCLUDE_DIRS include
#  LIBRARIES ur_msgs
   CATKIN_DEPENDS message_runtime std_msgs
#  DEPENDS system_lib
)

## DEPENDS: system dependencies of this project that dependent projects also need
catkin_package(
#  INCLUDE_DIRS include
#  LIBRARIES testmsg3
#  CATKIN_DEPENDS message_generation roscpp rospy std_msgs
#  DEPENDS system_lib
)

###########
## Build ##
###########

## Specify additional locations of header files
## Your package locations should be listed before other locations
include_directories(
# include
  ${catkin_INCLUDE_DIRS}
)

package.xml:
<?xml version="1.0"?>
<package format="2">
  <name>testmsg3</name>
  <version>0.0.0</version>
  <description>The testmsg3 package</description>
  <maintainer email="linux80@todo.todo">linux80</maintainer>

  
  
  
  <license>TODO</license>

  <exec_depend>message_runtime</exec_depend>

  <buildtool_depend>catkin</buildtool_depend>
  <build_depend>message_generation</build_depend>
  <build_depend>roscpp</build_depend>
  <build_depend>rospy</build_depend>
  <build_depend>std_msgs</build_depend>
  <build_export_depend>roscpp</build_export_depend>
  <build_export_depend>rospy</build_export_depend>
  <build_export_depend>std_msgs</build_export_depend>
  <exec_depend>roscpp</exec_depend>
  <exec_depend>rospy</exec_depend>
  <exec_depend>std_msgs</exec_depend>

  
  <export>
    

  </export>
</package>

andoreso, 2019-12-01:
I had the same problem, I solved it by sourcing ROS. So i added `source /opt/ros/melodic/setup.bash` to the top of my bash script.

closed 2020-10-21 by MichaelGrupp