cartographer-project/cartographer_ros #979: How to save the trajectory and the submap corresponding to each pose？

woshidaye, 2018-08-12:
I want to save each submap and its corresponding pose. How to achieve it? I look forward to your answer.

cschuet, 2018-08-13:
If you are using the offline_node for mapping it should write out a .pbstream which contains submap poses and submaps. The file format is not documented but you can read the cartographer serialization code to piece it together. It is essentially a stream of protocol buffers, one written after the other.

Alternatively our ROS nodes publish the "submap_list" topic which consists of [these](https://github.com/googlecartographer/cartographer_ros/blob/168f4cc53fc792ad8db2f7bc5b8dd42f679e734b/cartographer_ros_msgs/msg/SubmapList.msg) SubmapList ROS messages. They contain a list of submap IDs (consisting of trajectory ID and submap index) with the global pose of the submap. To get the probability grids you could call into our ROS Submap service with the ID of the submap you want to retrieve the grid for.

woshidaye, 2018-08-30:
@cschuet .First of all, thank you for your answer. I saved the.pdstream file according to your answer. How can I open the file? Can the content of this .pdstream file continue to be used for graph optimization? Looking forward to your reply again.

closed 2020-10-21 by MichaelGrupp