cartographer-project/cartographer_ros #980: catkin_make failed!

cgdsss, 2018-08-12:
When I use the catkin_make to compile cartographer_ros,  the errors is reported as below:
`/home/cgd/catkin_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.cc:180:32: error: no match for ‘operator[]’ (operand types are ‘const value_type {aka const cartographer::sensor::TimedRangefinderPoint}’ and ‘int’)
   CHECK_LE(points.points.back()[3], 0);
                                ^
/home/cgd/catkin_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.cc:192:62: error: no match for ‘operator[]’ (operand types are ‘__gnu_cxx::__alloc_traits<std::allocator<cartographer::sensor::TimedRangefinderPoint> >::value_type {aka cartographer::sensor::TimedRangefinderPoint}’ and ‘int’)
     const double time_to_subdivision_end = subdivision.back()[3];
                                                              ^
/home/cgd/catkin_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.cc:207:35: error: invalid initialization of reference of type ‘Eigen::Vector4f& {aka Eigen::Matrix<float, 4, 1>&}’ from expression of type ‘cartographer::sensor::TimedRangefinderPoint’
     for (Eigen::Vector4f& point : subdivision) {
                                   ^
In file included from /usr/local/include/cartographer/common/lua_parameter_dictionary.h:27:0,
                 from /usr/local/include/cartographer/mapping/trajectory_builder_interface.h:25,
                 from /home/cgd/catkin_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.h:23,
                 from /home/cgd/catkin_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.cc:17:
/home/cgd/catkin_ws/src/cartographer_ros/cartographer_ros/cartographer_ros/sensor_bridge.cc:210:32: error: no match for ‘operator[]’ (operand types are ‘__gnu_cxx::__alloc_traits<std::allocator<cartographer::sensor::TimedRangefinderPoint> >::value_type {aka cartographer::sensor::TimedRangefinderPoint}’ and ‘int’)
     CHECK_EQ(subdivision.back()[3], 0);
`
**Below is my entire compilation and installation steps.**
`
sudo apt-get update
sudo apt-get install -y \
    g++ \
    git \
    google-mock \
    libboost-all-dev \
    libcairo2-dev \
    libeigen3-dev \
    libgflags-dev \
    libgoogle-glog-dev \
    liblua5.2-dev \
    libsuitesparse-dev \
    ninja-build \
    python-sphinx

git clone https://github.com/ceres-solver/ceres-solver.git
cd ceres-solver
git checkout tags/1.13.0
mkdir build
cmake .. -G Ninja -DCXX11=ON
ninja
CTEST_OUTPUT_ON_FAILURE=1 ninja test
sudo ninja install

git clone https://github.com/google/protobuf.git
cd protobuf
git checkout tags/v3.4.1
mkdir build
cd build
cmake -G Ninja \
  -DCMAKE_POSITION_INDEPENDENT_CODE=ON \
  -DCMAKE_BUILD_TYPE=Release \
  -Dprotobuf_BUILD_TESTS=OFF \
  ../cmake
ninja
sudo ninja install

git clone https://github.com/googlecartographer/cartographer.git
cd cartographer
mkdir build
cd build
cmake .. -G Ninja
ninja
CTEST_OUTPUT_ON_FAILURE=1 ninja test
sudo ninja install

cd ~/catkin_ws/src
git clone https://github.com/googlecartographer/cartographer_ros.git
cd ~/catkin_ws
catkin_make --only-pkg-with-deps cartographer_ros
`

cgdsss, 2018-08-12:
Can someone help me? Thank you very much.

ZhouYixuanRobtic, 2018-08-12:
There's a official install [process](https://google-cartographer-ros.readthedocs.io/en/latest/#building-installation) recommended. I've installed cartographer right under its instruction with UBUNTU 16.04 + ROS kinetic.Hope it's' helpful

cgdsss, 2018-08-12:
@ZhouYixuanRobtic thank you！ I will try. But I still want to know where I was wrong.

cgdsss, 2018-08-12:
@ZhouYixuanRobtic 
https://raw.githubusercontent.com/googlecartographer/cartographer_ros/master/cartographer_ros.rosinstall
```
- git: {local-name: cartographer, uri: 'https://github.com/googlecartographer/cartographer.git', version: '1.0.0'}
- git: {local-name: cartographer_ros, uri: 'https://github.com/googlecartographer/cartographer_ros.git', version: '1.0.0'}
- git: {local-name: ceres-solver, uri: 'https://ceres-solver.googlesource.com/ceres-solver.git', version: '1.13.0'}
```
Using the specified version of the code, smoothly compile and pass.

closed 2018-08-12 by cgdsss

ojura, 2018-08-12:
#964 is not merged yet in cartographer_ros, while the corresponding PR in cartographer was merged.