cartographer-project/cartographer_ros #985: how can I install imu into cartographer?

forlinchow, 2018-08-21:
I'm using rplidar to do 2d SLAM, and I change to tracking_frame to match the laser data.

now I buy a new 9dof imu and hope the imu data can help it correct the map.( because sometime we can't keep it perfectly horizontal.

forlinchow, 2018-08-21:
and I already setup the imu.  

I can see the imu data and the /imu topic in rostopic

kdaun, 2018-08-21:
What is your question?
What do you need help with?

If you want to use the IMU messages in Cartographer you need to:

1. Remap your imu message topic to `imu`
2. Enable using IMU in your Cartographer configurations file
3. Add the IMU link to your URDF

Check the [documentation](https://google-cartographer-ros.readthedocs.io/en/latest/ros_api.html) for more information and see the [backpack2d launch and configuration files](https://github.com/googlecartographer/cartographer_ros/tree/master/cartographer_ros) as an example for 2D SLAM with IMU.

forlinchow, 2018-08-21:
Does it need to initialize?

kdaun, 2018-08-22:
Cartographer supports calibrating the orientation of the IMU, however the general transform between LiDAR and IMU has to be provided by in the URDF. 
An example is given in setup for [backpack_2d](https://github.com/googlecartographer/cartographer_ros/tree/master/cartographer_ros).
See the [configuration documentation](https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html) for an general overview on setting up your TF/URDF.

forlinchow, 2019-01-15:
thanku

closed 2019-01-15 by forlinchow

duandongbin000, 2019-07-03:
which type of imu is used in cartographer official website bags ？