cartographer-project/cartographer_ros #986: IMU  question

litzar, 2018-08-22:
Dear all:

Is it that the IMU chip should be installed in the center of the robot, so that the translation().norm()<1e-5?
If so, why, how to understand it? 

Thanks.

void SensorBridge::HandleImuMessage(const string& topic,
                                    const sensor_msgs::Imu::ConstPtr& msg) {
  CHECK_NE(msg->linear_acceleration_covariance[0], -1);
  CHECK_NE(msg->angular_velocity_covariance[0], -1);
  const carto::common::Time time = FromRos(msg->header.stamp);
  const auto sensor_to_tracking = tf_bridge_->LookupToTracking(
      time, CheckNoLeadingSlash(msg->header.frame_id));
  if (sensor_to_tracking != nullptr) {
    **CHECK(sensor_to_tracking->translation().norm() < 1e-5)
        << "The IMU frame must be colocated with the tracking frame. "
           "Transforming linear acceleration into the tracking frame will "
           "otherwise be imprecise.";**
    trajectory_builder_->AddImuData(
        topic, time,
        sensor_to_tracking->rotation() * ToEigen(msg->linear_acceleration),
        sensor_to_tracking->rotation() * ToEigen(msg->angular_velocity));
  }
}

kdaun, 2018-08-22:
For Cartographer the IMU does not necessarily be installed in the center of the robot, but it has to be collocated with the tracking frame.
The reason is already given in your issue:
> "The IMU frame must be colocated with the tracking frame. Transforming linear acceleration into the tracking frame will otherwise be imprecise."

The tracking_frame is the ROS frame ID of the frame that is tracked by the SLAM algorithm. If an IMU is used, it should be at its position, although it might be rotated. A common choice is “imu_link”. See the [documentation](https://google-cartographer-ros.readthedocs.io/en/latest/configuration.html) for more information about the frames.

litzar, 2018-08-22:
Dear Kevin: is it mean that the tracking_frame should be the same as the IMU frame, at least the translation should be same?

litzar, 2018-08-22:
@kdaun Hi Kevin, 

I found the imu_link now. Thanks very much.

<link name="imu_link">
    <visual>
      <origin xyz="0 0 0" />
      <geometry>
        <box size="0.06 0.04 0.02" />
      </geometry>
      <material name="orange" />
    </visual>
  </link>

 <link name="base_link" />

  <joint name="imu_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="imu_link" />
    <origin xyz="0 0 0" />
  </joint>

  <joint name="horizontal_laser_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="horizontal_laser_link" />
    <origin xyz="0.1007 0 0.0558" />
  </joint>

closed 2018-08-22 by litzar

Duke-Allen, 2019-02-28:
@litzar @kdaun 
Hello, I am using a 16-lines LIDAR and IMU to build 3D map by cartographer.
I tun the parameters and run it with no error, the results are as follows:
[image]
[image]

There is a big gap with the real environment. I think the map is in a mess. What is the problem of this result? How can I solve it?
This is the validate result of my bag:

- E0227 22:30:57.092067 19056 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.6084 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
I0227 22:30:57.093359 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 37968  Min: 0.004133  Max: 0.016211  Mean: 0.010000
[0.004133, 0.005340)	                    	Count: 2 (0.005268%)	Total: 2 (0.005268%)
[0.005340, 0.006548)	                    	Count: 3 (0.007901%)	Total: 5 (0.013169%)
[0.006548, 0.007756)	                    	Count: 8 (0.021070%)	Total: 13 (0.034239%)
[0.007756, 0.008964)	                    	Count: 828 (2.180784%)	Total: 841 (2.215023%)
[0.008964, 0.010172)	 ###################	Count: 35958 (94.706070%)	Total: 36799 (96.921089%)
[0.010172, 0.011380)	                    	Count: 345 (0.908660%)	Total: 37144 (97.829750%)
[0.011380, 0.012588)	                    	Count: 820 (2.159714%)	Total: 37964 (99.989464%)
[0.012588, 0.013796)	                    	Count: 1 (0.002634%)	Total: 37965 (99.992096%)
[0.013796, 0.015004)	                    	Count: 0 (0.000000%)	Total: 37965 (99.992096%)
[0.015004, 0.016211]	                    	Count: 3 (0.007901%)	Total: 37968 (100.000000%)
E0227 22:30:57.094650 19056 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101409 s, recommended is [0.0005, 0.05] s with no jitter.
I0227 22:30:57.094745 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 3766  Min: 0.100213  Max: 0.101409  Mean: 0.100801
[0.100213, 0.100333)	                    	Count: 1 (0.026553%)	Total: 1 (0.026553%)
[0.100333, 0.100452)	                    	Count: 1 (0.026553%)	Total: 2 (0.053107%)
[0.100452, 0.100572)	                    	Count: 9 (0.238980%)	Total: 11 (0.292087%)
[0.100572, 0.100691)	                    	Count: 33 (0.876261%)	Total: 44 (1.168348%)
[0.100691, 0.100811)	      ##############	Count: 2717 (72.145515%)	Total: 2761 (73.313858%)
[0.100811, 0.100931)	               #####	Count: 972 (25.809877%)	Total: 3733 (99.123741%)
[0.100931, 0.101050)	                    	Count: 25 (0.663834%)	Total: 3758 (99.787575%)
[0.101050, 0.101170)	                    	Count: 6 (0.159320%)	Total: 3764 (99.946892%)
[0.101170, 0.101289)	                    	Count: 1 (0.026553%)	Total: 3765 (99.973450%)
[0.101289, 0.101409]	                    	Count: 1 (0.026553%)	Total: 3766 (100.000000%)

[image]

Here are my files:
[demo_rsldiar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
[my data bag](https://pan.baidu.com/s/1chlbMpfUJeY-JdXoPzxzPw)
Could you help me?