cartographer-project/cartographer_ros #987: Can we use two two-dimensional lasers as a three-dimensional  laser?

sunmk2006, 2018-08-22:
A two-dimensional laser placed horizontally and a two-dimensional laser placed vertically constitute a data acquisition device, which can obtain both horizontal and vertical data. If we convert the two data into the form of point clouds, can we use them as three-dimensional data?

kdaun, 2018-08-22:
Yes, Cartographer supports the use case of using two 2D Lidars as input for the 3D SLAM pipeline. We use a similar setup for our mapping backpack. See the [backpack_3d launch and configuration files](https://github.com/googlecartographer/cartographer_ros/tree/master/cartographer_ros).
Demo instructions are available in the [demo documentation](https://google-cartographer-ros.readthedocs.io/en/latest/demos.html).

sunmk2006, 2018-08-22:
Can you specify which configuration file uses two two-dimensional lasers as input for 3Dslam input?

kdaun, 2018-08-22:
See [backpack_3d.launch](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/launch/backpack_3d.launch) and [backpack_3d.lua](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/configuration_files/backpack_3d.lua).
In our case we are using two multi-line LiDARs. Depending on your system setup using two 2D LiDARs can work as well.

sunmk2006, 2018-08-22:
I have done this already, but the result is not good. Can you help me look at it? [the config file ](https://github.com/sunmk2006/cartographer_ros/blob/master/2D_lidar_3Dslam.tar.gz)and [the bag file](https://google-cartographer-ros.readthedocs.io/en/latest/data.html) is b0-2014-07-11-10-58-16.bag

kdaun, 2018-08-23:
Ok, with only two 2D LiDARs there is not enough overlap between the scans and the mapped area to accurately resolve the 6D pose. Therefore, the quality of the map is degraded.
Reducing the resolution might lead to somewhat better matches as there should be more overlap between the scans and the map.
To gain a high quality 3D map with full 6D pose estimation with Cartographer 3D LiDARs are needed.

closed 2020-10-23 by MichaelGrupp