cartographer-project/cartographer_ros #990: Strange Map Rotation on Global Loop Closure

Sanderi44, 2018-08-24:
1. Rosbag Validate: https://gist.github.com/Sanderi44/fd56510dc060941e2eca7d4ff9c24bd5
2. Configuration Files: https://github.com/Sanderi44/cartographer_ros_tuning
3. Bag File: https://drive.google.com/open?id=1_rAGQ9Ivw5dnOoNIGpWVti4lVY-XR-4Y
4. Setup - My setup is two Velodyne VLP16's and a Varisens IMU mounted as shown in the URDF.

Hello,

Thank you guys for all the help!  I have been getting better and better results lately, since I was able to improve my URDF to be more accurate.  However, I am still seeing a very strange thing happening in rviz.  When there is a global loop closure, the entire map seems to rotate at 20+ degrees about the z axis.  It is not a consistent rotation and I cannot seem to pin down why this is happening.  Here is a [video](https://drive.google.com/open?id=123V8C30wtePjeyLIFSQxvovYiIuFtFzm) of this phenomenon.  Other than that, the map looks pretty good.  Any help with this topic would be much appreciated!  If there are any glaring problems that you can find with my tuning, please mention them because I would also very much like to continue to improve my results. 

Thanks,

Sander

Sanderi44, 2018-09-18:
Any idea what this could be?

gaschler, 2018-09-19:
> W0824 11:25:45.566849 11772 rosbag_validate_main.cc:103] frame_id imu_link time 1534960294547956764: IMU linear acceleration is 2.34354 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 01.61065 0.238903 -1.68549
> W0824 11:25:45.567081 11772 rosbag_validate_main.cc:103] frame_id imu_link time 1534960294553710380: IMU linear acceleration is 2.34354 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 01.61065 0.238903 -1.68549

Your acceleration values are wrong. Fix this.
Perhaps this even explains the large change in optimization. Optimization looks at the acceleration values to fine-tune IMU extrinsic rotation.

> When there is a global loop closure, the entire map seems to rotate at 20+ degrees about the z axis. Here is a video of this phenomenon.

Haven't seen this before. Sure your setup creates only one trajectory? The beginning of the first active trajectory is fixed in the optimization.

Sorry for the delay. Hope this helps.

ojura, 2018-09-19:
Is your tracking frame (`imu_link`) Z-up? If not, could be a case of https://github.com/googlecartographer/cartographer/issues/882 (optimization locks rotation of the first trajectory's first submap around the wrong axis).

ojura, 2018-09-19:
I have taken a look at your video. It can be observed that your submaps are X-up (just like your tracking frame). I am 100% positive now that this is googlecartographer/cartographer#882.

Sanderi44, 2018-09-19:
I pulled the latest cartographer and the problem seems to be fixed.  Now the tracking frames are Z-up instead of X-up, which prevents any yaw rotation.  I am, however, finding that I am receiving some strange IMU data and am working to track down what is going on. 

Thanks for all the help!

ojura, 2018-09-19:
> Now the tracking frames are Z-up instead of X-up

You mean the submap poses? Thanks for reporting.

Sanderi44, 2018-09-19:
Yes, apologies, the submap poses.

closed 2020-10-21 by MichaelGrupp