cartographer-project/cartographer_ros #992: running warn

carl1218, 2018-08-26:
When i am running 2d slam online，I will have the following warning

`[ INFO] [1535273031.324569290]: I0826 16:43:51.000000 25121 collated_trajectory_builder.cc:81] scan rate: 38.10 Hz 2.62e-02 s +/- 3.60e-03 s (pulsed at 99.92% real time)
[ WARN] [1535273035.202418632]: W0826 16:43:55.000000 25121 range_data_collator.cc:77] Dropped 22 earlier points.
[ WARN] [1535273037.564914245]: W0826 16:43:57.000000 25121 range_data_collator.cc:77] Dropped 131 earlier points.
[ WARN] [1535273038.017008142]: W0826 16:43:58.000000 25121 range_data_collator.cc:77] Dropped 96 earlier points.
[ WARN] [1535273038.498344371]: W0826 16:43:58.000000 25121 range_data_collator.cc:77] Dropped 59 earlier points.`

This is my lua：

```
include "map_builder.lua"
include "trajectory_builder.lua"
options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "laser_link",
  published_frame = "laser_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_pose_extrapolator = true,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}
MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.submaps.num_range_data = 35
TRAJECTORY_BUILDER_2D.min_range = 0.1
TRAJECTORY_BUILDER_2D.max_range = 10.
TRAJECTORY_BUILDER_2D.missing_data_ray_length = 1.
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.1
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.translation_delta_cost_weight = 10.
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.rotation_delta_cost_weight = 1e-1
POSE_GRAPH.optimization_problem.huber_scale = 1e2
POSE_GRAPH.optimize_every_n_nodes = 35
POSE_GRAPH.constraint_builder.min_score = 0.65
```
This is launch:

```
<launch>
 <node name="urg_node" pkg="urg_node" type="urg_node">
     <param name="port" type="string" value="/dev/ttyACM0"/>
     <param name="frame_id" value="laser_link" />
     <param name="calibrate_time" value="ture" />
     <param name="cluster" value="1"/>
</node>
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename hokuyo.lua"
      output="screen">
      <remap from="scan" to="/scan" />
  </node>
 <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.1" />
 <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />
</launch>
```

kdaun, 2018-08-31:
Please record a rosbag without Cartographer running. Check that the warning occurs when you launch Cartographer with the rosbag. Then:

1. run `rosbag_validate` which does some checks on your sensor data. This
   tool often finds issues that can explain poor performance and must be fixed
   at recording time. Please post the full output of the tool into a
   GitHub Gist at https://gist.github.com/, then link it in the issue even if
   it does not report anything. You can run the tool like this:

   rosrun cartographer_ros cartographer_rosbag_validate -bag_filename <bag filename>

2. post a link to a Git repository containing a branch of
   `cartographer_ros` containing all the configuration, launch, and URDF files
   required to reproduce your issue.
3. post a link to a bag file we can use to reproduce your issue. Put it on
   Google Drive, Dropbox, any webserver or wherever it is publicly
   downloadable.

closed 2018-09-28 by carl1218

JackyGiang, 2018-12-26:
I meet the same issue and it took me already one week,how could you solved this issue,please inform me,thank you very much

JackyGiang, 2018-12-27:
Hi, today i found out the reason,its caused by the lidar type inyegrated with cartographer, when i replaced the lidar with rplidar A2, everything is ok.the old lidar is tianhe lidar,GL-50. but this lidar works well with gmapping slam.

JackyGiang, 2019-01-04:
> @JackyGiang
> Hi, Maybe not for this reason, i have two lidars,one HOKUYO-30lx one RPLIDAR-A2, But when i using them alone on 2d ，There's no big problem (just a little warning).When I use them together on 3d, it will Always appear this warn.

when I used 2d lidar, I did not use IMU, I attached the LUA and launch file for your referrence.
If use 3d lidar, cartographer requires to use IMU.
I think it may caused by the IMU working with cartographer.

iLanceS, 2022-03-04:
> You are receiving this because you were mentioned.

Did you finally solve this problem?

jimmy71131, 2022-06-04:
reset your system time
[image]