cartographer-project/cartographer_ros #996: Cartographer on an initial map.

santiagobellas, 2018-08-29:
Hello, would it be possible to run cartographer from an existing map?
The problem that I am having is that the maps obtained have a certain curvature or some type of deformation that could be corrected with the introduction of an initial map.

It is known that some walls are straight or that others form a right angle.

In my application it is very common to have an autocad or a base cartography to enter as an initial restriction, but I would not know how to do it.

Could I create a pbstream with an initial map from which to start?

Thanks greetings.

Jdu16, 2019-07-05:
Hi, 
Have you managed to find an approach on this subject ? I am also interested in doing SLAM with a prior map. I read a lot of articles on the subject but I am not enough experienced with SLAM to reproduce the methods they use. So if you have any hints on the subject it would greatly help me. 
Thanks a lot.

MichaelGrupp, 2020-10-21:
We only support pbstream files because Cartographer needs special data structures that are not saved in other formats.

closed 2020-10-21 by MichaelGrupp