cartographer-project/cartographer_ros #997: Option to set output pbstream filename in offline node

maracuya-robotics, 2018-08-29:
Currently the offline node writes the final state to a file which name is derived from the first bag file:

https://github.com/googlecartographer/cartographer_ros/blob/edf25139a2ad004f17c2c4b92a7a7c88ce9088f0/cartographer_ros/cartographer_ros/offline_node.cc#L371-L374

This overwrites previous runs without confirmation and makes it hard to compare different runs that use the same bag files.

It would be great to have an option like `-output_state_filename` which can be used to set path and name of the output pbstream file.

kdaun, 2018-08-31:
Sounds reasonable to be me. Feel free to implement it and create a pull request :+1: . Please make sure the default behavior does not change if no extra option is set.

closed 2020-10-21 by MichaelGrupp

maracuya-robotics, 2020-10-21:
See #1204 for the corresponding PR.