cartographer-project/cartographer_ros #998: Lidar data format

3pi10, 2018-08-30:
Hi, 

I am using velodyne hdl64 and per ros pointcloud2 message I have only 18 degree of a whole circle worth of data. Wondering if Cartographer support this or I have to merge one whole spin per message. Or if there is a setting that I can tell cartographer to use 20 messages before matching it to a submap? The result I am getting is a dense spherical point cloud that amlost everything is not matched and after playing with parameters the spherical transferemed to a cylinder of dense points still no matching. For some reason the orientation of scans changing unreastically big and I am guessing that cause mapping fails. No bias in imu data.

Sorry I can not provide any data!

Thank you for filing an issue!

It would help us tremendously if you run through the following checklist. This
ensures that we have the most information to quickly understand, analyze,
reproduce and eventually resolve your issue.

Please

1. run `rosbag_validate` which does some checks on your sensor data. This
   tool often finds issues that can explain poor performance and must be fixed
   at recording time. Please post the full output of the tool into a
   GitHub Gist at https://gist.github.com/, then link it in the issue even if
   it does not report anything. You can run the tool like this:

   rosrun cartographer_ros cartographer_rosbag_validate -bag_filename <bag filename>

2. post a link to a Git repository containing a branch of
   `cartographer_ros` containing all the configuration, launch, and URDF files
   required to reproduce your issue.
3. post a link to a bag file we can use to reproduce your issue. Put it on
   Google Drive, Dropbox, any webserver or wherever it is publicly
   downloadable.
4. remove this boilerplate text before submitting your issue.

We will likely be unable to help you without all of these points addressed.

kdaun, 2018-08-31:
The parameter `num_accumulated_range_data` sets the "_number of range data to accumulate into one unwarped, combined range data to use for scan matching_" (see [documentation](https://google-cartographer.readthedocs.io/en/latest/configuration.html)).
If you are using the 3D pipeline, try adding `TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 20` to your configuration file to combine 20 scans per scan match.

bochen87, 2020-06-17:
Cartographer supports both 2D and 3D. However, if you only have 18 degrees of the full circle worth of Scans from you velodyne, something is wrong. You’re effectively only using 5% of the data that your Lidar scan returns. Any sort of slam will be difficult with that.

closed 2020-06-17 by bochen87