cartographer-project/cartographer_ros #1000: Extract pdstream file data

woshidaye, 2018-09-06:
I got a pdstream file by running the offline_backpack_2d.launch. What is the content of this document? How do I extract the contents of the document and save it to .txt?

gaschler, 2018-09-06:
A pbstream file is Cartographer's own file format, which contains serialized, compressed protobuf messages of cartographer.mapping.proto.SerializationHeader and cartographer.mapping.proto.SerializedData, which in turn contain various submessages that contain essentially all data of a PoseGraph.

To extract data, see the implementation of MapBuilder::LoadState.

tigerk0430, 2018-09-06:
I can get a pose (x,y,z,quaternion) of EndOfTrajectory by using this code

```
  ::cartographer::io::ProtoStreamReader reader(FLAGS_pbstream_filename);
  cartographer_ros::NodeOptions node_options;
  cartographer_ros::TrajectoryOptions trajectory_options;
  std::tie(node_options, trajectory_options) =
      cartographer_ros::LoadOptions(FLAGS_configuration_directory, FLAGS_configuration_basename);
  auto map_builder =
      cartographer::common::make_unique<cartographer::mapping::MapBuilder>(
          node_options.map_builder_options);

  // ed: load pbstream
  map_builder->LoadState(&reader, true);

  // ed: get TrajectoryNodePose
  const auto node_poses = map_builder->pose_graph()->GetTrajectoryNodePoses();
  double pbstream_x = std::prev(node_poses.EndOfTrajectory(0))->data.global_pose.translation().x();
  double pbstream_y = std::prev(node_poses.EndOfTrajectory(0))->data.global_pose.translation().y();
  double pbstream_z = std::prev(node_poses.EndOfTrajectory(0))->data.global_pose.translation().z();
  double pbstream_quat_x = (std::prev(node_poses.EndOfTrajectory(0)))->data.global_pose.rotation().x();
  double pbstream_quat_y = (std::prev(node_poses.EndOfTrajectory(0)))->data.global_pose.rotation().y();
  double pbstream_quat_z = (std::prev(node_poses.EndOfTrajectory(0)))->data.global_pose.rotation().z();
  double pbstream_quat_w = (std::prev(node_poses.EndOfTrajectory(0)))->data.global_pose.rotation().w();
```

see more details from here
https://gist.github.com/tigerk0430/7f67a9ca10824ea56875ad4b7a449b49

woshidaye, 2018-09-11:
@tigerk0430 First of all, thank you for your answer. I want to extract the trajectory and submaps  in the pbstream file and save them as TXT files. Do you have any suggestions? I look forward to your reply.

tigerk0430, 2018-09-13:
@woshidaye 
well, If I were you, I could save `trajectory.txt` file by using code I posted using while loop.

but I have no idea how to save `submap.txt` file because I don't know how to extract submap data from `.pbstream` file. 
sorry for it and hope you post the way to extract `submap` data from `.pbstream`

MichaelGrupp, 2018-09-13:
With the latest Cartographer you can at least run `cartographer_pbstream info <pbstream_file> --all_debug_strings`... have fun with all that text :P

closed 2020-10-21 by MichaelGrupp