cartographer-project/cartographer_ros #1004: read_metrics service does not exist

dan9thsense, 2018-09-08:
I set the collect_metrics flag in the launch file, but the read_metrics service does not exist.  Checking rosservice list, there is no entry for read_metrics but the other cartographer services are there and they do work.

Running cartographer_node with --help does not show --collect_metrics as an option for the command line arguments.

cartographer version is 1.0.0, using ROS Kinetic

What am I doing wrong?

MichaelGrupp, 2018-09-09:
If you installed the Debian package from ROS: I added `/read_metrics` after the 1.0.0 release tag.

You would either need to wait until the next version is shipped as a package or install from source in the meantime.

dan9thsense, 2018-09-10:
Oh, ok, I will update the source code.  I made some custom mods anyway.

closed 2018-09-10 by dan9thsense