cartographer-project/cartographer_ros #1006: Fail to build 3D map with 4 Kinects

MetalSlugxx, 2018-09-10:
Hi, I am using 4 kinects to build 3D map with imu, but fail to generate generate a map. 
Cartographer node has received imu data topic(imu/data_raw) and point cloud topic(points2_1/points2_2/points2_3/points2_4) generated from four kinect cameras at different posotions and orientations. However in the rviz, the point cloud only shows around the origin of "map" and "odom". The other links, such as base_link, imu_link and 4 kinect links just appear anywhere randomly away from the mapping center. 

1. The "rosbag_validate" output result Gist link is [here](https://drive.google.com/open?id=1CHydEnjml7zsVz28aSPzShwB3m6QEqJJ). 

2. The configuration, launch, and URDF files link is [here].(https://drive.google.com/open?id=1CHydEnjml7zsVz28aSPzShwB3m6QEqJJ)
The launch file to run is [kinect_3d.launch](https://github.com/MetalSlugxx/cartographer_ros_kinect/blob/master/cartographer_ros/cartographer_ros/launch/kinect_3d.launch).

3. The tested rosbag is uploaded to [GoogleDrive]( https://drive.google.com/open?id=1CHydEnjml7zsVz28aSPzShwB3m6QEqJJ).

Any idea about that? Thank you in advance for your help and providing such good SLAM system!

MetalSlugxx, 2018-09-25:
Oh, I made a mistake on sim_time. I successfully build the map but still can't close the loop, so the result isn't good enough. I think it's the problem of parameters and imu data. I'll try on solving that.

closed 2018-09-25 by MetalSlugxx

christophTUM, 2021-08-15:
What mistake was that?