cartographer-project/cartographer_ros #1009: Warning: TF_OLD_DATA ignoring data from the past for frame base_link at time X according to authority unknown_publisher. Even with official demo rosbags and launch files

kajep09, 2018-09-11:
Hello

I have recently encountered a problem while using cartographer_ros on bagged data. I haven't had an opportunity to use it on my robot with live data recently. Whenever I try to do offline mapping with a rosbag file or localization using a pbstream file and another rosbag file I keep getting TF_OLD_DATA warnings from cartographer. 

So my thinking was that something had happend with my robot_state_publisher / tf tree or with my rosbag files, but then I decided to just try one of the official demo launch files on the rosbags from the public data section and I still get the warning.

Sometimes cartographer actually seems to be doing the mapping even with all the warning spam and other times there is no usable output (according to what I can see in rviz) and sometimes rviz crashes instantly.

What is even more strange is that sometimes cartographer launches completely fine without any warnings and runs as expected. 

All these possible outcomes are from launching the exact same launch file with the same rosbag and it is seemingly random whether or not it launches correctly or not.

Using the cartographer_paper_deutsches_museum.bag from:

https://storage.googleapis.com/cartographer-public-data/bags/backpack_2d/cartographer_paper_deutsches_museum.bag

Output of rosbag_validate:

`rosrun cartographer_ros cartographer_rosbag_validate -bag_filename=cartographer_paper_deutsches_museum.bag`
https://gist.github.com/kajep09/b5cb1c6ca67a7dfc6d48401b24ae7a37

I run `roslaunch cartographer_ros demo_backpack_2d.launch bag_filename:=${HOME}/Downloads/cartographer_paper_deutsches_museum.bag` and get the following output spammed at a very high frequency in the terminal:

```
Warning: TF_OLD_DATA ignoring data from the past for frame odom at time 1.45987e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.18/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame base_link at time 1.45987e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.18/src/buffer_core.cpp
```

Anyone got an idea about what is wrong or how to track down the problem? I have only experienced this pretty recently, but on two separate computers and with rosbag files that have previously worked just fine.

kajep09, 2018-09-14:
SInce creating the issue I have updated to the newest version of cartographer and cartographer_ros and done a rebuild and I have not seen the error/warning since. However, I still don't know the origin of the issue and will leave the issue open for a little while longer in case anyone wants to coment.

gollth, 2018-09-26:
We experience the same issue today. All out of sudden, the cartographer_node was publishing ridiculous timestamps, which did not change at all, causing other parts of the system to fail with the *TF_OLD_DATA* warning. 

**Clean & rebuilding the catkin workspace helped!**

As a small side note: before cartographer ros was producing this error, we had the same issue with a different hardware driver, which published a TF and a stamped message. Both time stamps had this problem, therefore it might be a case with `ros::Time::now()`. Maybe a ROS developer can comment on this

kajep09, 2018-10-05:
I am reopening this issue in hopes that anyone can help me track down the cause. After running well for a while I am now experiencing the issues again and it only happens when I launch cartographer (every other ROS node that relies on TF starts complaining along with cartographer_ros).

From 4 consecutive launches I got the following output in the terminal where I launched cartographer:
https://gist.github.com/kajep09/47c1c53ccedaa8ded895650d8c71a7f2

As you can see I get wildly varying time stamps in the warning message, from something close to the actual current epoch time (~1.53874e+09) to something totally crazy like 2.62019e+09. I always see the a few of the incorrect time stamps and then one seemingly correct one  (1.53874e+09), but the incorrect one changes every time I relaunch cartographer. All of my other nodes stop complaining about TF when I terminate cartographer.

The frame in question in the warning (in this case base_footprint) is the frame set as "published_frame" in the configuration files, or the odom frame if provide_odom is true.

```
  published_frame = "base_footprint",
  odom_frame = "odom",
  provide_odom_frame = false,
```

Excerpt below:

```
#1

Warning: TF_OLD_DATA ignoring data from the past for frame base_footprint at time 2.2677e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.18/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame base_footprint at time 1.53874e+09 according to authority unknown_publisher

#2

Warning: TF_OLD_DATA ignoring data from the past for frame base_footprint at time 1.62223e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.18/src/buffer_core.cpp
[safecop1_transducerm-18] killing on exit
Warning: TF_OLD_DATA ignoring data from the past for frame base_footprint at time 1.53874e+09 according to authority unknown_publisher

#3

Warning: TF_OLD_DATA ignoring data from the past for frame base_footprint at time 1.53874e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.18/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame base_footprint at time 1.53874e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.18/src/buffer_core.cpp

#4

Warning: TF_OLD_DATA ignoring data from the past for frame base_footprint at time 2.62019e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.18/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame base_footprint at time 1.53874e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 277 in /tmp/binarydeb/ros-kinetic-tf2-0.5.18/src/buffer_core.cpp
```

kajep09, 2018-10-05:
I have attached a picture of the top links of my TF tree. Everything except the map to `published_frame` are static transforms handled by robot_state_publisher. 

Both the average rate and the most recent / oldest transform being reported by cartographer_node are totally insane.

[image]

kajep09, 2018-10-17:
Closing the issue again as it seems to be resolved by a complete cleaning of all related workspaces and a full rebuild. 

Along with the TF warnings I originally posted about, I found out that RVIZ also complained and crashed sometimes with `'std::runtime_error' what(): Duration is out of dual 32-bit range`. This error let me to a lot of other people reporting similar errors around mid-september and as other ROS users reported there was an update to roscpp_core that relates to rostime on september the 6th that seems to have broken some ABI compatibility.

The proposed solution to those issues was to do a complete clean and rebuild of the workspace and at the same time making sure that there are no outstanding or broken updates with other ROS pkgs.

Hopefully this helps people looking for this issue in the future.

https://github.com/ros-planning/moveit/issues/1062
https://github.com/ros/roscpp_core/pull/82

closed 2018-10-17 by kajep09

nickvaras, 2019-02-11:
`sudo apt-get upgrade` solved this for me.

cgr71ii, 2019-03-04:
In my case, i solved this changing this:

`rospy.ServiceProxy('/gazebo/reset_simulation', Empty)`

for this:

`rospy.ServiceProxy('/gazebo/reset_world', Empty)`

churoc, 2023-06-30:
> `sudo apt-get upgrade` solved this for me.

I also fixed the issue after executing this command, and it was surprising to find out that it had nothing to do with ROS