cartographer-project/cartographer_ros #1010: no meaningful result

frqstn, 2018-09-11:
I tried to run the cartographer on my data and it is not producing any result . 
[image]

I am wondering if you can help me with this.  I have uploaded the information you asked for.
1) Here is the ros bag check result 
E0911 13:52:58.210934 603901 rosbag_validate_main.cc:389] IMU data (frame_id: "imu") has a large gap, largest is 0.99911 s, recommended is [0.0005, 0.005] s with no jitter.
I0911 13:52:58.211391 603901 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "imu" (frame_id: "imu"):
Count: 186  Min: 0.000110  Max: 0.999110  Mean: 0.107529
[0.000110, 0.100010)      ##################    Count: 166 (89.247314%) Total: 166 (89.247314%)
[0.100010, 0.199910)                            Count: 0 (0.000000%)    Total: 166 (89.247314%)
[0.199910, 0.299810)                            Count: 0 (0.000000%)    Total: 166 (89.247314%)
[0.299810, 0.399710)                            Count: 0 (0.000000%)    Total: 166 (89.247314%)
[0.399710, 0.499610)                            Count: 0 (0.000000%)    Total: 166 (89.247314%)
[0.499610, 0.599510)                            Count: 0 (0.000000%)    Total: 166 (89.247314%)
[0.599510, 0.699410)                            Count: 0 (0.000000%)    Total: 166 (89.247314%)
[0.699410, 0.799310)                            Count: 0 (0.000000%)    Total: 166 (89.247314%)
[0.799310, 0.899210)                            Count: 0 (0.000000%)    Total: 166 (89.247314%)
[0.899210, 0.999110]                      ##    Count: 20 (10.752688%)  Total: 186 (100.000000%)
E0911 13:52:58.211477 603901 rosbag_validate_main.cc:382] Point data (frame_id: "velodyne") has a large gap, largest is 0.99912 s, recommended is [0.0005, 0.05] s with no jitter.
I0911 13:52:58.211515 603901 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "lidar1" (frame_id: "velodyne"):
Count: 171  Min: 0.000119  Max: 0.999120  Mean: 0.116962
[0.000119, 0.100019)      ##################    Count: 151 (88.304092%) Total: 151 (88.304092%)
[0.100019, 0.199919)                            Count: 0 (0.000000%)    Total: 151 (88.304092%)
[0.199919, 0.299819)                            Count: 0 (0.000000%)    Total: 151 (88.304092%)
[0.299819, 0.399719)                            Count: 0 (0.000000%)    Total: 151 (88.304092%)
[0.399719, 0.499619)                            Count: 0 (0.000000%)    Total: 151 (88.304092%)
[0.499619, 0.599519)                            Count: 0 (0.000000%)    Total: 151 (88.304092%)
[0.599519, 0.699420)                            Count: 0 (0.000000%)    Total: 151 (88.304092%)
[0.699420, 0.799320)                            Count: 0 (0.000000%)    Total: 151 (88.304092%)
[0.799320, 0.899220)                            Count: 0 (0.000000%)    Total: 151 (88.304092%)
[0.899220, 0.999120]                      ##    Count: 20 (11.695907%)  Total: 171 (100.000000%)

2. post a link to a Git repository containing a branch of
   `cartographer_ros`  The files are test.urdf, test.lua and  offline_test.launch
https://github.com/frqstn/test

3. post a link to a bag file we can use to reproduce your issue. 
https://drive.google.com/open?id=12HvCawVZ7JvTZJkVR2Qz5hblpNoCWeHG

gaschler, 2018-09-12:
> [0.899220, 0.999120] ## Count: 20 (11.695907%) Total: 171 (100.000000%)

This means you have 20 large 0.9 seconds gaps in your laser data (and imu).
The recording cannot be used and needs to be recorded again.
(Do you send messages over network before recording?
Ideally, record on the device that is connected to the sensors.)

Thanks for following the issue template.

frqstn, 2018-09-13:
looked at the issue, it was some conversion problem. I had time fractions for unix epoch however ros bag was expecting nsec. I also removed imu sampling and removed combining a spin worth of lidar data 
in one message 
The new xray result 
[image]

1 ) This is the new bag file validation 

I0913 09:50:49.848582 535821 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "imu" (frame_id: "imu"):
Count: 2058  Min: 0.009999  Max: 0.010001  Mean: 0.010000
[0.009999, 0.009999)                       #    Count: 83 (4.033042%)   Total: 83 (4.033042%)
[0.009999, 0.009999)                            Count: 0 (0.000000%)    Total: 83 (4.033042%)
[0.009999, 0.010000)                            Count: 0 (0.000000%)    Total: 83 (4.033042%)
[0.010000, 0.010000)                            Count: 0 (0.000000%)    Total: 83 (4.033042%)
[0.010000, 0.010000)                            Count: 0 (0.000000%)    Total: 83 (4.033042%)
[0.010000, 0.010000)      ##################    Count: 1892 (91.933914%)        Total: 1975 (95.966957%)
[0.010000, 0.010000)                            Count: 0 (0.000000%)    Total: 1975 (95.966957%)
[0.010000, 0.010001)                            Count: 0 (0.000000%)    Total: 1975 (95.966957%)
[0.010001, 0.010001)                            Count: 0 (0.000000%)    Total: 1975 (95.966957%)
[0.010001, 0.010001]                       #    Count: 83 (4.033042%)   Total: 2058 (100.000000%)
I0913 09:50:49.849045 535821 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "lidar1" (frame_id: "velodyne"):
Count: 3103  Min: 0.005708  Max: 0.007496  Mean: 0.006636
[0.005708, 0.005887)                            Count: 4 (0.128908%)    Total: 4 (0.128908%)
[0.005887, 0.006066)                            Count: 43 (1.385756%)   Total: 47 (1.514663%)
[0.006066, 0.006244)                       #    Count: 202 (6.509829%)  Total: 249 (8.024492%)
[0.006244, 0.006423)                       #    Count: 229 (7.379955%)  Total: 478 (15.404448%)
[0.006423, 0.006602)                   #####    Count: 751 (24.202385%) Total: 1229 (39.606831%)
[0.006602, 0.006781)                ########    Count: 1221 (39.349018%)        Total: 2450 (78.955849%)
[0.006781, 0.006960)                      ##    Count: 348 (11.214953%) Total: 2798 (90.170799%)
[0.006960, 0.007138)                       #    Count: 216 (6.961006%)  Total: 3014 (97.131805%)
[0.007138, 0.007317)                       #    Count: 80 (2.578150%)   Total: 3094 (99.709961%)
[0.007317, 0.007496]                            Count: 9 (0.290042%)    Total: 3103 (100.000000%)

2)  the new bag file 
https://drive.google.com/open?id=1fSZDLAsEIdpAxZGZGwZ2sxulK8-y8_D5

3) the repo is the same I just changed 
TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 19

Thanks,

gaschler, 2018-09-13:
Likely your IMU to lidar transform is wrong.
If you didn't set it, the default won't match your setup with a very high probability.
You can set it in the urdf file.

frqstn, 2018-09-13:
I have changed that (test.urdf) . The Lidar is x (right), y (forward) and imu is x forward and y right and both right hand.

closed 2020-10-21 by MichaelGrupp