cartographer-project/cartographer_ros #1011: compile error: undefined reference to `absl::Mutex::Mutex()'

cccstudy, 2018-09-12:
there are no problem before #969 , but when I pull code and use catkin_make_isolated --install --use-ninja to build, the errors is reported 

I check the build_isolated/cartographer_ros/abseil/src/abseil-build/absl space, there has libabsl_synchronization.a

log  :
==> Processing catkin package: 'cartographer_ros'
==> Building with env: '/home/xx/catk/install_isolated/env.sh'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/xx/catk/build_isolated/cartographer_ros'
ninja: warning: multiple rules generate abseil/src/abseil-build/absl/synchronization/libabsl_synchronization.a. builds involving this target will not be correct; continuing anyway
ninja: no work to do.
==> ninja -j4 -l4 in '/home/xx/catk/build_isolated/cartographer_ros'
ninja: warning: multiple rules generate abseil/src/abseil-build/absl/synchronization/libabsl_synchronization.a. builds involving this target will not be correct; continuing anyway
[3/19] Performing configure step for 'abseil'
loading initial cache file /home/xx/catk/build_isolated/cartographer_ros/abseil/tmp/abseil-cache-Release.cmake
-- Configuring done
-- Generating done
-- Build files have been written to: /home/xx/catk/build_isolated/cartographer_ros/abseil/src/abseil-build
[4/19] Performing build step for 'abseil'
ninja: no work to do.
[10/16] Linking CXX executable /home/h...r_ros/cartographer_occupancy_grid_node
FAILED: : && /usr/bin/c++   -O3 -DNDEBUG   cartographer_ros/CMakeFiles/cartographer_occupancy_grid_node.dir/occupancy_grid_node_main.cc.o  -o /home/xx/catk/devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_occupancy_grid_node  -rdynamic /home/xx/catk/devel_isolated/cartographer_ros/lib/libcartographer_ros.a /home/xx/catk/install_isolated/lib/libcartographer.a /home/xx/catk/install_isolated/lib/libceres.a -lglog -lgflags -lpthread -lspqr -ltbb -ltbbmalloc -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lblas -lsuitesparseconfig -lrt -lcxsparse -lgomp -lpthread -lspqr -ltbb -ltbbmalloc -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lblas -lsuitesparseconfig -lrt -lcxsparse -lgomp -llua5.2 -lm -lglog -lgflags -lcairo /usr/local/lib/libprotobuf.a -lpthread abseil/src/abseil-build/absl/synchronization/libabsl_synchronization.a abseil/src/abseil-build/absl/debugging/libabsl_symbolize.a abseil/src/abseil-build/absl/time/libabsl_time.a abseil/src/abseil-build/absl/strings/libabsl_strings.a abseil/src/abseil-build/absl/strings/libstr_format_internal.a abseil/src/abseil-build/absl/strings/libstr_format_extension_internal.a abseil/src/abseil-build/absl/strings/libabsl_str_format.a abseil/src/abseil-build/absl/algorithm/libabsl_algorithm.a abseil/src/abseil-build/absl/base/libabsl_base.a abseil/src/abseil-build/absl/base/libabsl_dynamic_annotations.a abseil/src/abseil-build/absl/base/libabsl_malloc_internal.a abseil/src/abseil-build/absl/base/libabsl_spinlock_wait.a abseil/src/abseil-build/absl/base/libabsl_throw_delegate.a abseil/src/abseil-build/absl/container/libabsl_container.a abseil/src/abseil-build/absl/container/libtest_instance_tracker_lib.a abseil/src/abseil-build/absl/debugging/libabsl_debugging.a abseil/src/abseil-build/absl/debugging/libabsl_examine_stack.a abseil/src/abseil-build/absl/debugging/libabsl_failure_signal_handler.a abseil/src/abseil-build/absl/debugging/libabsl_leak_check.a abseil/src/abseil-build/absl/debugging/libabsl_stack_consumption.a abseil/src/abseil-build/absl/debugging/libabsl_stacktrace.a abseil/src/abseil-build/absl/memory/libabsl_memory.a abseil/src/abseil-build/absl/meta/libabsl_meta.a abseil/src/abseil-build/absl/numeric/libabsl_int128.a abseil/src/abseil-build/absl/numeric/libabsl_numeric.a abseil/src/abseil-build/absl/synchronization/libabsl_synchronization.a abseil/src/abseil-build/absl/types/libabsl_any.a abseil/src/abseil-build/absl/types/libabsl_bad_any_cast.a abseil/src/abseil-build/absl/types/libabsl_bad_optional_access.a abseil/src/abseil-build/absl/types/libabsl_optional.a abseil/src/abseil-build/absl/types/libabsl_span.a abseil/src/abseil-build/absl/types/libabsl_variant.a abseil/src/abseil-build/absl/utility/libabsl_utility.a -lboost_system -lboost_filesystem -lboost_thread -lboost_date_time -lboost_serialization -lboost_chrono -lboost_atomic -lpthread -lpcl_common /opt/ros/kinetic/lib/libeigen_conversions.so /opt/ros/kinetic/lib/liborocos-kdl.so.1.3.0 /opt/ros/kinetic/lib/librosbag.so /opt/ros/kinetic/lib/librosbag_storage.so /opt/ros/kinetic/lib/libroslz4.so -llz4 /opt/ros/kinetic/lib/libtopic_tools.so /opt/ros/kinetic/lib/libroslib.so /opt/ros/kinetic/lib/librospack.so -lpython2.7 -lboost_program_options /opt/ros/kinetic/lib/libtf2_ros.so /opt/ros/kinetic/lib/libactionlib.so /opt/ros/kinetic/lib/libmessage_filters.so /opt/ros/kinetic/lib/libtf2.so /opt/ros/kinetic/lib/liburdf.so -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world -ltinyxml /opt/ros/kinetic/lib/librosconsole_bridge.so /opt/ros/kinetic/lib/libroscpp.so -lboost_signals /opt/ros/kinetic/lib/librosconsole.so /opt/ros/kinetic/lib/librosconsole_log4cxx.so /opt/ros/kinetic/lib/librosconsole_backend_interface.so -llog4cxx /opt/ros/kinetic/lib/libxmlrpcpp.so /opt/ros/kinetic/lib/libroscpp_serialization.so /opt/ros/kinetic/lib/librostime.so /opt/ros/kinetic/lib/libcpp_common.so -lboost_system -lboost_filesystem -lboost_thread -lboost_date_time -lboost_serialization -lboost_chrono -lboost_atomic -lpthread -lpcl_common /opt/ros/kinetic/lib/libeigen_conversions.so /opt/ros/kinetic/lib/liborocos-kdl.so.1.3.0 /opt/ros/kinetic/lib/librosbag.so /opt/ros/kinetic/lib/librosbag_storage.so /opt/ros/kinetic/lib/libroslz4.so -llz4 /opt/ros/kinetic/lib/libtopic_tools.so /opt/ros/kinetic/lib/libroslib.so /opt/ros/kinetic/lib/librospack.so -lpython2.7 -lboost_program_options /opt/ros/kinetic/lib/libtf2_ros.so /opt/ros/kinetic/lib/libactionlib.so /opt/ros/kinetic/lib/libmessage_filters.so /opt/ros/kinetic/lib/libtf2.so /opt/ros/kinetic/lib/liburdf.so -lurdfdom_sensor -lurdfdom_model_state -lurdfdom_model -lurdfdom_world -ltinyxml /opt/ros/kinetic/lib/librosconsole_bridge.so /opt/ros/kinetic/lib/libroscpp.so -lboost_signals /opt/ros/kinetic/lib/librosconsole.so /opt/ros/kinetic/lib/librosconsole_log4cxx.so /opt/ros/kinetic/lib/librosconsole_backend_interface.so -llog4cxx /opt/ros/kinetic/lib/libxmlrpcpp.so /opt/ros/kinetic/lib/libroscpp_serialization.so /opt/ros/kinetic/lib/librostime.so /opt/ros/kinetic/lib/libcpp_common.so -lboost_iostreams -lboost_regex -lconsole_bridge -Wl,-rpath,/opt/ros/kinetic/lib: && :
cartographer_ros/CMakeFiles/cartographer_occupancy_grid_node.dir/occupancy_grid_node_main.cc.o: In function `main':
occupancy_grid_node_main.cc:(.text.startup+0x333): undefined reference to `absl::Mutex::Mutex()'
collect2: error: ld returned 1 exit status
[10/16] Linking CXX executable /home/h...lib/cartographer_ros/cartographer_node
FAILED: : && /usr/bin/c++   -O3 -DNDEBUG   cartographer_ros/CMakeFiles/cartographer_node.dir/node_main.cc.o  -o /home/xx/catk/devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_node  -rdynamic /home/xx/catk/devel_isolated/cartographer_ros/lib/libcartographer_ros.a /home/xx/catk/install_isolated/lib/libcartographer.a /home/xx/catk/install_isolated/lib/libceres.a -lglog -lgflags -lpthread -lspqr -ltbb -ltbbmalloc -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lblas -lsuitesparseconfig -lrt -lcxsparse -lgomp -lpthread -lspqr -ltbb -ltbbmalloc -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lblas -lsuitesparseconfig -lrt -lcxsparse -lgomp -llua5.2 -lm -lglog -lgflags -lcairo /usr/local/lib/libprotobuf.a -lpthread abseil/src/abseil-build/absl/synchronization/libabsl_synchronization.a abseil/src/abseil-build/absl/debugging/libabsl_symbolize.a abseil/src/abseil-build/absl/time/libabsl_time.a abseil/src/abseil-build/absl/strings/libabsl_strings.a abseil/src/abseil-build/absl/strings/libstr_format_internal.a abseil/src/abseil-build/absl/strings/libstr_format_extension_internal.a abseil/src/abseil-build/absl/strings/libabsl_str_format.a abseil/src/abseil-build/absl/algorithm/libabsl_algorithm.a abseil/src/abseil-build/absl/base/libabsl_base.a abseil/src/abseil-build/absl/base/libabsl_dynamic_annotations.a abseil/src/abseil-build/absl/base/libabsl_malloc_internal.a abseil/src/abseil-build/absl/base/libabsl_spinlock_wait.a abseil/src/abseil-build/absl/base/libabsl_throw_delegate.a abseil/src/abseil-build/absl/container/libabsl_container.a abseil/src/abseil-build/absl/container/libtest_in [... cut ...] ing<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&, std::vector<double, std::allocator<double> > const&)':
histogram.cc:(.text+0x8ff): undefined reference to `absl::Mutex::Mutex()'
collect2: error: ld returned 1 exit status
[10/16] Building CXX object cartograph..._map.dir/pbstream_to_ros_map_main.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer_ros':

cschuet, 2018-09-13:
Can you try nuking your workspace (incl ceres, cartographer, cartographer_ros...) and starting over again?

alireza-hosseini, 2018-09-18:
@cschuet @cccstudy I'm getting the same exact error after fetching the recent changes.

slamer-run, 2018-10-04:
@alireza-hosseini @cschuet @cccstudy 
I'm getting same error for now...

alireza-hosseini, 2018-10-09:
@slamer-run @cccstudy I could fix the problem by removing an existing installation of **abseil-cpp** which was installed using **apt** and I was not aware of that (probably installed when I was trying the released version of cartographer_ros). So make sure you don't have any abseil installation somewhere else.

cccstudy, 2018-10-11:
@alireza-hosseini  Thanks for your answer !!! I never using apt install abseil, but I find there have lib(abseil/ and absl/) in folder (/opt/ros/kinetic/...), then I fixed this problem through delete them.
  It may used ( --install ) parameter to install those older lib, but there are not update in the latest version.

closed 2018-10-24 by cccstudy