cartographer-project/cartographer_ros #1015: cannot switch to pure localization

danielm1405, 2018-09-13:
Hi, 
I am trying to run cartographer in pure 2D localization. To achieve it I set 
`TRAJECTORY_BUILDER.pure_localization = true`
in my configuration file. However I receive fatal error:
`Check failed: 1 == reference_counts_.count(key) (1 vs. 0) Key 'pure_localization' was used the wrong number of times.`
How should to set to pure localization mode?

cschuet, 2018-09-13:
That option changed. You have to set 

```
TRAJECTORY_BUILDER.pure_localization_trimmer = {
  max_submaps_to_keep = 3,
}
```

danielm1405, 2018-09-13:
Is it possible that the simple environment causes the localization problem? Should I expect better results while using pure localization in more complex surrounding?

cschuet, 2018-09-13:
I am not sure I understand the visualization. When you look in rviz do you see two trajectories? Trajectory 0 should contain the map you are trying to localize against. Trajectory 1 should be a trajectory that contains only 3 submaps (when the next one is created the oldest one should disappear).

Can you enable only INTER constraint visualizations? Are there any constraints going from trajectory 0 to trajectory 1?

danielm1405, 2018-09-17:
The issue above occurs while trying to use offline localization (playing from .bag file), however for some reason robot localizes itself perfectly while running online version. So the problem lays somewhere in the bag file. Do you know what can be the source of the problem?

closed 2019-04-25 by danielm1405