cartographer-project/cartographer_ros #1016: ROS Buildfarm: Build failed in Jenkins: Idoc__cartographer_ros__ubuntu_trusty_amd64 #151

cschuet, 2018-09-13:
The ROS build farm seems to fail to compile cartographer_ros

Seems to related to the ReadMetrics ROS service

```
CMake Error at /tmp/catkin_workspace/build_isolated/cartographer_ros_msgs/cmake/cartographer_ros_msgs-genmsg.cmake:3 (message):
  Could not find messages which
  '/tmp/catkin_workspace/src/cartographer_ros/cartographer_ros_msgs/srv/ReadMetrics.srv'
  depends on.  Did you forget to specify generate_messages(DEPENDENCIES ...)?

  Cannot locate message [MetricFamily] in package [cartographer_ros_msgs]
  with paths
  [['/tmp/catkin_workspace/src/cartographer_ros/cartographer_ros_msgs/msg']]
Call Stack (most recent call first):
  /opt/ros/indigo/share/genmsg/cmake/genmsg-extras.cmake:307 (include)
  CMakeLists.txt:91 (generate_messages)

-- Configuring incomplete, errors occurred!
See also "/tmp/catkin_workspace/build_isolated/cartographer_ros_msgs/CMakeFiles/CMakeOutput.log".
See also "/tmp/catkin_workspace/build_isolated/cartographer_ros_msgs/CMakeFiles/CMakeError.log".
<== Failed to process package 'cartographer_ros_msgs': 
  Command '['cmake', '/tmp/catkin_workspace/src/cartographer_ros/cartographer_ros_msgs', '-DCATKIN_DEVEL_PREFIX=/tmp/catkin_workspace/devel_isolated/cartographer_ros_msgs', '-DCMAKE_INSTALL_PREFIX=/tmp/catkin_workspace/install_isolated', '-DCATKIN_SKIP_TESTING=1', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /tmp/catkin_workspace/build_isolated/cartographer_ros_msgs && cmake /tmp/catkin_workspace/src/cartographer_ros/cartographer_ros_msgs -DCATKIN_DEVEL_PREFIX=/tmp/catkin_workspace/devel_isolated/cartographer_ros_msgs -DCMAKE_INSTALL_PREFIX=/tmp/catkin_workspace/install_isolated -DCATKIN_SKIP_TESTING=1 -G 'Unix Makefiles'

Command failed, exiting.
# END SUBSECTION
Build step 'Execute shell' marked build as failure
$ ssh-agent -k
unset SSH_AUTH_SOCK;
unset SSH_AGENT_PID;
echo Agent pid 26927 killed;
[ssh-agent] Stopped.

```

MichaelGrupp, 2018-09-13:
Thought this fixed it? https://github.com/googlecartographer/cartographer_ros/pull/927

cschuet, 2018-09-13:
Email came in earlier today. Maybe we need to cherry pick this commit?

@mikaelarguedas Can you give some direction here?

mikaelarguedas, 2018-09-14:
@cschuet I think the job is failing because the messages are placed in subfolders of the `msg` directory.
ROS recommends placing all `msg` files at the root of the `msg` folder (looking at all packages indexed in kinetic it looks like cartographer_ros_msgs is the only package using subfolders).
That assumption is kept by the documentation jobs that locate all message files using https://github.com/ros-infrastructure/ros_buildfarm/blob/a9f0554840dda39272d76ff19f67c1675d77f8d6/ros_buildfarm/templates/doc/CMakeLists.txt.em#L75

I recommend you to move the messages in the `msg` folder and that will fix the failing job.

As of why only that job is failing: all the other distributions are tracking release branches for documentation and not master. e.g. for ROS melodic, the 1.0.0 branch is [used for documentation](https://github.com/ros/rosdistro/blob/81ac76c5a86a662a6970a319facb839620c465fa/melodic/distribution.yaml#L303) so the job passes as all messages are at the root of `msg` on that branch

closed 2018-10-17 by cschuet