cartographer-project/cartographer_ros #1023: Cartographer has no access to my sensor data

Semih67, 2018-09-18:
I want to use google cartographer and I have a RGB (3d camera) sensor. Its publishing the topic /ifm3d/camera/cloud. And I wrote a node for the imu data to test if it works. I looked in rviz and I know that I get data out of my imu node. The topic for this is /imu/data_row. So I have changed the launch file of cartographer_ros. I added `<remap from="/points2" to="/ifm3d/camera/cloud"/>` and `<remap from="/imu" to="/imu/data_row"/>` I deleted the rows with points2_1 and points2_2. I changed nothing in the urdf files. I changed the backpack_3d.lua file. So my num_point_clouds = 1. I noticed that the remapping is not working. I also tried to start cartographer with this command: roslaunch cartographer_ros backpack_3d.launch imu:=/imu/data_raw points2:=/ifm3d/camera/cloud . When I start it I get a warning: ..... Queue waiting for data (0,imu). If I start the node for the imu I get this warning after 5 min. But if I don't start the node there is no warning just the [INFO].

And in rviz I get the warning: No map received. And under submaps if I select the topic /submap_list I get the error that the frame [map] doesn't exist (Transform[sender=unknown_publisher]). My rqt graph looks like this: 

[image]

I don't know if it says something to you but if I start the backpack_3d.launch, I get the information in the end: [Info] ... added submap 1 and [Info] ... added trajectory with ID '0'.

So my problem is that cartographer is not publishing a map. And I think the reason for this is that I do the remapping wrong and it doesn't get the sensor data.

gaschler, 2018-09-18:
Since a few weeks, the master version warns you about missing topics and mismatches.
Look closely for warnings like

> [ WARN] [1537257047.284720607, 1432647018.922066652]: W0918 09:50:47.000000 212505 node.cc:821] Expected topic "points2_1" (trajectory 0) (resolved topic "/horizontal_laser_3d") but no publisher is currently active.
> [ WARN] [1537257047.284845243, 1432647018.922066652]: W0918 09:50:47.000000 212505 node.cc:821] Expected topic "points2_2" (trajectory 0) (resolved topic "/vertical_laser_3d") but no publisher is currently active.
> [ WARN] [1537257047.284940290, 1432647018.922066652]: W0918 09:50:47.000000 212505 node.cc:830] Currently available topics are: /horizontal_laser_2d,/vertical_laser_2d,/move_base_simple/goal,/constraint_list,/submap_list,/scan_matched_points2,/rosout,/initialpose,/tf,/clicked_point,/tf_static,/clock,/imu,/rosout_agg,/map,/trajectory_node_list,/landmark_poses_list,

(These are only warnings because the node cannot decide whether these topics are only starting up late.)
Correct the lua and launch files to resolve all topic mismatches.

Also, the following should not happen:

> [ WARN] [1537257047.318613837, 1432647018.952311050]: W0918 09:50:47.000000 212505 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2_1)

It indicates that you have configured `num_point_clouds > 0` but no PointCloud2 messages are coming in.

> And in rviz I get the warning: No map received. And under submaps if I select the topic /submap_list I get the error that the frame [map] doesn't exist (Transform[sender=unknown_publisher]). My rqt graph looks like this:

There seems to be confusion about topics and frames. You show the frame/transform graph. This is unrelated to topic names (even though some people choose similar names).

To fix frame issues, correct your urdf file. The frames in urdf need to match the frame_id in the incoming message headers.

Semih67, 2018-09-18:
"  [ WARN] [1537257047.318613837, 1432647018.952311050]: W0918 09:50:47.000000 212505 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2_1)

It indicates that you have configured num_point_clouds > 0 but no PointCloud2 messages are coming in." Instead of this I get a warning for imu. And I'm working with points not points2_1. I didn't understand the difference.
"To fix frame issues, correct your urdf file. The frames in urdf need to match the frame_id in the incoming message headers." Where can I see the incoming message headers?

MichaelGrupp, 2020-10-21:
ROS topic remappings are a way to fix mismatching topic names. See `rosnode info /cartographer_node` for what Cartographer subscribes to.

closed 2020-10-21 by MichaelGrupp