cartographer-project/cartographer_ros #1024: Error while using pure localization demo

alfred31, 2018-09-19:
Hello,
I am trying to use pure localization demos, so I am simply following the steps shown in the cartographer, this are the steps I am following:
1. I downloaded the bags from the website
2. I run : roslaunch cartographer_ros offline_backpack_2d.launch bag_filenames:=${HOME}/Downloads/b2-2016-04-05-14-44-52.bag
I can see that rviz opens and the submaps are being build , I can see that there is a problem though, since the submaps are added and then cleared from the map, which means that I only see 3 or 4 submaps at a time.
3. I tried to run then : 
roslaunch cartographer_ros demo_backpack_2d_localization.launch \ load_state_filename:=${HOME}/Downloads/b2-2016-04-05-14-44-52.bag.pbstream \bag_filename:=${HOME}/Downloads/b2-2018-04-27-12-31-41.bag

And I get this error:
[FATAL] [1537337030.165972614]: F0919 08:03:50.000000  7704 proto_stream_deserializer.cc:69] Check failed: pose_graph_.pose_graph().trajectory_size() == all_trajectory_builder_options_.all_trajectory_builder_options() .options_with_sensor_ids_size() (0 vs. 1)
I assume this is due to problem detected in point 2 above.
¿How can I solve this issue? ¿What am I doing wrong?
Any help would be appreciated.
Thanks.
Kind Regards.

gaschler, 2018-09-19:
>  I only see 3 or 4 submaps at a time.

Sure you don't have the pure localization trimmer in trajectory_builder.lua?
(You need to turn that one off for the mapping step.)

alfred31, 2018-09-19:
This is my trajectory_builder.lua. 

include "trajectory_builder_2d.lua"
include "trajectory_builder_3d.lua"

TRAJECTORY_BUILDER = {
  trajectory_builder_2d = TRAJECTORY_BUILDER_2D,
  trajectory_builder_3d = TRAJECTORY_BUILDER_3D,
  pure_localization = false,
}

I tried it also by setting the pure_localization to true, it did not work though.
I don't see any pure localization trimmer.

gaschler, 2018-09-19:
The syntax of trajectory_builder.lua has changed.

For the mapping step, don't have pure_localization_trimmer

For the localization step, have
```
 pure_localization_trimmer = {
   max_submaps_to_keep = 3,
 },
```

alfred31, 2018-09-19:
Should I add:
pure_localization_trimmer = {
   max_submaps_to_keep = 3,
 },

to my trajectory builder.lua?

gaschler, 2018-09-19:
Again, you need to use separate lua configs for the mapping and the localization phase.
The two launch files in https://github.com/googlecartographer/cartographer_ros/issues/1024#issue-361593391
need to specify separate lua files, with include separate lua files that contain the TRAJECTORY_BUILDER part.

alfred31, 2018-09-19:
Ok, thanks for your help.

I'll try to get this done, and post the outcomes.

Kind Regards

alfred31, 2018-09-19:
Hi,
I finally managed to make it work with your explanations.

Thank you very much!

Kind Regards

closed 2018-09-19 by alfred31

ivan140, 2021-07-20:
@alfred31 I'm experiencing the same issue you had when launching the pure localization demo. Could you please explain how you solved the issue?

bainro, 2023-05-14:
@ivan140 The original mapping and localization configuration .lua files need to be different. The localization_trimmer option can only be on after originally mapping, when in localization mode.