cartographer-project/cartographer_ros #1027: Example of using ROS google Cartographer with HDL32 or VLP16 pcap?

MontyTHall, 2018-09-24:
I'm trying to get the cartographer to run with HDL32/VLP32/VLP16 pcaps.  Anybody know how to setup cartographer to work w/ pcaps?  Is there an IRC or forum forum for this project?  I've cloned backpack_2d.launch, backpack_2d.urdf, backpack_2d.lua as rick.launch, rick.urdf, rick.lua.

In other programs, like LOAM, I'd launch it, then launch the 'roslaunch velodyne_pointcloud VLP16.launch pcap="blah"'  I was hoping to "roslaunch cartographer_ros rick.launch" and then launch the velodyne point cloud pcap.  When rviz pops up, I see nothing.

My Lua File:
-- Copyright 2016 The Cartographer Authors
--
-- Licensed under the Apache License, Version 2.0 (the "License");
-- you may not use this file except in compliance with the License.
-- You may obtain a copy of the License at
--
--      http://www.apache.org/licenses/LICENSE-2.0
--
-- Unless required by applicable law or agreed to in writing, software
-- distributed under the License is distributed on an "AS IS" BASIS,
-- WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
-- See the License for the specific language governing permissions and
-- limitations under the License.

include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_pose_extrapolator = true,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 10

return options

My URDF:
<robot name="rick_2d">

  <link name="base_link" />
  <link name="velodyne" />
  <joint name="imu_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="velodyne" />
    <origin xyz="0 0 0" />
  </joint>

</robot>

My Launch file:
<launch>

  <param name="robot_description"
    textfile="$(find cartographer_ros)/urdf/rick.urdf" />

  <node name="robot_state_publisher" pkg="robot_state_publisher"
    type="robot_state_publisher" />

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename rick.lua"
      output="screen">
    
  </node>

  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />

  <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />

</launch>

gaschler, 2018-09-25:
> [1537812445.970344829]: W0924 14:07:25.000000 22978 ordered_multi_queue.cc:155] Queue waiting for data: (0, imu)

If you don't publish imu data, you need to set `TRAJECTORY_BUILDER_2D.use_imu_data = false`

What message type is your laser scanner publishing? Since you seem to use 3D lidars, you probably publish PointCloud2 messages, in which case you need to set ` num_point_clouds = 1,`

As the issue template says, posting the output of cartographer_rosbag_validate would help (finding these  configuration/input mismatches)

MontyTHall, 2018-09-25:
Isn't cartographer_rosbag_validate for bag files?  I'm only using pcaps.  However, "TRAJECTORY_BUILDER_2D.use_imu_data=false" silenced the IMU messages.  "stuff" starts to appear, but it looks like it's it's updating only occasionaly and isn't building a map.  I was going to post a screen shot, but it doesn't look like I can.  My pcap is me driving around some city blocks.  All I get is a white blob after a few minutes.

MontyTHall, 2018-09-25:
Using "num_laser_scan = 1 & num_point_clouds=0" I get "something/junk".  If I "num_laser_scan=0 & num_point_clouds=1" I get nothing.  I tried "<remap from"velodyne_points" to "point2"/> in my cartographer_node with no effect.  Note, I used "point2" because of this message on the console: "[ WARN] [1537894907.378683923]: W0925 13:01:47.000000 21935 node.cc:821] Expected topic "points2" (trajectory 0) (resolved topic "/points2") but no publisher is currently active.
"

gaschler, 2018-09-26:
> Isn't cartographer_rosbag_validate for bag files? I'm only using pcaps. 

Sure, but we can hardly help unless you record a bag file and post the rosbag_validate output (as text).

MontyTHall, 2018-09-26:
Here's what I get run I run rosbag validate:

```
E0926 10:19:03.495780  9345 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636735682078657500, current one at time 636735682078657500
E0926 10:19:03.496206  9345 rosbag_validate_main.cc:316] Sensor with frame_id "velodyne" jumps backwards in time, i.e. timestamps are not strictly increasing. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0926 10:19:03.496220  9345 rosbag_validate_main.cc:327] frame_id "velodyne" is send on multiple topics. It was seen at least on /scan and /velodyne_points
E0926 10:19:03.499403  9345 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636735682079658630, current one at time 636735682079658630
E0926 10:19:03.499581  9345 rosbag_validate_main.cc:316] Sensor with frame_id "velodyne" jumps backwards in time, i.e. timestamps are not strictly increasing. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0926 10:19:03.499591  9345 rosbag_validate_main.cc:327] frame_id "velodyne" is send on multiple topics. It was seen at least on /scan and /velodyne_points
E0926 10:19:03.502071  9345 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636735682080659790, current one at time 636735682080659790
E0926 10:19:03.502099  9345 rosbag_validate_main.cc:316] Sensor with frame_id "velodyne" jumps backwards in time, i.e. timestamps are not strictly increasing. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0926 10:19:03.502105  9345 rosbag_validate_main.cc:327] frame_id "velodyne" is send on multiple topics. It was seen at least on /scan and /velodyne_points
E0926 10:19:03.502382  9345 rosbag_validate_main.cc:131] Input contains transform message from frame_id "map" to child_frame_id "odom". This is almost always output published by cartographer and should not appear as input. (Unless you have some complex remove_frames configuration, this is will not work. Simplest solution is to record input without cartographer running.)
E0926 10:19:03.519035  9345 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636735682087667380, current one at time 636735682087667380
E0926 10:19:03.525486  9345 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636735682090670650, current one at time 636735682090670650
E0926 10:19:03.536233  9345 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636735682095676160, current one at time 636735682095676160
W0926 10:19:03.818801  9345 rosbag_validate_main.cc:203] Sensor with frame_id "velodyne" range measurements have longest overlap of 0 s
E0926 10:19:03.818845  9345 rosbag_validate_main.cc:383] Point data (frame_id: "map") has a large gap, largest is 1.00114 s, recommended is [0.0005, 0.05] s with no jitter.
I0926 10:19:03.818882  9345 rosbag_validate_main.cc:399] Time delta histogram for consecutive messages on topic "/scan_matched_points2" (frame_id: "map"):
Count: 13  Min: 1.001073  Max: 1.001144  Mean: 1.001106
[1.001073, 1.001080)	                 ###	Count: 2 (15.384615%)	Total: 2 (15.384615%)
[1.001080, 1.001087)	                 ###	Count: 2 (15.384615%)	Total: 4 (30.769230%)
[1.001087, 1.001094)	                  ##	Count: 1 (7.692307%)	Total: 5 (38.461540%)
[1.001094, 1.001101)	                    	Count: 0 (0.000000%)	Total: 5 (38.461540%)
[1.001101, 1.001109)	               #####	Count: 3 (23.076923%)	Total: 8 (61.538460%)
[1.001109, 1.001116)	                  ##	Count: 1 (7.692307%)	Total: 9 (69.230766%)
[1.001116, 1.001123)	                  ##	Count: 1 (7.692307%)	Total: 10 (76.923080%)
[1.001123, 1.001130)	                    	Count: 0 (0.000000%)	Total: 10 (76.923080%)
[1.001130, 1.001137)	                  ##	Count: 1 (7.692307%)	Total: 11 (84.615387%)
[1.001137, 1.001144]	                 ###	Count: 2 (15.384615%)	Total: 13 (100.000000%)
E0926 10:19:03.818898  9345 rosbag_validate_main.cc:383] Point data (frame_id: "velodyne") has a large gap, largest is 0.100157 s, recommended is [0.0005, 0.05] s with no jitter.
I0926 10:19:03.818928  9345 rosbag_validate_main.cc:399] Time delta histogram for consecutive messages on topic "/scan" (frame_id: "velodyne"):
Count: 297  Min: 0.000000  Max: 0.100157  Mean: 0.049887
[0.000000, 0.010016)	          ##########	Count: 149 (50.168350%)	Total: 149 (50.168350%)
[0.010016, 0.020031)	                    	Count: 0 (0.000000%)	Total: 149 (50.168350%)
[0.020031, 0.030047)	                    	Count: 0 (0.000000%)	Total: 149 (50.168350%)
[0.030047, 0.040063)	                    	Count: 0 (0.000000%)	Total: 149 (50.168350%)
[0.040063, 0.050079)	                    	Count: 0 (0.000000%)	Total: 149 (50.168350%)
[0.050079, 0.060094)	                    	Count: 0 (0.000000%)	Total: 149 (50.168350%)
[0.060094, 0.070110)	                    	Count: 0 (0.000000%)	Total: 149 (50.168350%)
[0.070110, 0.080126)	                    	Count: 0 (0.000000%)	Total: 149 (50.168350%)
[0.080126, 0.090141)	                    	Count: 0 (0.000000%)	Total: 149 (50.168350%)
[0.090141, 0.100157]	          ##########	Count: 148 (49.831650%)	Total: 297 (100.000000%)
```

MontyTHall, 2018-09-26:
This is after running 180 seconds, I think the above was only for 20-30 seconds.
```
E0926 10:28:44.323010 12648 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636735687418775430, current one at time 636735687418775430
E0926 10:28:44.323343 12648 rosbag_validate_main.cc:316] Sensor with frame_id "velodyne" jumps backwards in time, i.e. timestamps are not strictly increasing. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0926 10:28:44.323354 12648 rosbag_validate_main.cc:327] frame_id "velodyne" is send on multiple topics. It was seen at least on /scan and /velodyne_points
E0926 10:28:44.325423 12648 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636735687419776410, current one at time 636735687419776410
E0926 10:28:44.325450 12648 rosbag_validate_main.cc:316] Sensor with frame_id "velodyne" jumps backwards in time, i.e. timestamps are not strictly increasing. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0926 10:28:44.325456 12648 rosbag_validate_main.cc:327] frame_id "velodyne" is send on multiple topics. It was seen at least on /scan and /velodyne_points
E0926 10:28:44.327921 12648 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636735687420777710, current one at time 636735687420777710
E0926 10:28:44.328073 12648 rosbag_validate_main.cc:316] Sensor with frame_id "velodyne" jumps backwards in time, i.e. timestamps are not strictly increasing. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0926 10:28:44.328083 12648 rosbag_validate_main.cc:327] frame_id "velodyne" is send on multiple topics. It was seen at least on /scan and /velodyne_points
E0926 10:28:44.341763 12648 rosbag_validate_main.cc:131] Input contains transform message from frame_id "map" to child_frame_id "odom". This is almost always output published by cartographer and should not appear as input. (Unless you have some complex remove_frames configuration, this is will not work. Simplest solution is to record input without cartographer running.)
E0926 10:28:44.343950 12648 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636735687427785110, current one at time 636735687427785110
E0926 10:28:44.346051 12648 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636735687428786330, current one at time 636735687428786330
E0926 10:28:44.360522 12648 rosbag_validate_main.cc:160] Sensor with frame_id "velodyne" is not sequential in time.Previous range message ends at time 636735687434793090, current one at time 636735687434793090
W0926 10:28:48.734419 12648 rosbag_validate_main.cc:203] Sensor with frame_id "velodyne" range measurements have longest overlap of 0 s
E0926 10:28:48.734469 12648 rosbag_validate_main.cc:383] Point data (frame_id: "map") has a large gap, largest is 1.25172 s, recommended is [0.0005, 0.05] s with no jitter.
I0926 10:28:48.734508 12648 rosbag_validate_main.cc:399] Time delta histogram for consecutive messages on topic "/scan_matched_points2" (frame_id: "map"):
Count: 163  Min: 1.001061  Max: 1.251719  Mean: 1.008982
[1.001061, 1.026127)	 ###################	Count: 154 (94.478531%)	Total: 154 (94.478531%)
[1.026127, 1.051193)	                    	Count: 1 (0.613497%)	Total: 155 (95.092026%)
[1.051193, 1.076258)	                    	Count: 2 (1.226994%)	Total: 157 (96.319016%)
[1.076258, 1.101324)	                    	Count: 1 (0.613497%)	Total: 158 (96.932518%)
[1.101324, 1.126390)	                    	Count: 0 (0.000000%)	Total: 158 (96.932518%)
[1.126390, 1.151456)	                    	Count: 0 (0.000000%)	Total: 158 (96.932518%)
[1.151456, 1.176522)	                    	Count: 2 (1.226994%)	Total: 160 (98.159508%)
[1.176522, 1.201587)	                    	Count: 0 (0.000000%)	Total: 160 (98.159508%)
[1.201587, 1.226653)	                    	Count: 1 (0.613497%)	Total: 161 (98.773003%)
[1.226653, 1.251719]	                    	Count: 2 (1.226994%)	Total: 163 (100.000000%)
E0926 10:28:48.734534 12648 rosbag_validate_main.cc:383] Point data (frame_id: "velodyne") has a large gap, largest is 0.195791 s, recommended is [0.0005, 0.05] s with no jitter.
I0926 10:28:48.734596 12648 rosbag_validate_main.cc:399] Time delta histogram for consecutive messages on topic "/scan" (frame_id: "velodyne"):
Count: 3283  Min: 0.000000  Max: 0.195791  Mean: 0.050431
[0.000000, 0.019579)	          ##########	Count: 1642 (50.015228%)	Total: 1642 (50.015228%)
[0.019579, 0.039158)	                    	Count: 0 (0.000000%)	Total: 1642 (50.015228%)
[0.039158, 0.058737)	                    	Count: 0 (0.000000%)	Total: 1642 (50.015228%)
[0.058737, 0.078316)	                    	Count: 0 (0.000000%)	Total: 1642 (50.015228%)
[0.078316, 0.097896)	                    	Count: 0 (0.000000%)	Total: 1642 (50.015228%)
[0.097896, 0.117475)	          ##########	Count: 1601 (48.766373%)	Total: 3243 (98.781601%)
[0.117475, 0.137054)	                    	Count: 35 (1.066098%)	Total: 3278 (99.847702%)
[0.137054, 0.156633)	                    	Count: 2 (0.060920%)	Total: 3280 (99.908623%)
[0.156633, 0.176212)	                    	Count: 1 (0.030460%)	Total: 3281 (99.939079%)
[0.176212, 0.195791]	                    	Count: 2 (0.060920%)	Total: 3283 (100.000000%)
```

pephabet, 2019-01-30:
Has this issue been resolved? Where is the solution?

MichaelGrupp, 2020-10-21:
Closing for inactivity. I would recommend to take a look at the messages printed by rosbag_validate, e.g. duplicate timestamps are problematic for Cartographer.

closed 2020-10-21 by MichaelGrupp