cartographer-project/cartographer_ros #1030: [ WARN]Queue waiting for data: (0, imu)

oneway234, 2018-09-27:
Hi, 
I tried to slam by cartographer but I got an issue. I tried to figure  the problem but couldn't find. I think that the topic name of imu which is /imu/data  maybe the problem, but I've write "<remap from="imu" to="/imu/data" />" in launch file. And I also checked others, but still got nothing.

1. Result of rosbag_validate: 
   https://gist.github.com/oneway234/bdd70f275f473e444b456f266152eb23

2. Link to my issue Git repository :
    https://github.com/oneway234/cartographer_vlp16
    launch file : playbag.launch
    urdf : backpack_3d.urdf
    lua file : 3d_vlp.lua

3. Link to my bag file :
    https://drive.google.com/file/d/10qMvY2Bf_-UpQAQsdGUXFcFJzkgu9iXO/view?usp=sharing

gaschler, 2018-09-27:
> E0927 20:25:57.264914 21519 rosbag_validate_main.cc:382] Point data (frame_id: "velodyne") has a large gap, largest is 2.28095e+08 s, recommended is [0.0005, 0.05] s with no jitter.
> I0927 20:25:57.265089 21519 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/velodyne_points" (frame_id: "velodyne"):
> Count: 3458  Min: -228094976.000000  Max: 228094976.000000  Mean: -32723.992188
> [-228094976.000000, -182475968.000000)	          ##########	Count: 1729 (50.000000%)	Total: 1729 (50.000000%)
> [-182475968.000000, -136856992.000000)	                    	Count: 0 (0.000000%)	Total: 1729 (50.000000%)
> [-136856992.000000, -91237984.000000)	                    	Count: 0 (0.000000%)	Total: 1729 (50.000000%)
> [-91237984.000000, -45619000.000000)	                    	Count: 0 (0.000000%)	Total: 1729 (50.000000%)
> [-45619000.000000, 0.000000)	                    	Count: 0 (0.000000%)	Total: 1729 (50.000000%)
> [0.000000, 45619000.000000)	                    	Count: 0 (0.000000%)	Total: 1729 (50.000000%)
> [45619000.000000, 91237984.000000)	                    	Count: 0 (0.000000%)	Total: 1729 (50.000000%)
> [91237984.000000, 136856992.000000)	                    	Count: 1 (0.028918%)	Total: 1730 (50.028919%)
> [136856992.000000, 182475968.000000)	                    	Count: 0 (0.000000%)	Total: 1730 (50.028919%)
> [182475968.000000, 228094976.000000]	          ##########	Count: 1728 (49.971081%)	Total: 3458 (100.000000%)

Header timestamps of your velodyne points are corrupted, you need to fix this first.

liushuo1201, 2019-07-31:
I also meet the same problem and I want to know how did you solve this problem.
Looking forward to your reply, thanks a lot.

closed 2020-10-21 by MichaelGrupp