cartographer-project/cartographer_ros #1031: Why the cartographer_ros public "map" topic can not be accepted correctly by ros navigation?

CccYunxiao, 2018-09-28:
When i use cartographer to create a map that display in rviz correctly, but when i use map_server to save the map which without include the occupancy information.

Then i create a pbstream file and run the node "cartographer_pbstream_to_ros_map", in the end the map contains occupation info.

But i have some questions:

1. why the cartographer_ros public "map" topic can not be accepted correctly by ros navigation?

2. how can i make ros navigation package(eg. map_server, move_base ...) correctly read the map create by cartographer_ros?

3. in the file  cartographer_ros/cartographer_ros/cartographer_ros/occupancy_grid_node_main.cc, the function "void PublishOccupancyGrid()" just has declaration, but without realization. Did your team ignore the function or did i not find the realization?

CccYunxiao, 2018-09-28:
@SirVer @MichaelGrupp @jihoonl @ojura @gaschler @cschuet 
I hope get your help. Thx!

MichaelGrupp, 2018-09-28:
You said you successfully created a ROS map with `cartographer_pbstream_to_ros_map`, do you have problems with the `cartographer_occupancy_grid_node`?

> "void PublishOccupancyGrid()" just has declaration, but without realization. Did your team ignore the function or did i not find the realization?

`git blame` blames me :boom: I forgot to remove it while refactoring. It's not used anymore, the publisher logic is now completely in `DrawAndPublish()`. I opened a pull request to remove the declaration.

CccYunxiao, 2018-09-29:
@MichaelGrupp Thank you for your reply. Now the third question has been done. 

When i launch "move_base" node, "cartographer_node" node(which creates submaps) and "cartographer_occupancy_grid_node" node(which integrates submaps and public "map" topic),  and when i specify a goal, there is a path create by "move_base" node, but the path will across the obstacle.

When i launch "move_base" node and "slam_gmapping" node, the path will bypasses obstacle. 

This reason is the same to why do i use "map_server" node to save a map which does not contain obstacle occupation. In other words, the maps created by "gmapping" and "cartographer" respectively have different style data. 

So how can i do to create real-time cartographer map which has some features:

1. include obstacle occupation in grid map

2. this map can be correctly accept by ros navigation function packages

3. this map is dynamic , not created by "cartographer_pbstream_to_ros_map" node though read the pbstream file.

I hope you can enlighten me. Thank very much!

alliansmover, 2018-09-29:
> @MichaelGrupp感谢您的回复。现在第三个问题已经完成。
> 
> 当我启动“move_base”节点，“cartographer_node”节点（创建子图）和“cartographer_occupancy_grid_node”节点（它集成了子图和公共“地图”主题）时，当我指定目标时，有一个由“move_base”创建的路径节点，但路径将越过障碍物。
> 
> 当我启动“move_base”节点和“slam_gmapping”节点时，路径将绕过障碍物。
> 
> 这个原因与我为什么使用“map_server”节点保存不包含障碍物占用的地图相同。换句话说，由“gmapping”和“cartographer”创建的地图分别具有不同的样式数据。
> 
> 那么我该如何创建具有一些功能的实时制图师地图：
> 
> 1. 包括网格图中的障碍占用
> 2. 这个地图可以通过ros导航功能包正确接受
> 3. 此映射是动态的，不是由“cartographer_pbstream_to_ros_map”节点创建的，而是读取pbstream文件。
> 
> 我希望你能开导我。非常感谢！

I have the same problem

CccYunxiao, 2018-09-29:
@alliansmover Hi friend, do you want to use dynamic map created by cartographer and use move_base node to achieve dynamic navigation?

If you want to achieve this goal, you will meet a problem that the map created by cartographer can not be accept correctly by move_base node. Right?

alliansmover, 2018-10-08:
> @alliansmover嗨朋友，您想使用由制图师创建的动态地图并使用move_base节点来实现动态导航吗？
> 
> 如果要实现此目标，您将遇到一个问题，即由move_base节点无法正确接受由制图师创建的地图。对？

Yes, I have the same problem.Have you solved it?

alliansmover, 2018-10-08:
> @alliansmover Hi friend, do you want to use dynamic map created by cartographer and use move_base node to achieve dynamic navigation?
> 
> If you want to achieve this goal, you will meet a problem that the map created by cartographer can not be accept correctly by move_base node. Right?

Yes, I have the same problem.Have you solved it?

CccYunxiao, 2018-10-08:
@alliansmover yes, i think i had solved this problem. Now the move_base node can identity the obstacle publiced by cartographer_occupancy_grid_node; at the same time, the path created by planner can correctly avoid obstacles. If you want to have the same effects, i can give you some tips:

1. do you understand the word @MichaelGrupp said -- **the publisher logic is now completely in DrawAndPublish().**

1. do you know the map data (free, unknown, obstacle) publiced by cartographer and slam-gmapping? If you do not know, you can launch these nodes separately and use tool **rostopic** to public these maps.

alliansmover, 2018-10-09:
> @alliansmover yes, i think i had solved this problem. Now the move_base node can identity the obstacle publiced by cartographer_occupancy_grid_node; at the same time, the path created by planner can correctly avoid obstacles. If you want to have the same effects, i can give you some tips:
> 
> 1. do you understand the word @MichaelGrupp said -- **the publisher logic is now completely in DrawAndPublish().**
> 2. do you know the map data (free, unknown, obstacle) publiced by cartographer and slam-gmapping? If you do not know, you can launch these nodes separately and use tool **rostopic** to public these maps.

Thanks, I think I seem to know how to solve this problem.

CccYunxiao, 2018-10-09:
@alliansmover Good luck.

closed 2018-10-09 by CccYunxiao