cartographer-project/cartographer_ros #1032: How to save the map

carl1218, 2018-09-28:
Hi:
 I runed the 2d slam online,but when i call rosservice call /finish_trajectory "trajectory_id: 0"  ,i can not find the map .so how to save the map?

ZhouYixuanRobtic, 2018-09-28:
- If you wanna get a `.pgm`and `yaml` map. 
ROS provides map_server to do so.Try `rosrun map_server map_saver -f mapname` 
And another thing must be noted,
 > The map_saver output only contains free and unknown cells, occupied cells seem to be missing. This is probably because of the fact that other ROS integrated SLAM algorithms provide different values as the map data. For example GMapping only defines -1, 0 or 100 instead of the full range from from -1 to 100 like Cartographer does. 

  So you may need to do some manual changes, please refer to this #475.
- If you wanna get a `.pbstream` files for cartographer
 Please refer to [assets writer](https://github.com/googlecartographer/cartographer_ros/issues/963) or the offline node

Hope this would be helpful

carl1218, 2018-09-28:
@ZhouYixuanRobtic 
Thank you very much.

MichaelGrupp, 2018-09-28:
You can also use the `write_state` service to write a `.pbstream` file and then use the `cartographer_pbstream_to_ros_map` tool.

HaoQChen, 2019-01-14:
You can use map_server, but you need to change some code like this [map_server for Cartographer](https://github.com/HaoQChen/map_server)