cartographer-project/cartographer_ros #1033: How to set the new trajectory

carl1218, 2018-09-28:
Hi:
I am running 2d slam online,when i call  rosservice call /finish_trajectory 0 ,than set a new trajectory,
i call : 
` rosrun cartographer_ros cartographer_start_trajectory -configuration_basename=backpack_2d.lua -configuration_directory=/home/calyu/catkin_ws/install_isolated/share/cartographer_ros/configuration_files -initial_pose="{to_trajectory_id=0,relative_pose={translation={0.0,0.0,10},rotation={0.,0.,0.0}}}`

I set the Z-axis to 10m，I wan to drafting the map in this plane.
At first, i can see the cloud points in this plane(Z-axis = 10m).but when start to  drawing the map , the Z axis will change to the initial position.(Z-axis = 0m).
How can i to set the new trajectory in the new plane (Z-axis ≠ 0)

this is cloud points picture:
[image]

alfred31, 2018-10-03:
From my point of view two possible reasons:
1) It maybe caused due to the loop closure of cartographer, the algorithm detects your position in the map, that is Z=0 and then it corrects the base_link pose within the map
2) Since it is a 2D problem cartographer might not allow Z axis different from 0

Regards

carl1218, 2018-10-31:
@alfred31 
Thank you for your reply.Is there any way to change  it?
Now,i can drafting the map in this plane(Z-axis ≠ 0),but
when i runned :rosservice call /finish_trajectory 0 want to save this map ,the Z axis will change to the initial position.look like this:
[image]
Than runned finish trajectory:
[image]

MichaelGrupp, 2020-10-21:
This is not really a supported use case of 2D SLAM.

closed 2020-10-21 by MichaelGrupp