cartographer-project/cartographer_ros #1036: When I run Demo_pr2.launch, the map frame inside the TF tree is not, Why ?

Daivesi, 2018-09-30:
i have provide /scan data and /odom data. so my configfile like this:

demo_pr2.launch:

<launch>
  <param name="/use_sim_time" value="false" />

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory
              $(find cartographer_ros)/configuration_files
          -configuration_basename pr2.lua"
      output="screen">
    <remap from="scan" to="/scan" />
  </node>

  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />

  <node name="tf_remove_frames" pkg="cartographer_ros"
      type="tf_remove_frames.py">
    <remap from="tf_out" to="/tf" />
    <rosparam param="remove_frames">
      - map
      - odom_combined
    </rosparam>
  </node>

</launch>
---------------------------------------------------------------------------------------------

pr2.lua；
    include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_footprint",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,  
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.15
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.angular_search_window = math.rad(35.)

POSE_GRAPH.optimization_problem.huber_scale = 1e2

return options

than I run " rosrun rqt_tf_tree rqt_tf_tree", the resulat like this:

                                                              odom               
                                                                  include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_footprint",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = true,
  publish_frame_projected_to_2d = false,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 1,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,  
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

MAP_BUILDER.use_trajectory_builder_2d = true

TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = true
TRAJECTORY_BUILDER_2D.use_imu_data = false
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.15
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.angular_search_window = math.rad(35.)

POSE_GRAPH.optimization_problem.huber_scale = 1e2

return options

-------------------------------------------------------------------------------------------
Than i run "rosrun rqt_tf_tree rqt_tf_tree", the resualt like this:

                                                          odom
                                                              ↓    
                                                    base_footprint
                                                              ↓
                                                       base_link
                                                              ↓
                                                            laser

the map have disappear。 It was still there at first, but it disappeared after a while。
the “/map” topic still have data.

Daivesi, 2018-09-30:
demo_pr2.launch:
  <param name="/use_sim_time" value="false" />

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory
              $(find cartographer_ros)/configuration_files
          -configuration_basename pr2.lua"
      output="screen">
    <remap from="scan" to="/scan" />
  </node>

  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />

  <node name="tf_remove_frames" pkg="cartographer_ros"
      type="tf_remove_frames.py">
    <remap from="tf_out" to="/tf" />
    <rosparam param="remove_frames">
      - map
      - odom_combined
    </rosparam>
  </node>

Daivesi, 2018-09-30:
demo_pr2.launch:
-  <param name="/use_sim_time" value="false" />
-
 -<node name="cartographer_node" pkg="cartographer_ros"
  -    type="cartographer_node" args="
   -       -configuration_directory
    -          $(find cartographer_ros)/configuration_files
          -configuration_basename pr2.lua"
    - output="screen">
  -  <remap from="scan" to="/scan" />
-  </node>

 - <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
 -     type="cartographer_occupancy_grid_node" args="-resolution 0.05" />

 - <node name="tf_remove_frames" pkg="cartographer_ros"
  -    type="tf_remove_frames.py">
 -   <remap from="tf_out" to="/tf" />
 -   <rosparam param="remove_frames">
      - map
      - odom_combined
 -   </rosparam>
-  </node>
-

Daivesi, 2018-09-30:
/home/anbot/Pictures/1.png
[image]:

while .........

Daivesi, 2018-09-30:
[image]

MichaelGrupp, 2020-11-06:
Closing for inactivity, we can't invest time to help you with your setup anymore unfortunately.

closed 2020-11-06 by MichaelGrupp