cartographer-project/cartographer_ros #1038: Not publishing map

Semih67, 2018-10-01:
I'm running google cartographer but it doesn't publish the map. I'm working with a laser scanner (rplidar). 

The rqt_graph looks like this: 
[image]

launch: 
[image]

urdf: 
[image]

rqt: 
[image]

map: 
[image]

lua: 
[image]

trajectory_2d: 
[image]

I don't know what exactly my problem is. I have these assumptions: 

-cartographer is not receiving any sensor data
-I don't know if I need a tf_listener. I didn't transform the sensor data from base_laser to base_link

gaschler, 2018-10-01:
Please check the message type that your laser scanner publishes.
When you set num_laser_scans = 1, cartographer_ros subscribes to type "LaserScan" messages on the topic "scan".

Semih67, 2018-10-02:
the topic "scan" has the message type "LaserScan"

Semih67, 2018-10-04:
Ok, it's working now. After I inserted a world and map frame it worked.

closed 2018-10-04 by Semih67

zchao9456, 2018-11-08:
Can you tell me how you did it? My question is the same as yours.

zchao9456, 2018-11-08:
> Ok, it's working now. After I inserted a world and map frame it worked.

Can you tell me how you did it? My question is the same as yours.