cartographer-project/cartographer_ros #1050: Failing debian package build because of Abseil

AlexReimann, 2018-10-08:
I'm trying to build an updated cartographer_ros debian package (commit c0e713c404d3087cec3dbe07e35bd162ce802dc0 from  Wed Aug 29 12:21:47 2018)  for internal use as following:

```
cd ros_cartographer
bloom-generate rosdebian --os-name ubuntu --os-version xenial --ros-distro kinetic
fakeroot debian/rules binary
```

The `cartographer_ros_msgs` and `cartographer` (commit a351a8e from Wed Sep 5 16:59:16 2018) debian packages build fine.
Building with catkin works fine (even building install).

Errors:
```yaml
[ 25%] Building CXX object CMakeFiles/cartographer_ros.dir/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc.o
/usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"cartographer_ros\" -isystem /usr/include/lua5.2 -isystem /usr/include/pcl-1.7 -isystem /usr/include/eigen3 -isystem /opt/ros/kinetic/include -isystem /opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu -I/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros -I/usr/src/gmock/gtest/include -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/freetype2 -isystem /usr/include/libpng12 -isystem /opt/protobuf3/include -isystem /opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil  -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2     -DEIGEN_USE_NEW_STDVECTOR -DEIGEN_YES_I_KNOW_SPARSE_MODULE_IS_NOT_STABLE_YET -pthread -std=c++11 -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -o CMakeFiles/cartographer_ros.dir/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc.o -c /opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:111:1: error: ‘MutexImpl’ does not name a type
 MutexImpl::MutexImpl() {
 ^
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:116:1: error: ‘MutexImpl’ does not name a type
 MutexImpl::~MutexImpl() {
 ^
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:123:6: error: ‘MutexImpl’ has not been declared
 void MutexImpl::Lock() {
      ^
In file included from /opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/base/port.h:24:0,
                 from /opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/base/internal/low_level_alloc.h:51,
                 from /opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/mutex.h:65,
                 from /opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:19:
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc: In function ‘void absl::synchronization_internal::Lock()’:
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:124:38: error: ‘pthread_mutex_’ was not declared in this scope
   ABSL_RAW_CHECK(pthread_mutex_lock(&pthread_mutex_) == 0, "pthread error");
                                      ^
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:125:3: error: ‘locked_’ was not declared in this scope
   locked_ = true;
   ^
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc: At global scope:
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:128:6: error: ‘MutexImpl’ has not been declared
 bool MutexImpl::TryLock() {
      ^
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc: In function ‘bool absl::synchronization_internal::TryLock()’:
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:129:46: error: ‘pthread_mutex_’ was not declared in this scope
   bool locked = (0 == pthread_mutex_trylock(&pthread_mutex_));
                                              ^
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:130:15: error: ‘locked_’ was not declared in this scope
   if (locked) locked_ = true;
               ^
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc: At global scope:
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:134:6: error: ‘MutexImpl’ has not been declared
 void MutexImpl::Unlock() {
      ^
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc: In function ‘void absl::synchronization_internal::Unlock()’:
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:135:3: error: ‘locked_’ was not declared in this scope
   locked_ = false;
   ^
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:136:3: error: ‘released_’ was not declared in this scope
   released_.SignalAll();
   ^
In file included from /opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/base/port.h:24:0,
                 from /opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/base/internal/low_level_alloc.h:51,
                 from /opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/mutex.h:65,
                 from /opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:19:
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:137:40: error: ‘pthread_mutex_’ was not declared in this scope
   ABSL_RAW_CHECK(pthread_mutex_unlock(&pthread_mutex_) == 0, "pthread error");
                                        ^
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc: At global scope:
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:140:1: error: ‘CondVarImpl’ does not name a type
 CondVarImpl::CondVarImpl() {
 ^
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:145:1: error: ‘CondVarImpl’ does not name a type
 CondVarImpl::~CondVarImpl() {
[... 41 lines cut ...]
            ^
/opt/enway/cartographer_ws/src/cartographer_ros/cartographer_ros/obj-x86_64-linux-gnu/abseil/src/abseil/absl/synchronization/internal/mutex_nonprod.cc:46:12: warning: ‘absl::Time absl::synchronization_internal::{anonymous}::LimitedDeadline(absl::Time)’ defined but not used [-Wunused-function]
 absl::Time LimitedDeadline(absl::Time deadline) {
            ^
[... cut ...]
Shouldn't the `Abseil` build from the `cartographer` package be used for the ros package? (then I wouldn't run into these problems)

We got it to compile once, but had to manually install `https://github.com/google/benchmark` and also add the `${ABSEIL_INCLUDE_DIRS}`, add `ABSEIL` to the catkin package and then it worked with some weird combination of running `fakeroot debian/rules binary` then `fakeroot debian/rules` and then `fakeroot debian/rules binary` which points to some missing dependencies or compile race conditions...

pifon2a, 2018-10-09:
Interesting. Why did you need google/benchmark? As soon as https://github.com/abseil/abseil-cpp/pull/182 is submitted, it will be possible to `make install` abseil and reuse it in cartographer/cartographer_ros.

AlexReimann, 2018-10-09:
It is used in some of the Abseil tests, which also got build at some point.

redkite, 2018-10-30:
I found the issue, the CMakeLists.txt for `cartographer_ros` includes all files with the ending `*.cc`.

If you build the package with the fakeroot command separately to generate a .deb package the `file(GLOB_RECURSE ALL_SRCS "*.cc" "*.h")` statement will pull in not only `cartographer_ros` files, but also all the Abseil files. Since all `cartographer_ros` files are located in a sub-folder with the same name, it's sufficient to traverse only this sub-folder, such as: `file(GLOB_RECURSE ALL_SRCS "cartographer_ros/*.cc" "cartographer_ros/*.h")`.

The same applies for similar statements. If you want I can submit a PR with changes to the CMakeLists.txt so that it builds also with bloom-generate/fakeroot.

MichaelGrupp, 2018-10-31:
@redkite Yes, a PR for restricting the GLOB pattern only to the required folders would be good.

PR #1085 (2018-11-06): Only include correct source files in cmake
This fixes #1050. Tested standalone compilation and with the debian fakeroot tool. I had to build with a custom Protobuf3 instance though, so another build-check on a regular setup would be appreciated.

Before this change all *.cc files would be included. If subprojects
were run individually with the Debian package generator. This resulted in an
inclusion of temporarily checked out *.cc from the abseil include. This
change fixes the import behaviour and enables the creation of a valid
package.
commit c4933695 (2018-11-06): Only include correct source files in cmake (#1085)

This fixes #1050. Tested standalone compilation and with the debian fakeroot tool. I had to build with a custom Protobuf3 instance though, so another build-check on a regular setup would be appreciated.

Before this change all *.cc files would be included. If subprojects
were run individually with the Debian package generator. This resulted in an
inclusion of temporarily checked out *.cc from the abseil include. This
change fixes the import behaviour and enables the creation of a valid
package.
closed 2018-11-06 by wally-the-cartographer