cartographer-project/cartographer_ros #1051: [cartographer_node-2] process has died

alliansmover, 2018-10-09:
When I use the cartographer to get the map, the cartographer will get the following errors:
can you tell me why?
thanks.
[FATAL] [1539065013.815399731, 19.711000000]: F1009 14:03:33.000000 24461 map_by_time.h:43] Check failed: data.time > std::prev(trajectory.end())->first (621355968195650000 vs. 621355968195650000) 
*** Check failure stack trace: ***
    @     0x7f45bd895daa  (unknown)
    @     0x7f45bd895ce4  (unknown)
    @     0x7f45bd8956e6  (unknown)
    @     0x7f45bd898687  (unknown)
    @           0x66adee  cartographer::mapping::optimization::OptimizationProblem2D::AddOdometryData()
    @           0x6ac79b  cartographer::mapping::PoseGraph2D::AddOdometryData()
    @           0x667a1c  cartographer::mapping::(anonymous namespace)::GlobalTrajectoryBuilder<>::AddSensorData()
    @           0x663ce8  cartographer::mapping::CollatedTrajectoryBuilder::HandleCollatedSensorData()
    @           0x663e51  _ZNSt17_Function_handlerIFvRKSsSt10unique_ptrIN12cartographer6sensor4DataESt14default_deleteIS5_EEEZNS3_7mapping25CollatedTrajectoryBuilderC1EPNS4_17CollatorInterfaceEiRKSt3setINSA_26TrajectoryBuilderInterface8SensorIdESt4lessISG_ESaISG_EES2_ISF_S6_ISF_EEEUlS1_S8_E_E9_M_invokeERKSt9_Any_dataS1_S8_
    @           0x605ecb  _ZNSt17_Function_handlerIFvSt10unique_ptrIN12cartographer6sensor4DataESt14default_deleteIS3_EEEZNS2_8Collator13AddTrajectoryEiRKSt13unordered_setISsSt4hashISsESt8equal_toISsESaISsEERKSt8functionIFvRKSsS6_EEEUlS6_E_E9_M_invokeERKSt9_Any_dataS6_
    @           0x604843  std::function<>::operator()()
    @           0x603d42  cartographer::sensor::OrderedMultiQueue::Dispatch()
    @           0x60432b  cartographer::sensor::OrderedMultiQueue::Add()
    @           0x606038  cartographer::sensor::Collator::AddSensorData()
    @           0x6631bd  cartographer::mapping::CollatedTrajectoryBuilder::AddData()
    @           0x664961  cartographer::mapping::CollatedTrajectoryBuilder::AddSensorData()
    @           0x5ca2c4  cartographer_ros::SensorBridge::HandleOdometryMessage()
    @           0x59a93f  cartographer_ros::Node::HandleOdometryMessage()
    @           0x5a8d4c  boost::detail::function::void_function_obj_invoker1<>::invoke()
    @           0x5b6ef5  ros::SubscriptionCallbackHelperT<>::call()
    @     0x7f45bb0b1645  (unknown)
    @     0x7f45bb06b107  (unknown)
    @     0x7f45bb06bc53  (unknown)
    @     0x7f45bb0b4175  (unknown)
    @     0x7f45bb09bd9b  (unknown)
    @           0x596f4b  cartographer_ros::(anonymous namespace)::Run()
    @           0x5940f6  main
    @     0x7f45b9b64f45  (unknown)
    @           0x596cf7  (unknown)
    @              (nil)  (unknown)
[cartographer_node-2] process has died [pid 24461, exit code -6, cmd /home/zj/google/install_isolated/lib/cartographer_ros/cartographer_node -configuration_directory /home/zj/google/install_isolated/share/cartographer_ros/configuration_files -configuration_basename revo_lds.lua scan:=front/scan __name:=cartographer_node __log:=/home/zj/.ros/log/e5cf24c4-cb88-11e8-b2f2-d46a6aa4b3a3/cartographer_node-2.log].
log file: /home/zj/.ros/log/e5cf24c4-cb88-11e8-b2f2-d46a6aa4b3a3/cartographer_node-2*.log

alliansmover, 2018-10-09:
> 看起来您正在为您的测距发布无序的ROS消息。请注意，制图师需要在每个传感器数据流中严格按时间顺序（例如，测距，IMU，激光雷达），但允许不同流之间的任意时间顺序。

alliansmover, 2018-10-09:
> It looks like you are posting out-of-order ROS messages for your odometry. Note that Cartographer requires strict time-order within each sensor data stream (e.g. odometry, IMU, lidar) but allows arbitrary time-order between different streams.

hello friend,I set <param name="/use_sim_time" value="true" />.Is it causing this problem?thanks

closed 2020-10-21 by MichaelGrupp

numb0824, 2021-04-16:
Hi，
i have meet the same problems,could you please tell me how to solve it ?
thanks