cartographer-project/cartographer_ros #1052: Cartographer 3D with IMU, 1 Velodyne-16 LiDAR. No map output

NTUwilliam, 2018-10-09:
Hi all !
I'm trying to use my **own 3D bagfile** (collected by Husky Robot, using only one Velodyne-16 LiDAR and one 3DM-GX3-25 IMU) to run on Cartographer but no map generated. 
Before this, I've successfully run Cartographer on my own 2D bagfile.

My bag file has 2 topics and here is what **rosbag info** gives me:
[image]

And the result of running the **rosbag_validate** gives me:
[image]

I had tuned the parameters and tried to run this bag by `roslaunch cartographer_ros  offline_backpack_3d.launch bag_filename:=/home/williamwoo/husky-tf-scan-_2018-10-08-08-14-41.bag` but there's no map showing on the Rviz. 
The running results (some repetitive part ignored) and Rviz screenshots are shown as below:
[image]

[image]

[image]

[image]

[image]

[image]

The related files are uploaded for your reference:
3D Bagfile with my sensor data: https://entuedu-my.sharepoint.com/:u:/g/personal/zhenyu002_e_ntu_edu_sg/EbOR1SYYAeFBk5bVflIcrvsBYmshHWGSTDlUIxx8LrG4Cw?e=78UbLr
Launch file: https://github.com/NTUwilliam/Google_cartographer/blob/master/offline_backpack_3d.launch
lua file: https://github.com/NTUwilliam/Google_cartographer/blob/master/backpack_3d.lua
URDF file: https://github.com/NTUwilliam/Google_cartographer/blob/master/backpack_3d.urdf

 
 Is there any solution to this problem?  Any comment is appreciated. Thanks a lot.  @gaschler @cschuet 
@SirVer @ojura

ojura, 2018-10-15:
@NTUwilliam, your bag has a single `PointCloud2` per revolution, with a sample rate of 10 Hz.

Cartographer's demo bags consist of small point clouds which correspond to individual Velodyne UDP packets. There is around 100-200 of these per revolution. Therefore, the following parameter in the .lua file is set to accumulate 160 point clouds into one scan: 

`TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 160`

This is wrong for your setup, since in your data, 1 cloud = 1 revolution. Therefore, this parameter should be set to 1 or 2 for your case.

@cschuet I've seen this issue occur many times. Most people are using the open source ROS velodyne driver, which accumulates a whole revolution into one point cloud. They use the default provided tuning and forget to change this. Perhaps we should document this in the .lua.

junzhang2016, 2018-10-18:
@ojura Yes, you are right!

I got it worked and the generated map in the Rviz is:
For `TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 0.1`
[image]

For `TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 1`
[image]

For `TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 3`
[image]

I've noticed that the generated map is still kind of **unclear** and the **ROS time runs pretty fast** (running finished in around 10 seconds, although the actual duration time is 136s). Any suggestions for tuning the parameters? 
@ojura @cschuet Thanks!!

NTUwilliam, 2018-10-18:
@ojura Yes, you are right!

I got it worked and the generated map in the Rviz is:
For `TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 0.1`
[image]

For `TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 1`
[image]

For `TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 3`
[image]

I've noticed that the generated map is still kind of **unclear** and the **ROS time runs pretty fast** (running finished in around 10 seconds, although the actual duration time is 136s). Any suggestions for tuning the parameters? 
@ojura @cschuet Thanks!!

NTUwilliam, 2018-10-18:
@ojura Thanks for your speedy reply.
I see. Then I think `TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 1` works great for my case. 

I already chose to display high-resolution submaps but it's still not clear, as shown below:
[image]

Duke-Allen, 2019-02-21:
@NTUwilliam I'm running cartographer with IMU and Rslidar-16 LiDAR, but no map output. There is a warning:"No tf data, Actual error: [map] does not exist". Could you help me?

2505381597, 2019-02-25:
Hello, I have seen your result, but I got a question about it. You have built a 2D map, but do you know how to build a 3D map using one velodyne-16 LIDAR ?

simongtan, 2019-02-25:
@NTUwilliam
> I see. Then I think `TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 1` works great for my case.
> 
> I already chose to display high-resolution submaps but it's still not clear, as shown below:

Hi William,
Try using demo_my_robot.launch instead of offline_my_robot.launch in your launch file.

(From cartographer documentation) "offline_my_robot.launch is very similar to demo_my_robot.launch but tries to execute SLAM as fast as possible. This can make map building significantly faster. This launch file can also use multiple bag files provided to the bag_filenames argument."

NTUwilliam, 2019-03-04:
@2505381597 
I tried to build a 3D map but failed. So that's why I got the 2D map.....
You may search for some other issues about how to build a 3D map using one velodyne-16 LIDAR (I think they got a lot of similar issues)

NTUwilliam, 2019-03-04:
@simongtan 
Thanks for replying! I'll give it a try in the future.

NTUwilliam, 2019-03-04:
@Duke-Allen Hi
Sorry that I browsed through your issue but have no idea how to solve it. 
Maybe you can post a new issue and @ the contributors who will be willing to solve your issue.

Duke-Allen, 2019-03-04:
@NTUwilliam 
Do you build 3D map successfully? I am trying build 3D map by LIDAR and IMU, but I always have some problems I don't know how to solve.

NTUwilliam, 2019-03-04:
@Duke-Allen No I didn't :( 
So it's better for you to post a new issue and ask for help.

Duke-Allen, 2019-03-04:
@NTUwilliam 
Thanks for your reply. I will try it. Wish me good luck.

Akash-Pardasani, 2019-10-19:
@Duke-Allen Were you able to get the 3D mapping work?

Duke-Allen, 2019-10-19:
@Akash-Pardasani Yeah, I just built a 3D map. But I did not check to see if the map was correct.

Akash-Pardasani, 2019-10-19:
@Duke-Allen Thanks for the speedy reply. It would be really great if you could share the final configuration that worked for you in Google Cartographer?

Duke-Allen, 2019-10-19:
@Akash-Pardasani This is my configuration. [rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
I hope it could help you.

closed 2020-10-21 by MichaelGrupp