cartographer-project/cartographer_ros #1055: z position drifting.

Howonjung, 2018-10-10:
Hello,

I am currently testing cartographer package to make a outdoor map and I found that there is still z

position drifting even after I calibrate my extrinsic IMU orientation from [This_issue](https://github.com/googlecartographer/cartographer_ros/issues/1029)

The result of optimization is just from **2 to 3.xx deg**.(IMU correction value) as you can check the pic below
[image]

from [This_issue](https://github.com/googlecartographer/cartographer/issues/1199#issuecomment-398080355) 

"**When the IMU/lidar rotation is correct +/- a few degrees, global optimization will do the rest for you, see googlecartographer/cartographer_ros#902 (comment)**"

so, I thought global optimization will handle this problem but it's not.
I really don't know what is reason of this z position drifting problem if calibration is fine from the result above.

any comments would be appreciated.

 ==============================================================================

**The result of rosbag_validation** 
[image]

The test environment is same as in [This_issue](https://github.com/googlecartographer/cartographer_ros/issues/1029)
bag file : [bag_file](https://drive.google.com/file/d/1V-qSMat-CESRfDInS-d2K54YOXndglLB/view?usp=sharing)
launch file : [launch_file](https://drive.google.com/file/d/1hfwX5Ziqo3Okr9jV8gqUe0hkQXc0sTwI/view?usp=sharing)
urdf file : [urdf_file](https://drive.google.com/file/d/1czchSWzM3BXPK7KXEV6VtkKgebyp46hG/view?usp=sharing)
lua file : [lua_file](https://drive.google.com/file/d/1Tbdp3POuurGswb6OLJBy-koNU4iyJOjS/view?usp=sharing)
map builder lua file : [map_builder_lua](https://drive.google.com/file/d/1nBotxuXiBlZaVZ1Skckqjcx5Gwl4bBvp/view?usp=sharing)
trajectory_builder_3d lua file : [trajectory_builder_3d](https://drive.google.com/file/d/1LNQsOH1E07h6c2Tj9qGO5iQqeiCZDLKZ/view?usp=sharing)

ojura, 2018-10-10:
Hi,

if you want to use the IMU correction quaternion without changing your URDF, check out the following pull request:

https://github.com/googlecartographer/cartographer_ros/pull/1025

You can just copy the correction quaternion as printed in the report into your .lua file.

Howonjung, 2018-10-11:
@ojura  Thanks for replying. 

so, based on what you recommend to do, I modified Cartographer_ros source code following [this_link](https://github.com/googlecartographer/cartographer_ros/pull/1025/commits/b1ba4b2fbf892ec17b06ef819afe0e8723618ad2) , build it successfully.

so, what I am planning to do is that, just following the small area to make a loop closure so that i can get a proper IMU correction quaternion.
 but, before that I have a one question. 
 
from  [this_link](https://github.com/googlecartographer/cartographer_ros/pull/1025/commits/b1ba4b2fbf892ec17b06ef819afe0e8723618ad2) 
there is part " **CHECK_NEAR(options.imu_correction.norm(), 1., 1e-9);** " 
it looks like options.imu_correction.norm() value should be between (1 -  1e-9) and  (1 + 1e-9).
and I think this condition looks very tight even for IMU correction quaternion of less than 1 deg.

so can i modify this value ? like 1e-9 -> 1e-5 ?

ojura, 2018-10-12:
@Howonjung you were right, 1e-9 for the normalization check is a bit too strict. I lowered this to 1e-4.

Howonjung, 2018-11-05:
@ojura  I have several questions regrading this Issue !

First one is that 
**"Is ceres matching or online_correlative_scan_matching only the source of Pose information If i only use lidar and IMU ( No external Odometry Info ) ?"**
(I just want to make sure about this... since I thought IMU gives initial Pose value for ceres-matching but looks like only for rotational components..)
 
I suspect that the reason of z drifting is coming from IMU data (Gravity data). the bag data shows that the vehicle is going through the uphills even thought it is almost plain load (just small pot hole or bump). 
I try to minimize this effect by increasing the value "imu_gravity_time_constant" (moving average related value) but it doesn't affect much (It works fine on the plain road.)
So, the second question is that 
**"Is there any other way I can try to solve this problem ?"**

ojura, 2018-11-05:
@Howonjung the problem is in your IMU-lidar transform. Incorrect pitch of the IMU causes SLAM to think you are going uphills.

Your options are:
 - correct the IMU transform, by changing your urdf or using #1025 
 - enable the option for fixing z=0 in 3D SLAM

Flandoe, 2018-12-03:
@ojura You said ‘enable the option for fixing z=0 in 3D SLAM’, but I have no idea how to enable the option. Adding fixing_z = 0 in the options{} is wrong. Could you point out how to enable it ? Thanks.

aidamQin, 2019-07-03:
> https://github.com/googlecartographer/cartographer/blob/a33ef5de86306380663ecb476d187c71e35d8370/configuration_files/pose_graph.lua#L76
> 
> 因此，您将添加`POSE_GRAPH.optimization_problem.fix_z_in_3d = true`到您的调整。

I would like to know what is the  use_online_imu_extrinsics_in_3d  = true, is it the same as above?

aidamQin, 2019-07-03:
> @Howonjung问题在于你的IMU-lidar变换。IMU的音高不正确会导致SLAM认为你正在上升。
> 
> 你的选择是：
> 
> * 通过更改您的urdf或使用[＃1025](https://github.com/googlecartographer/cartographer_ros/pull/1025)来更正IMU转换[](https://github.com/googlecartographer/cartographer_ros/pull/1025)
> * 启用在3D SLAM中修复z = 0的选项

I would like to know what is the `use_online_imu_extrinsics_in_3d = true` do ,is it the same as above?

MichaelGrupp, 2020-10-21:
> > @Howonjung问题在于你的IMU-lidar变换。IMU的音高不正确会导致SLAM认为你正在上升。
> > 你的选择是：
> > 
> > * 通过更改您的urdf或使用[＃1025](https://github.com/googlecartographer/cartographer_ros/pull/1025)来更正IMU转换
> > * 启用在3D SLAM中修复z = 0的选项
> 
> I would like to know what is the `use_online_imu_extrinsics_in_3d = true` do ,is it the same as above?

It means that the IMU extrinsic calibration is also optimized in the optimization.

closed 2020-10-21 by MichaelGrupp