cartographer-project/cartographer_ros #1056: Lookup would require extrapolation into the past. when trying 3D Cartographer with and IMU/odometory/LIDAR

SarahMars, 2018-10-10:
Hello, and thanks in advance.
I am trying to do 3D cartographer using an IMU, odometry, and a single Lidar but I get warning messages and never able to get a proper map. (I had no problem with 2D cartographer.(I converted pointcloud2 data to LaserScan for 2D cartographer.))
The warning message (copied and pasted from terminal) is [here](https://github.com/SarahMars/cartographer_configuration/blob/master/error_message)

Here are the links for (everything is in this post but in case if links were convenient)
[lua file](https://github.com/SarahMars/cartographer_configuration/blob/master/config/backpack_3d.lua), 
[launch file for cartographer & cartographer_node](https://github.com/SarahMars/cartographer_configuration/blob/master/launch/pointcloud_cartographer.launch),
[launch file for robot_state_publisher](https://github.com/SarahMars/cartographer_configuration/blob/master/launch/myrobot_frames.launch),
[xml configuration file for the robot](https://github.com/SarahMars/cartographer_configuration/blob/master/models/myrobot.xml),
and [screen shot of rqt_tf_tree](https://github.com/SarahMars/cartographer_configuration/blob/master/screen_shots/tf_tree_without_map.png). (same as the following picture)
[image] 

Once every 30~50 seconds, warning message stops for few seconds and provides [this map](https://github.com/SarahMars/cartographer_configuration/blob/master/screen_shots/map_not_working.png) data. [Tf_tree looks like this](https://github.com/SarahMars/cartographer_configuration/blob/master/screen_shots/tf_tree_with_map.png) at this time.

I've already tried /checked these;
- waitForTransform in track_odometry(the broadcaster of odom frame) `ros::Time(0)`, `ros::Time::now()`, and tried `ros::Duration(1.0)` or even longer duration time.
- syncing the clock of my laptop and the robot.
- observed `$ top` result and it is not the issue caused by CPU
- changed lookup_transform_timeout_sec to 1.0 and 10.0

Although it kept giving me the same error.

I will appreciate any kind of advices.
Thank you, 

Sarah 

# lua file

```
include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "imu_link",
  published_frame = "odom",
  odom_frame = "odom_cg",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = false,
  use_odometry = true,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 0,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 1,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 160

MAP_BUILDER.use_trajectory_builder_3d = true
MAP_BUILDER.num_background_threads = 7
POSE_GRAPH.optimization_problem.huber_scale = 5e2
POSE_GRAPH.optimize_every_n_nodes = 320
POSE_GRAPH.constraint_builder.sampling_ratio = 0.03
POSE_GRAPH.optimization_problem.ceres_solver_options.max_num_iterations = 10
POSE_GRAPH.constraint_builder.min_score = 0.62
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.66
[... 99 lines cut ...]
[ INFO] [1539165775.071520880]: I1010 19:02:55.000000  6315 configuration_file_resolver.cc:41] Found '/home/mars/catkin_ws/src/slam_study/configs/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1539165775.071829364]: I1010 19:02:55.000000  6315 configuration_file_resolver.cc:41] Found '/home/mars/catkin_ws/src/slam_study/configs/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1539165775.071871880]: I1010 19:02:55.000000  6315 configuration_file_resolver.cc:41] Found '/home/mars/catkin_ws/src/slam_study/configs/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1539165775.264291450]: I1010 19:02:55.000000  6315 submap_3d.cc:321] Added submap 1
[ INFO] [1539165775.264821790]: I1010 19:02:55.000000  6315 map_builder_bridge.cc:130] Added trajectory with ID '0'.
[ INFO] [1539165775.844728931]: I1010 19:02:55.000000  6315 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '636747625757570701'.
[ INFO] [1539165776.145594557]: I1010 19:02:56.000000  6315 local_trajectory_builder_3d.cc:86] IMU not yet initialized.
[ INFO] [1539165790.451548494]: I1010 19:03:10.000000  6315 collated_trajectory_builder.cc:72] imu rate: 49.44 Hz 2.02e-02 s +/- 1.00e-04 s (pulsed at 92.60% real time)
[ INFO] [1539165790.451671028]: I1010 19:03:10.000000  6315 collated_trajectory_builder.cc:72] odom rate: 40.00 Hz 2.50e-02 s +/- 7.49e-04 s (pulsed at 91.23% real time)
[ INFO] [1539165790.451788181]: I1010 19:03:10.000000  6315 collated_trajectory_builder.cc:72] points2 rate: 4.35 Hz 2.30e-01 s +/- 1.09e-01 s (pulsed at 93.03% real time)
[ INFO] [1539165805.545608982]: I1010 19:03:25.000000  6315 collated_trajectory_builder.cc:72] imu rate: 49.44 Hz 2.02e-02 s +/- 1.20e-04 s (pulsed at 100.42% real time)
[ INFO] [1539165805.545797143]: I1010 19:03:25.000000  6315 collated_trajectory_builder.cc:72] odom rate: 40.01 Hz 2.50e-02 s +/- 9.20e-04 s (pulsed at 99.69% real time)
[ INFO] [1539165805.545921336]: I1010 19:03:25.000000  6315 collated_trajectory_builder.cc:72] points2 rate: 3.84 Hz 2.60e-01 s +/- 9.33e-02 s (pulsed at 99.89% real time)
[ INFO] [1539165820.715214538]: I1010 19:03:40.000000  6315 collated_trajectory_builder.cc:72] imu rate: 49.45 Hz 2.02e-02 s +/- 8.97e-05 s (pulsed at 101.63% real time)
[ INFO] [1539165820.715337548]: I1010 19:03:40.000000  6315 collated_trajectory_builder.cc:72] odom rate: 40.00 Hz 2.50e-02 s +/- 8.51e-04 s (pulsed at 102.50% real time)
[ INFO] [1539165820.715433502]: I1010 19:03:40.000000  6315 collated_trajectory_builder.cc:72] points2 rate: 3.37 Hz 2.97e-01 s +/- 1.79e-01 s (pulsed at 101.19% real time)
[ WARN] [1539165827.266965214]: W1010 19:03:47.000000  6315 tf_bridge.cc:52] Lookup would require extrapolation into the past.  Requested time 1539165817.346780000 but the earliest data is at time 1539165817.383995962, when looking up transform from frame [odom] to frame [imu_link]
[ WARN] [1539165827.267340201]: W1010 19:03:47.000000  6315 tf_bridge.cc:52] Lookup would require extrapolation into the past.  Requested time 1539165817.346780000 but the earliest data is at time 1539165817.383995962, when looking up transform from frame [odom] to frame [imu_link]
[ WARN] [1539165827.269819190]: W1010 19:03:47.000000  6315 tf_bridge.cc:52] Lookup would require extrapolation into the past.  Requested time 1539165817.346780000 but the earliest data is at time 1539165817.383995962, when looking up transform from frame [odom] to frame [imu_link]
[ WARN] [1539165827.274754261]: W1010 19:03:47.000000  6315 tf_bridge.cc:52] Lookup would require extrapolation into the past.  Requested time 1539165817.346780000 but the earliest data is at time 1539165817.383995962, when looking up transform from frame [odom] to frame [imu_link]
[ WARN] [1539165827.280080282]: W1010 19:03:47.000000  6315 tf_bridge.cc:52] Lookup would require extrapolation into the past.  Requested time 1539165817.346780000 but the earliest data is at time 1539165817.383995962, when looking up transform from frame [odom] to frame [imu_link]
[ WARN] [1539165827.284358848]: W1010 19:03:47.000000  6315 tf_bridge.cc:52] Lookup would require extrapolation into the past.  Requested time 1539165817.346780000 but the earliest data is at time 1539165817.383995962, when looking up transform from frame [odom] to frame [imu_link]
[ WARN] [1539165827.289345815]: W1010 19:03:47.000000  6315 tf_bridge.cc:52] Lookup would require extrapolation into the past.  Requested time 1539165817.346780000 but the earliest data is at time 1539165817.383995962, when looking up transform from frame [odom] to frame [imu_link]
[ WARN] [1539165827.295566834]: W1010 19:03:47.000000  6315 tf_bridge.cc:52] Lookup would require extrapolation into the past.  Requested time 1539165817.346780000 but the earliest data is at time 1539165817.434059095, when looking up transform from frame [odom] to frame [imu_link]
```

SarahMars, 2018-10-18:
Any additional information needed?

gaschler, 2018-10-18:
> (I converted pointcloud2 data to LaserScan for 2D cartographer.)

You don't need to do this conversion. 2D Cartographer accepts PointCloud2 messages as an input.

> [ WARN] [1539165827.295566834]: W1010 19:03:47.000000  6315 tf_bridge.cc:52] Lookup would require extrapolation into the past.  Requested time 1539165817.346780000 but the earliest data is at time 1539165817.434059095, when looking up transform from frame [odom] to frame [imu_link]

Please check what frame_id your odometry input is in. 
This should not be "odom" because this frame_id is used by Cartographer for output.
(Your urdf does not define any transform regarding odometry.)

Also, to rule out other issues, please publish without odometry data and configure use_odometry = false in the lua, and check if that works.

SarahMars, 2018-10-19:
Thank you so much for your reply!
I didn't know that I could directly feed PointCloud2 message for 2D Cartographer. I've tested with PointCloud2 and it is working fine. 

I have found out that even with tons of warning codes, I could see data via Point Viewer which implies that Cartographer 3D might be appropriately working. Although I still want to get rid of that warning.

> Please check what frame_id your odometry input is in. This should not be "odom" because this frame_id is used by Cartographer for output.
As you pointed out, the frame_id of my odometry was "odom". I will take a deeper look into this broadcaster and let you know the result shortly.

> Also, to rule out other issues, please publish without odometry data and configure use_odometry = false in the lua, and check if that works.
When I publish without odometry by setting use_odometry = false, it is working without warning.

SarahMars, 2018-10-23:
TF between odom frame and base_link is broadcasted by an open source package.

This part wasn't really clear to me,  
> Please check what frame_id your odometry input is in.
This should not be "odom" because this frame_id is used by Cartographer for output.

From my understanding, I thought the odom frame should be connected to the map frame.
Would you be able to explain more about this? 

- If I just simply change the frame id of odom to something else like this,  
[image]
Of course It just says
`[ WARN] [1540293815.575363621]: W1023 20:23:35.000000 30618 tf_bridge.cc:52] "odom" passed to lookupTransform argument source_frame does not exist. 
`

- If I change the published frame to "base_link", no warning shows up. Although the map status in RViz goes ok to error continuously and the visualized map flips every few seconds.
[image]
[image]

SarahMars, 2018-10-23:
Its frame id is "odom"

```
header: 
  seq: 216322
  stamp: 
    secs: 1540302923
    nsecs: 996946096
  frame_id: "odom"
child_frame_id: "base_link"
```

gaschler, 2018-10-23:
```
  frame_id: "odom"
child_frame_id: "base_link"
```
This conflicts with Cartographer's output.

SarahMars, 2018-10-23:
What could be a solution for this conflict when I am trying to use Cartographer with an odometry source?

gaschler, 2018-10-24:
Proper way would be to configure the node that publishes the odometry messages to change the frame_id.

A quick and dirty way to just get Cartographer running would be to set `provide_odom_frame = false,` and rename `odom_frame` in the lua file.

SarahMars, 2018-10-29:
@gaschler 

For 2D Cartographer, setting `use_odometry=false` was working fine, though I forgot to check it for 3D. For 3D, if I disable use odometry, its result looks like this. The map was spinning in the beginning and it gave me the same warning as always. 

```
[ WARN] [1540818900.009024555]: W1029 22:15:00.000000  5225 tf_bridge.cc:52] Lookup would require extrapolation into the past.  Requested time 1540818844.151189000 but the earliest data is at time 1540818890.037019062, when looking up transform from frame [odom] to frame [imu_link]
```
[image]

SarahMars, 2018-10-29:
Also I am looking for a way to rename the frame_id "odom".
The problem here is that even if I rename the frame_id of the odometry topic, I still need to put that inside a TF tree some how. I am not sure how people with `use_odometry=true` is dealing with TF tree.

Here is the current TF tree, (where "odometry source" is the conflicting frame)
[image]

Do you mean that I have to change TF tree like one of following?
**Possible TF tree1**
[image]

**Possible TF tree2**
[image]

Please let me know if I am doing something completely wrong...

Thank you

rdesc, 2018-11-08:
I am also confused/stuck with this. Should tf tree be map -> odom (cartographer) -> odom (odometry source) -> base_link ??

gaschler, 2018-11-09:
I'll try to give a general answer.
The underlying issue with frame_id "odom" or transform "odom -> base_link" it is both the default output of SLAM algorithms and sometimes also by wheel encoders or wheel odometry. This should be avoided.
(Don't have two components publish the same transform.)

Cartographer (and other SLAM algorithms) output two transforms, "map -> odom" for the global loop closure correction and "odom -> base_link" for the transform from current pose of the robot to the first one.

All input sensors are rigidly attached to the robot, so they should be in frame_id "base_link" or a frame attached to base_link.
https://google-cartographer-ros.readthedocs.io/en/latest/ros_api.html#required-tf-transforms
(It does not make sense to have wheel odometry input in the "odom" frame, that is the first pose of the robot trajectory.)
A common configuration is to receive messages from the **odometry/ topic** in a **frame_id "base_footprint"** or similar, and define the (static) transform "base_link <-> base_footprint" in urdf.

rdesc, 2018-11-09:
@gaschler could you just confirm that my configuration is valid? map -> odom -> odom_source -> base_footprint -> base_link

rdesc, 2018-11-09:
@gaschler Could you confirm my configuration shown above is valid?

rdesc, 2018-11-09:
This goes against the configuration for the TurtleBot + carto demo linked below since the transform odom -> base_footprint is done by rosbag which originally was by the mobile_base (contains the odometry) https://storage.googleapis.com/cartographer-public-data/bags/turtlebot/cartographer_turtlebot_demo.bag 
[image]

closed 2018-11-20 by SarahMars

johnchars, 2020-08-01:
I also meet this trouble, tf tree like this solved it.

[image]