cartographer-project/cartographer_ros #1058: Deutsches Museum - TSDF Tuning

alexmillane, 2018-10-11:
Thanks @kdaun for the addition of TSDF mapping; this is great work and I'm excited to use it!

I was just wondering if someone had played around and found some good tunings for the Deutsches Mueseum dataset? With the default params in the scans appear to "slide around" quite a bit relative to the (sub)map.

kdaun, 2018-10-12:
The TSDF is quite sensitive to the parameter configuration, here a [configuration for the Deutsches Museum dataset](https://drive.google.com/open?id=0BwEl1ypVvuqLSlRjWU5JazNLTThfY0toZ2F2S3ZaTVBTbGtB) and the resulting map:
[image]
In my experience, in comparison to the Occupancy Grid pipeline, TSDF can improve the accuracy, the robustness to poor initial guesses and the convergence rate of the scan matching at the cost of a more expensive map updated and worse upper bounds for the branch-and-bound constraint search.
The airplane hall in the Deutsche Museum yields a reduced accuracy, as the TSDF is more sensitive to the strongly moving point observations when scanning the wings.

I'm still working on the TSDF approach and looking forward to hear about your results.

alexmillane, 2018-10-12:
Cheers for this! Works great! Looking forward to seeing how things progress. I'll let you know if I manage to come up with anything interesting.

Again, great work and thanks for the help.

closed 2018-10-12 by alexmillane