cartographer-project/cartographer_ros #1059: Cartographer_ros fails to build

fgrigis, 2018-10-12:
_Hi

I'm trying to build cartographer-ros on a new machine. I'exactly following the instalation guide [here](https://google-cartographer-ros.readthedocs.io/en/latest/).

Setup: Ubuntu 16.04,  ROS kinetic

@HPZ240:~/catkin_ws$ protoc --version
libprotoc 3.4.0

@HPZ240:~/catkin_ws$ lua
Lua 5.2.4  Copyright (C) 1994-2015 Lua.org, PUC-Rio
> 

@HPZ240:~/catkin_ws$ gcc --version
gcc (Ubuntu 5.5.0-12ubuntu1~16.04) 5.5.0 20171010
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

I have done the exact same installation on another machine and it went through without any build errors. I've been trying to fix it for some time now but I'm lost. So I hope one of you could help me here is what I do:_

@HPZ240:~/catkin_ws$ catkin_make_isolated --install --use-ninja

and I get:
@HPZ240:~/catkin_ws$ catkin_make_isolated --install --use-ninja
Base path: /home/fabian/catkin_ws
Source space: /home/fabian/catkin_ws/src
Build space: /home/fabian/catkin_ws/build_isolated
Devel space: /home/fabian/catkin_ws/devel_isolated
Install space: /home/fabian/catkin_ws/install_isolated
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~~  traversing 5 packages in topological order:
~~  - cartographer_ros_msgs
~~  - ceres-solver (plain cmake)
~~  - cartographer (plain cmake)
~~  - cartographer_ros
~~  - cartographer_rviz
~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

==> Processing catkin package: 'cartographer_ros_msgs'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/fabian/catkin_ws/build_isolated/cartographer_ros_msgs'
ninja: no work to do.
==> ninja -j8 -l8 in '/home/fabian/catkin_ws/build_isolated/cartographer_ros_msgs'
[13/13] cd /home/fabian/catkin_ws/build_isolated/cartogr.../srv/WriteState.srv cartographer_ros_msgs/StatusResponse
==> ninja install in '/home/fabian/catkin_ws/build_isolated/cartographer_ros_msgs'
[14/14] Install the project...
-- Install configuration: ""
 
_..... Then there's a bunch of_ '--Up-to-date: /... _messages I don't think it's of any use to includ theme_ here. After the 'up-to-date-lines it continues like this:

Up-to-date: /home/fabian/catkin_ws/install_isolated/lib/cmake/Ceres/FindGflags.cmake
-- Up-to-date: /home/fabian/catkin_ws/install_isolated/lib/libceres.a
<== Finished processing package [2 of 5]: 'ceres-solver'

==> Processing plain cmake package: 'cartographer'
==> Building with env: '/home/fabian/catkin_ws/install_isolated/env.sh'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/home/fabian/catkin_ws/build_isolated/cartographer/install'
ninja: no work to do.
==> ninja -j8 -l8 in '/home/fabian/catkin_ws/build_isolated/cartographer/install'
[8/109] Linking CXX executable cartographer.mapping.internal.motion_filter_test
FAILED: : && /usr/bin/c++   -O3 -DNDEBUG   CMakeFiles/cartographer.mapping.internal.motion_filter_test.dir/cartographer/mapping/internal/motion_filter_test.cc.o  -o cartographer.mapping.internal.motion_filter_test  -rdynamic libcartographer.a gmock/libgmock_main.a -lpthread libcartographer_test_library.a libcartographer.a /home/fabian/catkin_ws/install_isolated/lib/libceres.a -lglog -lgflags -lspqr -ltbb -ltbbmalloc -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lf77blas -latlas -lsuitesparseconfig -lrt -lcxsparse -llapack -lf77blas -latlas -lsuitesparseconfig -lrt -lcxsparse -lgomp -lpthread -llua5.2 -lm -lboost_iostreams -lboost_regex -lglog -lgflags -lcairo /usr/local/lib/libprotobuf.so -lpthread -Wl,-rpath,/usr/local/lib && :
libcartographer.a(motion_filter_options.pb.cc.o): In function `cartographer::mapping::proto::MotionFilterOptions::SerializeWithCachedSizes(google::protobuf::io::CodedOutputStream*) const':
motion_filter_options.pb.cc:(.text+0x14a): undefined reference to `google::protobuf::internal::proto3_preserve_unknown_'

_Then an unbeliaveble amount of messages follow which all are of the type_ 'undefined reference to `google::protobuf'  _........_

_The last lines before exiting are as follows:_
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['/home/fabian/catkin_ws/install_isolated/env.sh', 'ninja', '-j8', '-l8']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/fabian/catkin_ws/build_isolated/cartographer && /home/fabian/catkin_ws/install_isolated/env.sh ninja -j8 -l8

Command failed, exiting.

Does anyone know what to do? sees to be something with the protobuf stuff but I'm kind of a newbie so not sure how to tackle this issue. Thank you for your help!!

erwincoumans, 2018-10-23:
Same error here:
~~~~~~~~~~~~~~~~~~~~
<== Finished processing package [2 of 5]: 'ceres-solver'

==> Processing plain cmake package: 'cartographer'
==> Building with env: '/usr/local/google/home/erwincoumans/catkin_ws/install_isolated/env.sh'
build.ninja exists, skipping explicit cmake invocation...
==> ninja build.ninja in '/usr/local/google/home/erwincoumans/catkin_ws/build_isolated/cartographer/install'
ninja: no work to do.
==> ninja -j12 -l12 in '/usr/local/google/home/erwincoumans/catkin_ws/build_isolated/cartographer/install'
[1/106] Linking CXX executable cartographer_migrate_serialization_format
FAILED: cartographer_migrate_serialization_format 
: && /usr/bin/c++  -O3 -DNDEBUG  -rdynamic CMakeFiles/cartographer_migrate_serialization_format.dir/cartographer/io/migrate_serialization_format_main.cc.o  -o cartographer_migrate_serialization_format  -Wl,-rpath,/usr/local/lib: libcartographer.a /usr/local/google/home/erwincoumans/catkin_ws/install_isolated/lib/libceres.a -lglog /usr/lib/x86_64-linux-gnu/libgflags.so.2.2.1 -lspqr -lcholmod -lccolamd -lcamd -lcolamd -lamd -llapack -lf77blas -latlas -lsuitesparseconfig -lrt -lcxsparse -llapack -lf77blas -latlas -lsuitesparseconfig -lrt -lcxsparse -lgomp -llua5.2 -lm -lboost_iostreams -lboost_regex -lglog /usr/lib/x86_64-linux-gnu/libgflags.so.2.2.1 -lpthread -lcairo /usr/local/lib/libprotobuf.so -lpthread && :
/usr/bin/ld: warning: libgflags.so.2, needed by /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libglog.so, may conflict with libgflags.so.2.2
/usr/bin/ld: warning: libgflags.so.2, needed by /usr/lib/gcc/x86_64-linux-gnu/7/../../../x86_64-linux-gnu/libglog.so, may conflict with libgflags.so.2.2
libcartographer.a(pose_graph.pb.cc.o): In function `cartographer::mapping::proto::protobuf_cartographer_2fmapping_2fproto_2fpose_5fgraph_2eproto::(anonymous namespace)::protobuf_RegisterTypes(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
pose_graph.pb.cc:(.text+0x661): undefined reference to `google::protobuf::GoogleOnceInitImpl(long*, google::protobuf::Closure*)'
libcartographer.a(pose_graph.pb.cc.o): In function `cartographer::mapping::proto::PoseGraph_LandmarkPose::GetMetadata() const':
pose_graph.pb.cc:(.text+0x6e1): undefined reference to `google::protobuf::GoogleOnceInitImpl(long*, google::protobuf::Closure*)'
~~~~~~~~~~~~~~~~~~~~
(etc)

cediy2088, 2018-11-14:
I have the same problem,who can help me?

cediy2088, 2018-11-27:
I have solved this error,in my ubuntu, 
in /usr/local/lib 
has another protobuf which version is not 3.4.1, remove it. Recompile is OK.

closed 2020-10-23 by MichaelGrupp