cartographer-project/cartographer_ros #1064: Cartographer not building map

boesenbe, 2018-10-15:
Hi,

I am trying to use cartographer with the following data:

[Bagfile](https://github.com/boesenbe/Issue_Cartographer/blob/master/2018-10-05-15-07-19.bag)

and with the following Lua and Launchfile

[Lua](https://github.com/boesenbe/Issue_Cartographer/blob/master/settings_cartographer.lua)

[Launchfile](https://github.com/boesenbe/Issue_Cartographer/blob/master/mapping_cartographer.launch)

Output of the 

 ```rosrun cartographer_ros cartographer_rosbag_validate -bag_filename <bag filename>```

command is:

```
E1015 16:39:03.178186  7276 rosbag_validate_main.cc:350] frame_id "" on topic /odometry has serialization time 1538744845.520863038 but sensor time 0.000000000 differing by -1.53874e+09 s.
E1015 16:39:03.178403  7276 rosbag_validate_main.cc:350] frame_id "" on topic /odometry has serialization time 1538744845.521473281 but sensor time 0.000000000 differing by -1.53874e+09 s.
E1015 16:39:03.178436  7276 rosbag_validate_main.cc:350] frame_id "" on topic /odometry has serialization time 1538744845.531494877 but sensor time 0.000000000 differing by -1.53874e+09 s.
E1015 16:39:03.178603  7276 rosbag_validate_main.cc:131] Input contains transform message from frame_id "map" to child_frame_id "odom". This is almost always output published by cartographer and should not appear as input. (Unless you have some complex remove_frames configuration, this is will not work. Simplest solution is to record input without cartographer running.)
I1015 16:39:03.227418  7276 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/odometry" (frame_id: ""):
Count: 4583  Min: 0.000000  Max: 0.000000  Mean: 0.000000
E1015 16:39:03.227432  7276 rosbag_validate_main.cc:382] Point data (frame_id: "back_laser") has a large gap, largest is 0.149883 s, recommended is [0.0005, 0.05] s with no jitter.
I1015 16:39:03.227540  7276 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/scan_rear" (frame_id: "back_laser"):
Count: 831  Min: 0.009310  Max: 0.149883  Mean: 0.053032
[0.009310, 0.023367)	                 ###	Count: 117 (14.079422%)	Total: 117 (14.079422%)
[0.023367, 0.037424)	                 ###	Count: 121 (14.560770%)	Total: 238 (28.640192%)
[0.037424, 0.051481)	              ######	Count: 269 (32.370636%)	Total: 507 (61.010830%)
[0.051481, 0.065539)	                   #	Count: 49 (5.896510%)	Total: 556 (66.907341%)
[0.065539, 0.079596)	                  ##	Count: 79 (9.506618%)	Total: 635 (76.413956%)
[0.079596, 0.093653)	                 ###	Count: 115 (13.838749%)	Total: 750 (90.252708%)
[0.093653, 0.107711)	                   #	Count: 31 (3.730445%)	Total: 781 (93.983154%)
[0.107711, 0.121768)	                   #	Count: 37 (4.452467%)	Total: 818 (98.435623%)
[0.121768, 0.135825)	                    	Count: 5 (0.601685%)	Total: 823 (99.037308%)
[0.135825, 0.149883]	                    	Count: 8 (0.962696%)	Total: 831 (100.000000%)
E1015 16:39:03.227582  7276 rosbag_validate_main.cc:382] Point data (frame_id: "front_laser") has a large gap, largest is 0.150487 s, recommended is [0.0005, 0.05] s with no jitter.
I1015 16:39:03.227661  7276 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/scan_front" (frame_id: "front_laser"):
Count: 832  Min: 0.009538  Max: 0.150487  Mean: 0.052945
[0.009538, 0.023633)	                 ###	Count: 124 (14.903846%)	Total: 124 (14.903846%)
[0.023633, 0.037728)	                 ###	Count: 116 (13.942307%)	Total: 240 (28.846153%)
[0.037728, 0.051823)	              ######	Count: 252 (30.288462%)	Total: 492 (59.134617%)
[0.051823, 0.065918)	                   #	Count: 50 (6.009615%)	Total: 542 (65.144234%)
[0.065918, 0.080013)	                 ###	Count: 129 (15.504807%)	Total: 671 (80.649040%)
[0.080013, 0.094108)	                 ###	Count: 107 (12.860577%)	Total: 778 (93.509613%)
[0.094108, 0.108203)	                    	Count: 17 (2.043269%)	Total: 795 (95.552887%)
[0.108203, 0.122298)	                   #	Count: 24 (2.884615%)	Total: 819 (98.437500%)
[0.122298, 0.136393)	                    	Count: 7 (0.841346%)	Total: 826 (99.278847%)
[0.136393, 0.150487]	                    	Count: 6 (0.721154%)	Total: 832 (100.000000%)
```

My problem is if I play the bag and start Cartographer no map is build. Just *tf* is showing me the Coordinate Systems.

[image]

Many thanks for your help! Have a nice day.

gaschler, 2018-10-15:
> E1015 16:39:03.178186  7276 rosbag_validate_main.cc:350] frame_id "" on topic /odometry has serialization time 1538744845.520863038 but sensor time 0.000000000 differing by -1.53874e+09 s.

Your odometry messages are corrupted.

Does it work without odometry as an input?

boesenbe, 2018-10-17:
Hi,

I used the command ```rosbag play <filename> --clock``` and now I get a **tf** warning. 

```[ WARN] [1539768285.605400493, 1539767248.506165225]: W1017 11:24:45.000000   556 sensor_bridge.cc:207] Ignored subdivision of a LaserScan message from sensor scan_2 because previous subdivision time 636753640380916281 is not before current subdivision time 636753640380916281
[ WARN] [1539768285.807042053, 1539767248.707688979]: W1017 11:24:45.000000   556 tf_bridge.cc:52] "base_frame" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1539768286.008355997, 1539767248.909093888]: W1017 11:24:46.000000   556 tf_bridge.cc:52] "base_frame" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1539768286.209643275, 1539767249.110810895]: W1017 11:24:46.000000   556 tf_bridge.cc:52] "base_frame" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1539768286.411425309, 1539767249.312319394]: W1017 11:24:46.000000   556 tf_bridge.cc:52] "base_frame" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1539768286.612931116, 1539767249.513774807]: W1017 11:24:46.000000   556 tf_bridge.cc:52] "base_frame" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1539768286.814957667, 1539767249.715318270]: W1017 11:24:46.000000   556 tf_bridge.cc:52] "base_frame" passed to lookupTransform argument source_frame does not exist. 
```
I also set the header time from my odometry with ```odom.header.stamp = rospy.Time.now();``` and I got rid of the 

```E1015 16:39:03.178186 7276 rosbag_validate_main.cc:350] frame_id "" on topic /odometry has serialization time 1538744845.520863038 but sensor time 0.000000000 differing by -1.53874e+09 s.``` 

Error.

MichaelGrupp, 2020-10-23:
You need to fix your TF frame IDs. Empty ones indicate that something is pretty broken.

closed 2020-10-23 by MichaelGrupp