cartographer-project/cartographer_ros #1069: Map frame lost when pure_localization

alfred31, 2018-10-19:
Hello,

First of all thanks for the work done.

I have been working with the cartographer the last months and I have been able to set up almost everything and made it work really good.
Now I am working with the pure localization node, but I am having problems time to time with it.
The point is that for some reason the pure localization node stops sending the "map" transform after a while, and therefore the link between "base_link" and "map" gets lost and the pure localization stops. Although the weird thing is that I do not received any message from cartographer telling that the node has crashed.

Thanks in advance.

Kind Regards.

gaschler, 2018-10-19:
What kind of crash did you experience? Any error messages or stack trace?

alfred31, 2018-10-19:
It just stops sending the transform, that was the weird thing, cartographer did not send any crash message I got the feeling that it was still running but the connection between the base_link of the robot and the map_frame sent by the cartographer disappeared and it is not able to take it back again so cartographer gets disconnected form my robot description.
I know it is a weird  but  any idea of what could be causing this error will be helpful. This is the last message I got before loosing the track of the frames:
```
[ INFO] [1539869984.708266908]: I1018 15:39:44.000000  8478 motion_filter.cc:42] Motion filter reduced the number of nodes to 15.6286%.
[ INFO] [1539869989.513644053]: I1018 15:39:49.000000  8478 collated_trajectory_builder.cc:72] imu rate: 99.95 Hz 1.00e-02 s +/- 4.73e-03 s (pulsed at 100.58% real time)
[ INFO] [1539869989.513731481]: I1018 15:39:49.000000  8478 collated_trajectory_builder.cc:72] odom rate: 7.47 Hz 1.34e-01 s +/- 2.11e-02 s (pulsed at 99.94% real time)
[ INFO] [1539869989.513797148]: I1018 15:39:49.000000  8478 collated_trajectory_builder.cc:72] scan rate: 29.99 Hz 3.33e-02 s +/- 2.28e-03 s (pulsed at 100.75% real time)
```

As you can see no error messages are received. I tried fixing it by increasing the odom rate because it seems too low, but the error still occurred.

Thanks for your quick response and your help.

alfred31, 2018-10-19:
Ok, thanks for your help!

I will track this issue and report the outcomes in case it is helpful for other people.

Kind Regards

alfred31, 2018-10-19:
Hello,
I discovered that the error occurs when I use both cartographer and move-it since ther is a robotic arm mounted on the mobile robot.
Don't know the reason why this is happening but it is for sure due to this.
Best Regards

alfred31, 2018-10-25:
Hello,
I found the problem.
As I am using an actual robot (no simulation) I had this problem with the frames due to the use_sim_time parameter set to true: A I was not simulating anymore It is clear that this parameter should be set to false. Doing this the problem vanished.

Thanks a lo for your kind help, and of course for the amazing job you have done with cartographer.

closed 2018-10-25 by alfred31