cartographer-project/cartographer_ros #1070: cartographer can't save map

haolingyi, 2018-10-21:
Thank you for filing an issue!

It would help us tremendously if you run through the following checklist. This
ensures that we have the most information to quickly understand, analyze,
reproduce and eventually resolve your issue.

Please

1. run `rosbag_validate` which does some checks on your sensor data. This
   tool often finds issues that can explain poor performance and must be fixed
   at recording time. Please post the full output of the tool into a
   GitHub Gist at https://gist.github.com/, then link it in the issue even if
   it does not report anything. You can run the tool like this:

   rosrun cartographer_ros cartographer_rosbag_validate -bag_filename <bag filename>

2. post a link to a Git repository containing a branch of
   `cartographer_ros` containing all the configuration, launch, and URDF files
   required to reproduce your issue.
3. post a link to a bag file we can use to reproduce your issue. Put it on
   Google Drive, Dropbox, any webserver or wherever it is publicly
   downloadable.
4. remove this boilerplate text before submitting your issue.

We will likely be unable to help you without all of these points addressed.

zhenliysu, 2018-11-05:
First, you should check whether  the "map" topic has been published？ 
The "map" topic is published by cartographer_occupancy_grid_node node.
e.g.
  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />

haolingyi, 2018-11-05:
Thank you,I have solved this problem by set cartographer_occupancy_grid_node.

zhaoyu24, 2019-03-27:
> Thank you,I have solved this problem by set cartographer_occupancy_grid_node.

I had the same problem,could please tell me how do you set the cartographer_occupancy_grid_node.

MichaelGrupp, 2020-10-21:
> > Thank you,I have solved this problem by set cartographer_occupancy_grid_node.
> 
> I had the same problem,could please tell me how do you set the cartographer_occupancy_grid_node.

You can rosrun cartographer_ros cartographer_occupancy_grid_node ...

closed 2020-10-21 by MichaelGrupp