cartographer-project/cartographer_ros #1072: Radar support

ShettyHarapanahalli, 2018-10-23:
Hi, 

Does Cartographer support radar interface like from TI, details of the radar is given below - https://github.com/radar-lab/ti_mmwave_rospkg

Best Regards,
:)

Entropy512, 2018-11-07:
The IWR1443 doesn't have even remotely enough resolution to be useful with Cartographer.

ShettyHarapanahalli, 2018-11-22:
Thanks for sharing this information. if I need the 3d SLAM can I use a 2d Lidar on a "pan and tilt platform" to get the 3D data. Planning to use rplidar A3 [http://www.slamtec.com/en/Lidar/A3] on a tilt platform.  

Ex for a pan and tilt platform - 
https://www.robotshop.com/eu/en/spt200-drive-pan-tilt-system-futaba.html?gclid=EAIaIQobChMI4YbtrJPo3gIVArTtCh2X5gUoEAQYASABEgJlf_D_BwE

ShettyHarapanahalli, 2018-11-22:
I see a paper on constructing a 3d point cloud from 2d lidar - 
https://www.hindawi.com/journals/js/2018/2937694/

closed 2019-02-05 by ShettyHarapanahalli