cartographer-project/cartographer_ros #1083: Triggering global localization on service request

mojovski, 2018-10-31:
I am currently running into problems when the robot moves into a corridor, which has ambiguous scan data.
In this case, cartographer relocates the robot on the map to somewhere else. When the robot passes the corridor, the correct map pose is restored.

Is there a way to trigger the global localization only on a service request? It would be more than enough to trigger this by the user, until the pose is OK. And after this, the odometry should be regarded as the main source of movement (without global map matching).

Or can I somehow disable the jumps but still using an existing map for pure localization?

I am not sure, if cartographer_ros or the pure cartographer is more appropriate for this issue. 
Please advice.

Best

MichaelGrupp, 2018-11-05:
>  It would be more than enough to trigger this by the user, until the pose is OK

Unless you really need global search, I would recommend to disable it and to use the [start_trajectory](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/start_trajectory_main.cc) executable with the initial pose argument. There was some discussion about this here: https://github.com/googlecartographer/cartographer_ros/issues/913#issuecomment-401522456

mojovski, 2018-11-06:
Hi @MichaelGrupp , thank you very much!
* how do I disable the global search? 
Setting one of these to some crazy value?
```
POSE_GRAPH.optimize_every_n_nodes = 2 
POSE_GRAPH.global_sampling_ratio=0.003
POSE_GRAPH.global_constraint_search_after_n_seconds=10 
```

* If I use the start_trajectory entry point, how precise has my estimate to be? Will cartographer still fix smaller errors in the passed start-pose? 
* If so, up to which number? 
* Can this parameter help to control the error correction coming from the initial pose?
```
TRAJECTORY_BUILDER_2D.real_time_correlative_scan_matcher.linear_search_window = 0.4
```

MichaelGrupp, 2018-11-06:
Setting `POSE_GRAPH.global_constraint_search_after_n_seconds` very high will "disable" it, yes. I don't think there is an explicit disabling option at the moment.

If global search is deactivated, Cartographer will only do a local constraint search just as in normal loop closure. The parameters for the search window of this search can be set for example like this:
```lua
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher.linear_search_window = 0.75
POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher.angular_search_window = math.rad(10.0)
```
The "real time correlative matcher" is used for local SLAM / motion estimation and has a different purpose.

> Will cartographer still fix smaller errors in the passed start-pose?

Yes. Usually it should "snap" into place after moving a bit around inside the search window.

closed 2020-10-21 by MichaelGrupp