cartographer-project/cartographer_ros #1088: Ouster LiDAR Scanner Configuration Files

dsb6063, 2018-11-04:
Hello, 

I have an Ouster OS-1 3D LiDAR Scanner with imu similar to Velodyne VLP-16 and having problems with the configuration files. 

System Ubuntu 14.05

Ouster Git.

bag files and drivers are publicly available at https://github.com/ouster-lidar/ouster_example

My files are below.

https://github.com/dsb6063/cartographer_ros

I would appreciate any assistance with these files. 

Thanks!

brh5047, 2018-11-09:
Have you made any progress on this? I've been working on integrating an OS-1 64 channel with cartographer. Having some troubles with the configuration files as well. I think you still need to adjust some of your topics, but I'm not 100% sure on that.

I've created new .lua .urdf and .rviz files in addition to a new launch file.  Will push them later today so that you can check it out.

dsb6063, 2018-11-09:
Hi Brennan,

 

Any help would be appreciated. I have not been able to view the tf tree to track down my error in the configuration files. That led me to  review a few of the issues on the Ouster Git, and it appears the tf tree is not working with the timestamp from the Ouster driver. I have not configured the ptp yet, and have started the short-term workaround where the person changed the timestamp source in the code. 

 

https://github.com/ouster-lidar/ouster_example/issues/1 

 

Thanks,

 

Daryl

dsb6063, 2018-11-10:
HI Brennan,

 

Any luck? The web docs for cartographer updated and I walked thru with a result. After checking the bag files has shown possible issues,  I did implement a static transform, which is best viewed in rviz. Still guessing

 

I did not believe timing was an issue since I am only using one device with no plans to sync or correlate with GPS positions. I am not sure if this is a mandatory to trigger off a master clock for 3D.

 

Along with the other terminal, in a separate terminal.

 

rosrun tf static_transform_publisher 0.0 0.0 0.0 0.0.0 0.0 os1 map 100

eric-schleicher, 2018-11-28:
I'll comment since this is still open...  getting the ouster running with ptp4l should be fairly straightforward.

`ptp4k -i eth0 -m -H`  for hardware timestamping on the ethernet
or
`ptp4k -i eth0 -m -S`  for software timestamping on the ethernet

dsb6063, 2018-11-30:
Thanks for the information on time stamping. I made a wrong assumption about the timestamp and configuration. I was directed to an example to get started using Ouster with these config files if anyone is interested. 

https://github.com/wilselby/ouster_example
 
Use his cartographer configuration, launch, and urdf files as a template. This is what you need to get started.

dsb6063, 2018-12-05:
After creating the pbstream file, I cannot create the point cloud using the asset_writer_cart_3d. I am having trouble with the output of the points.ply or .xyz. I have attached the files. The configuration files that need attention are the files inside the "test" folder. 

Basically, when I run the assets_writer, it runs for 10 seconds and stops at processing 0 out of  number of messages and exits. It seems that running the same files again should not have issues with the file? I tested against different configuration settings. Same error. Therefore, it sounds like a bagfile issue. 

Any help would be appreciated. This is my first field bag--Therefore, my understanding of all the settings withing the configuration is still fuzzy. 

My setup is VMWare Workstation 8BG running on Win 10 32GB.
Ubuntu 18.04, ROS Melodic.

After running rosbag validate, it did show 1 gap of 3 second; however it produced the pbstream file.
 
Here is the link to configuration and ROS log folder--

https://github.com/dsb6063/cartographer_ros/tree/master/test

Here is the link to the pbstream output file and the bag file (9GB)--

https://drive.google.com/open?id=1QcmwAOhb6Pae4nlsJSWsMjeGIk2tTnYt
 
https://github.com/dsb6063/cartographer_ros/tree/master/cartographer_ros

cmoxie, 2018-12-21:
Just a note on the time stamping, we have been unable to get the ouster working with ptp4l. Anyone have any tips for doing so? 

Instead, we did as was suggested here: https://github.com/ouster-lidar/ouster_example/issues/1

Time stamping the ouster point cloud messages with ros::Time::now(). 

As was mentioned in the post, this is not a perfect solution, it doesn't stamp when the data was obtained, but when it was parsed by ros. This lead to an issue with the transforms trying to request data that was a couple milliseconds out of date. We fixed this by using tf2 instead of tf to publish the static transforms between base_link and os1 and os1_imu. We also have the lookup_transform_timeout_sec set to 2 in the .lua . We're not sure which one of these changes fixed it or why, but it seems to be working for now. 

Overall, we're still not getting great data from the ouster. If you've made any progress we'd love to hear what you have done.

dsb6063, 2018-12-21:
One suggestion is to use a low cost hardware implementation to bypass those issues. We are implementing the Garmin GPS18x LVC 5m.  We are using it for the timestamp from the GPS Almanac and not for positional data to synchronize two or more sensors. It costs $60-80 US. Inside the interface box on the Ouster, there is a connection for pps-master.  Then, configure the sensor to receive the external PPS protocol. It would have been nice to already have this port on the interface box for an external timer. 

 

IEEE 1588 precision time protocol

External PPS

It’s been a challenge with the data. Once I have a good setup with workflow, I’ll post it. There are a few people that already have a master/slave ptp setup. Maybe they will chime in?

jattiogbe, 2019-02-05:
I tried using the configuration files as suggested in the message from december 1st, but none of the launch files  lets me see the bagfile in rviz. 
When I try  `roslaunch cartographer_ros demo_cart_3d.launch bag_filename:=~/u_2019-02-01.bag`, I get this error message : 

>  requires the 'bag_filenames' arg to be set

What should I do exactly based on those configuration files ?

dsb6063, 2019-02-05:
Add the “s” to bag_filename

 

roslaunch cartographer_ros demo_cart_3d.launch bag_filenames:=~/u_2019-02-01.bag

jattiogbe, 2019-02-06:
There are quite a few launch files, excuse me but I've really really discovering this for a few days only. So here is what I understand from it, correct me if I'm wrong : 

- **assets_writer_cart_2d/3D.launch** : 
uses a .bag and .pbstream to generate various files such as ply files. This is what I would like to obtain myself.

 
- **cart_2/3D.launch** : 
unlike the offline version, they don't need bagfiles to run, so I guess it's for using directly the os1 ?

- **demo_cart2d/3d.launch** : 
is this to play bagfiles in rviz ?

Another thing, I tried cartographer_rosbag_validate on a bagfile of mine and it just returned nothing...

Where should I start  ?
I thought I'd run some launch file (which one ?) to generate a pbstream, then asset_writer_3D to get a .ply file, but eventually I think I'm lost.

gaschler, 2019-02-06:
> Another thing, I tried cartographer_rosbag_validate on a bagfile of mine and it just returned nothing...

If cartographer_rosbag_validate does not print anything, please rerun it in a debugger (`gdb --args cartographer_rosbag_validate ...`) and report segfaults or the like. Even if there are no warning, it should print the time distribution histograms of the different topics that have sensor data.

jattiogbe, 2019-02-06:
Here is what I got from gdb : 
```
(gdb) r
Starting program: /home/julien/goo_carto_ws/install_isolated/bin/cartographer_rosbag_validate /home/julien/lidar_2019_01/cat_ws/a_2019-01-30-17-25-45.bag
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
F0206 14:00:24.347297 13011 rosbag_validate_main.cc:422] Check failed: !FLAGS_bag_filename.empty() -bag_filename is missing.
*** Check failure stack trace: ***
    @     0x7ffff7bb00cd  google::LogMessage::Fail()
    @     0x7ffff7bb1f33  google::LogMessage::SendToLog()
    @     0x7ffff7bafc28  google::LogMessage::Flush()
    @     0x7ffff7bb2999  google::LogMessageFatal::~LogMessageFatal()
    @     0x55555556bfff  (unknown)
    @     0x7ffff5be9b97  __libc_start_main
    @     0x55555556c8fa  (unknown)

Program received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: Aucun fichier ou dossier de ce type.
(gdb) 
```

gaschler, 2019-02-06:
> F0206 14:00:24.347297 13011 rosbag_validate_main.cc:422] Check failed: !FLAGS_bag_filename.empty() -bag_filename is missing.

Please run the tool like so:

```
cartographer_rosbag_validate  -bag_filename /home/julien/lidar_2019_01/cat_ws/a_2019-01-30-17-25-45.bag
```

jattiogbe, 2019-02-06:
Thank you, sorry I don't know how to use the tool...

Here is what I got : 
```
carto_ws$ cartographer_rosbag_validate -bag_filename /home/julien/lidar_2019_01/cat_ws/a_2019-01-30-17-25-45.bag
carto_ws$ 
```
or when I tried with gdb:  

```
Starting program: /home/julien/goo_carto_ws/install_isolated/bin/cartographer_rosbag_validate -bag_filename /home/julien/lidar_2019_01/cat_ws/a_2019-01-30-17-25-45.bag
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Inferior 1 (process 14128) exited normally]
(gdb) 
```

dsb6063, 2019-02-06:
These commands will help. 

*         rosrun cartographer_ros cartographer_rosbag_validate -bag_filename <bag filename>

Record a ROSbag using the OS1 ROS driver

*         roslaunch ouster_ros os1.launch os1_hostname:=<os1_hostname> os1_udp_dest:=<udp_data_dest_ip> scan_dur_ns:=50000000

*         Validate ROSbag

*         rosrun cartographer_ros cartographer_rosbag_validate -bag_filename <bag filename>

*         Run 2D online

*         $ roslaunch cartographer_ros demo_cart_2d.launch bag_filename:=/path/to/<file.bag>

*         Run 2D offline

*         $ roslaunch cartographer_ros offline_cart_2d.launch bag_filenames:=/path/to/<file.bag>

*         Run 2D assets_writer

*         $ roslaunch cartographer_ros assets_writer_cart_2d.launch bag_filenames:=/path/to/<file.bag> pose_graph_filename:=/path/to/<file.bag.pbstream>

 

*         Run 3D online

*         $ roslaunch cartographer_ros demo_cart_3d.launch bag_filename:=/path/to/<file.bag>

*         Run 3D offline

*         $ roslaunch cartographer_ros offline_cart_3d.launch bag_filenames:=/path/to/<file.bag>

*         Run 3D assets_writer

*         $ roslaunch cartographer_ros assets_writer_cart_3d.launch bag_filenames:=/path/to/<file.bag> pose_graph_filename:=/path/to/<file.bag.pbstream>

 

 

Thanks,

 

Daryl Banks, PSM

Banks & Banks Consulting, Inc.

Business:  <tel:%28904%29%20685-2800> (904) 685-2800

Facsimile:  <tel:%28904%29%20685-2345> (904) 685-2345

 <http://www.bankssurveyors.com/> Website   <http://www.linkedin.com/pub/daryl-banks/20/5b5/253/> Linked-In

 

Banks Family Business 29 years surveying Florida.

gaschler, 2019-02-06:
If cartographer_rosbag_validate has an empty result, there are no message types in the bag file that Cartographer can process.

jattiogbe, 2019-02-06:
Thank you both very much.
I ran ros_validate on Daryl's"input.bag", I saw the processing indeed. I 'm going to record a new bagfile with os1 and start over.

jattiogbe, 2019-02-11:
Hi @dsb6063 ,

I think I followed your exact steps but I still can't get to have a right rosbag_validation
* I am in Lidar_mode 1024x20
 * I do :

        `roslaunch ouster_ros os1.launch os1_hostname:=os1-991838000761.local os1_udp_dest:=192.168.1.7 viz:=true scan_dur_ns:=50000000`
      and
      ` rosbag record /os1_node/imu_packets /os1_node/lidar_packets`

* then when I want to validate :
`rosrun cartographer_ros cartographer_rosbag_validate -bag_filename ~/cat_ws/demo_2019-02.bag`
which returns nothing.

I guess I'm stuck there until I manage to validate my bagsfiles.

dsb6063, 2019-02-11:
You need to record/subscribe to the ROS compatible node.

 

This is the Ouster Raw format. This will not work directly with ROS topics. 

 

rosbag record /os1_node/imu_packets /os1_node/lidar_packets

 

You can either record a new bag, or use the existing bag (from your above format) and record using this ROS compatible format—

 

rosbag record /os1_node/imu /os1_node/points 

 

 

See below if you want only the points

 

terminal 1:

roslaunch ouster_ros os1.launch replay:=true

 

terminal 2:

rosbag record /os1_node/points

[ INFO] [1536258906.022794286]: Subscribing to /os1_node/points

[ INFO] [1536258906.025252932]: Recording to 2018-09-06-11-35-06.bag.

 

terminal 3:

rosbag play --clock ~/files/ouster-os1-100sec.bag

 

Then, when recording finishes:

rosrun pcl_ros bag_to_pcd  2018-09-06-11-35-06.bag  os1_node/points ./points

dsb6063, 2019-02-11:
Below is a more detailed explanation of the process.

 

- To publish OS1 data as ROS topic `roslaunch ouster_ros os1.launch os1_hostname:=<os1_hostname> os1_udp_dest:=<udp_data_dest>` where `<os1_hostname>` can be the hostname or IP of the OS1 device and `<udp_data_dest>` is the IP to which the sensor should send data to (i.e., your computer's IP address on the interface connected to the OS1)

	

 

- To record raw sensor output, run `rosbag record /os1_node/imu_packets /os1_node/lidar_packets` in another terminal

	

 

- Note: os1_node/lidar_packets and os1_node/imu_packets are the "raw data" topics, while os1_node/points is the ROS compatible XYZ topic and os1_node/imu is the ROS compatible IMU topic

	

 

- To visualize output, run `rviz -d /path/to/ouster_ros/viz.rviz` in another terminal

	

- *Note: `os1_node/lidar_packets` and `os1_node/imu_packets` are the "raw data" topics, while `os1_node/points` is the ROS compatible XYZ topic and `os1_node/imu` is the ROS compatible IMU topic*

jattiogbe, 2019-02-11:
thank you again Daryl,

My rostopic list says :

```
/clock
/os1_cloud_node/imu
/os1_cloud_node/points
/os1_node/imu_packets
/os1_node/lidar_packets
/rosout
/rosout_agg
/tf_static
```

So I used` os1_cloud_node/imu os1_cloud_node/points` instead of your 

> rosbag record /os1_node/imu /os1_node/points

I was then able to run rosbag_validate, that's a good thing, even if I still don't understand what it's trying to explain to me:
```
 Count: 1627  Min: 0.049318  Max: 0.050451  Mean: 0.049995
[0.049318, 0.049432)	                    	Count: 1 (0.061463%)	Total: 1 (0.061463%)
[0.049432, 0.049545)	                    	Count: 0 (0.000000%)	Total: 1 (0.061463%)
[0.049545, 0.049658)	                    	Count: 2 (0.122926%)	Total: 3 (0.184388%)
[0.049658, 0.049771)	                    	Count: 12 (0.737554%)	Total: 15 (0.921942%)
[0.049771, 0.049885)	                   #	Count: 108 (6.637984%)	Total: 123 (7.559926%)
[0.049885, 0.049998)	           #########	Count: 712 (43.761524%)	Total: 835 (51.321449%)
[0.049998, 0.050111)	            ########	Count: 683 (41.979103%)	Total: 1518 (93.300552%)
[0.050111, 0.050225)	                   #	Count: 97 (5.961893%)	Total: 1615 (99.262444%)
[0.050225, 0.050338)	                    	Count: 8 (0.491703%)	Total: 1623 (99.754150%)
[0.050338, 0.050451]	                    	Count: 4 (0.245851%)	Total: 1627 (100.000000%)
```

Finally, I did **rosrun pcl_ros bag_to_pcd** and now I'm in the same case as [this guy](https://answers.ros.org/question/235063/bag-file-to-pcd-files/)3 years ago.

Sorry for my being so "newbie", I actually am. It's gonna take me a while to handle all this, thanks for your help and patience !

dsb6063, 2019-02-11:
The histogram will display time deltas. Why is this important? It will let you know if there are gaps in timing. A gap over 0.5 second is quite significant with the cartographer during runtime. I have had issues with 1-3 second time gaps. Secondly, the order of the data packets can become an issue as well with the UDP protocol. The validation tool will help find these. 

 

When you get going, send out a best practice. I’m sure you are not the only one that has questions.

 

Here is a good read.

 

https://github.com/googlecartographer/cartographer_ros/issues/529

hachbani2019, 2019-06-12:
Did any of you figure out a way to solve the problem ? working as well with some bag files of an Ouster OS-1 64. Being a ROS beginner, can't figure out a way to solve this. If any of you can share his .launch & .lua files please, i'd be so thankful.

THanks

dsb6063, 2019-07-03:
Here is the link to the config files. They are for a specific implementation using the Ouster OS-1 16. 

https://github.com/wilselby/ouster_example/tree/master/cartographer_ros

I am still having issues getting it working.

fgomezrl87, 2020-01-28:
Hello,

After reading a lot, i could get ply points using cartographer. 

`cd ~/catkin_ws`
`source install_isolated/setup.bash`
`roslaunch cartographer_ros my_robot.launch bag_filenames:=${HOME}/casa.bag`
`roslaunch cartographer_ros my_robot2.launch bag_filenames:=${HOME}/casa.bag pose_graph_filename:=${HOME}/casa.bag.pbstream`

You will need my_robot.lua, my_robot.urdf, my_robot.launch, my_robot2.lua, my_robot2.launch (all attached).

It works very good outdoors. The problem is that the result is not good enought indoors. Hopefully you can help me at this point.

[cartfiles.zip](https://github.com/googlecartographer/cartographer_ros/files/4120343/cartfiles.zip)

closed 2020-10-21 by MichaelGrupp