cartographer-project/cartographer_ros #1091: Saved State makes Cartographer crash next time on load (corrupt pbstream?)

gollth, 2018-11-06:
Hi first of all thanks for the outstanding work with Cartographer! 

We have the following problem, though: Our system requires a method of "autosaving" the state of cartographer, so that we can continue SLAM in the event of a system crash the next time cartographer is started via the `load_state_filename` arg. It's nice, that cartographer saves its current state on graceful shutdowns in the `save_state_filename` but not if it crashes. Therefore we call

`rosservice call /robot/write_state "filename: map.autosave.pbstream"` 

in regular intervals to avoid data loss. However loading this file does not work. I get the following error if I use `visualize_pbstream.launch`:

```
$ roslaunch cartographer_ros visualize_pbstream.launch pbstream_filename:=/tmp/map.autosave.pbstream
... logging to /home/goll_th/.ros/log/afc7ea80-e19c-11e8-a358-7470fd49bc7c/roslaunch-felap-lx013-20162.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.2.180:42633/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    cartographer_offline_node (cartographer_ros/cartographer_offline_node)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [20172]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to afc7ea80-e19c-11e8-a358-7470fd49bc7c
process[rosout-1]: started with pid [20185]
started core service [/rosout]
process[rviz-2]: started with pid [20202]
process[cartographer_offline_node-3]: started with pid [20203]
[ INFO] [1541492372.987252511]: I1106 09:19:32.000000 20203 configuration_file_resolver.cc:41] Found '/home/goll_th/ros/src/morty_ros/cartographer_ros/cartographer_ros/configuration_files/visualize_pbstream.lua' for 'visualize_pbstream.lua'.
[ INFO] [1541492372.987526628]: I1106 09:19:32.000000 20203 configuration_file_resolver.cc:41] Found '/home/goll_th/ros/src/morty_ros/cartographer_ros/cartographer_ros/configuration_files/backpack_2d.lua' for 'backpack_2d.lua'.
[ INFO] [1541492372.987568477]: I1106 09:19:32.000000 20203 configuration_file_resolver.cc:41] Found '/home/goll_th/ros/src/morty_ros/cartographer_ros/cartographer_ros/configuration_files/backpack_2d.lua' for 'backpack_2d.lua'.
[ INFO] [1541492372.987679501]: I1106 09:19:32.000000 20203 configuration_file_resolver.cc:41] Found '/usr/local/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1541492372.987717543]: I1106 09:19:32.000000 20203 configuration_file_resolver.cc:41] Found '/usr/local/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1541492372.987781460]: I1106 09:19:32.000000 20203 configuration_file_resolver.cc:41] Found '/usr/local/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1541492372.987819173]: I1106 09:19:32.000000 20203 configuration_file_resolver.cc:41] Found '/usr/local/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1541492372.987981398]: I1106 09:19:32.000000 20203 configuration_file_resolver.cc:41] Found '/usr/local/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1541492372.988018469]: I1106 09:19:32.000000 20203 configuration_file_resolver.cc:41] Found '/usr/local/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1541492372.988084941]: I1106 09:19:32.000000 20203 configuration_file_resolver.cc:41] Found '/usr/local/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1541492372.988132896]: I1106 09:19:32.000000 20203 configuration_file_resolver.cc:41] Found '/usr/local/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1541492372.988310215]: I1106 09:19:32.000000 20203 configuration_file_resolver.cc:41] Found '/usr/local/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1541492372.988365432]: I1106 09:19:32.000000 20203 configuration_file_resolver.cc:41] Found '/usr/local/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ WARN] [1541492372.989377301]: W1106 09:19:32.000000 20203 fixed_ratio_sampler.cc:26] FixedRatioSampler is dropping all data.
[ INFO] [1541492372.998465426]: I1106 09:19:32.000000 20203 map_builder_bridge.cc:115] Loading saved state '/tmp/map.autosave.pbstream'...
[ WARN] [1541492373.002102452]: W1106 09:19:33.000000 20203 fixed_ratio_sampler.cc:26] FixedRatioSampler is dropping all data.
terminate called after throwing an instance of 'std::out_of_range'
  what():  map::at
[cartographer_offline_node-3] process has died [pid 20203, exit code -6, cmd /home/goll_th/ros/devel/lib/cartographer_ros/cartographer_offline_node -configuration_directory /home/goll_th/ros/src/morty_ros/cartographer_ros/cartographer_ros/configuration_files -configuration_basenames visualize_pbstream.lua -keep_running=true -load_state_filename /tmp/map.autosave.pbstream -load_frozen_state=false __name:=cartographer_offline_node __log:=/home/goll_th/.ros/log/afc7ea80-e19c-11e8-a358-7470fd49bc7c/cartographer_offline_node-3.log].
log file: /home/goll_th/.ros/log/afc7ea80-e19c-11e8-a358-7470fd49bc7c/cartographer_offline_node-3*.log
================================================================================REQUIRED process [rviz-2] has died!
process has finished cleanly
log file: /home/goll_th/.ros/log/afc7ea80-e19c-11e8-a358-7470fd49bc7c/rviz-2*.log
Initiating shutdown!
================================================================================
[rviz-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

However with a `map.pbstream` generated from cartographer when its shutdown via CTRL-C this error does not happen. 

I'm not an expert in protobuf and don't know how to debug the pbstreams. Am I missing something? Can you point me in the right direction?

Attachments to reproduce:
[rosbag_validate_log](https://gist.github.com/gollth/0b5677eedfe5a1fc7f613fe16a3d034d)
[map.autosave.pbstream.zip](https://github.com/googlecartographer/cartographer_ros/files/2552234/map.autosave.pbstream.zip)
[map.pbstream.zip](https://github.com/googlecartographer/cartographer_ros/files/2552235/map.pbstream.zip)

mojovski, 2018-11-06:
yes, check the saved pbstream: this will probably be very small. You will need to call the service including the parameter to save unfinished state:
```
rosservice call /write_state /map/agvstate.pbstream true
```

MichaelGrupp, 2020-10-21:
I agree, this should be improved. I'm not sure if it's trivial to fix, but I will take a look when I have time.

closed 2025-10-22 by gollth