cartographer-project/cartographer_ros #1093: cartographer build map error

zchao9456, 2018-11-08:
process[robot_state_publisher-1]: started with pid [11071]
process[cartographer_node-2]: started with pid [11072]
process[cartographer_occupancy_grid_node-3]: started with pid [11078]
process[base_to_laser_broadcaster-4]: started with pid [11086]
[ INFO] [1541676994.005781399]: I1108 19:36:34.000000 11072 configuration_file_resolver.cc:41] Found '/home/zc/catkin_ws/install_isolated/share/cartographer_ros/configuration_files/zc_build_map.lua' for 'zc_build_map.lua'.
[ INFO] [1541676994.006124286]: I1108 19:36:34.000000 11072 configuration_file_resolver.cc:41] Found '/home/zc/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1541676994.006183860]: I1108 19:36:34.000000 11072 configuration_file_resolver.cc:41] Found '/home/zc/catkin_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1541676994.006258424]: I1108 19:36:34.000000 11072 configuration_file_resolver.cc:41] Found '/home/zc/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1541676994.006300170]: I1108 19:36:34.000000 11072 configuration_file_resolver.cc:41] Found '/home/zc/catkin_ws/install_isolated/share/cartographer/configuration_files/pose_graph.lua' for 'pose_graph.lua'.
[ INFO] [1541676994.006450226]: I1108 19:36:34.000000 11072 configuration_file_resolver.cc:41] Found '/home/zc/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1541676994.006491885]: I1108 19:36:34.000000 11072 configuration_file_resolver.cc:41] Found '/home/zc/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1541676994.006558495]: I1108 19:36:34.000000 11072 configuration_file_resolver.cc:41] Found '/home/zc/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1541676994.006597822]: I1108 19:36:34.000000 11072 configuration_file_resolver.cc:41] Found '/home/zc/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_2d.lua' for 'trajectory_builder_2d.lua'.
[ INFO] [1541676994.006727725]: I1108 19:36:34.000000 11072 configuration_file_resolver.cc:41] Found '/home/zc/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1541676994.006776982]: I1108 19:36:34.000000 11072 configuration_file_resolver.cc:41] Found '/home/zc/catkin_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder_3d.lua' for 'trajectory_builder_3d.lua'.
[ INFO] [1541676994.020484346]: I1108 19:36:34.000000 11072 map_builder_bridge.cc:134] Added trajectory with ID '0'.
[ INFO] [1541676994.212808286]: I1108 19:36:34.000000 11072 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '621355968000000000'.
[ WARN] [1541676994.213847158]: W1108 19:36:34.000000 11072 pose_extrapolator.cc:168] Queue too short for velocity estimation. Queue duration: 0 ms
[ INFO] [1541676994.215562453]: I1108 19:36:34.000000 11072 pose_graph_2d.cc:142] Inserted submap (0, 0).
[ WARN] [1541676994.253293244]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.293853203]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.334359385]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.374953905]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.415488143]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.456220691]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.496982983]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.557598578]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.578224334]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.618762979]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.659404326]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.703223099]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.747280864]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.791182124]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.835301719]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.879255103]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.923160590]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676994.967305147]: W1108 19:36:34.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676995.011445719]: W1108 19:36:35.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[ WARN] [1541676995.055182978]: W1108 19:36:35.000000 11072 sensor_bridge.cc:211] Ignored subdivision of a LaserScan message from sensor scan because previous subdivision time 621355968000000000 is not before current subdivision time 621355968000000000
[... cut ...]
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename zc_build_map.lua"
      output="screen">
    <remap from="scan" to="/scan" />
   
  </node>
   <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />
</launch>

cartographer build map error,
What is the reason?
help me,thanks

zchao9456, 2018-11-08:
<launch>
  <param name="/use_sim_time" value="true" />
  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename zc_build_map.lua"
      output="screen">
    <remap from="scan" to="/scan" />
   
  </node>
   <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />
 
   

</launch>

zchao9456, 2018-11-08:
Can you tell me how to do it?I just started learning.thanks

jonra1993, 2019-09-03:
Hi @zchao9456  Could you solve that problem?

jonra1993, 2019-09-03:
Ensure that scan_time and time_increment in scan topic header are in seconds. I had same problem with YdLiDAR X4 those parameters were in ns at the beginning.
[image]

closed 2020-10-17 by MichaelGrupp