cartographer-project/cartographer_ros #1101: Tuning help with Vectornav IMU and VLP32 setup

SubMishMar, 2018-11-16:
Hi all,

Earlier I had run google cartographer in an indoor turtlebot like robot and I had got good results with it while doung 2D SLAM. Currently, I am trying to run cartographer_ros on a dataset collected from a feature rich downtown area using a sensor mounted on an autonomous golfcart. I use primarily two sensors that are relevant for our discussion here. It is a velodyne VLP32 Lidar and a Vectornav IMU. We also have GPS but currently I don't use it with cartographer, although I know that I can. 

I want to do both 2D and 3D slam and at the moment both seem to be not working too well. The environment has pedestrians and moving cars. I will be glad if some tuning help can be provided both for 2D and 3D scenario. Here I provide all the information on how to reproduce my result.

My fork of cartographer: https://github.com/SubMishMar/cartographer
My fork of cartographer_ros: https://github.com/SubMishMar/cartographer_ros
My bag file: https://drive.google.com/a/tamu.edu/file/d/1BAEg3ZGhshLQmAWkOZzj69zpAqlKiBKi/view?usp=sharing

Here is how to run it:
1. For 2D run `roslaunch cartographer_ros golfcart_2d.launch` and for 3D run `roslaunch cartographer_ros golfcart.launch` .
2. In another terminal play the bag file with the `--clock` option.

Here is the gist link to the result of running `rosbag_validate` on the provided bag file: https://gist.github.com/SubMishMar/9c5babe318e0b915d72a5804e0981c7a#file-rosbagvalidate-txt

I thank your for spending time on this.

SubMishMar, 2018-11-19:
@gaschler, @ojura 
 can you please help me with tuning this? 
I am getting considerable drift in Z while as can be seen in the image.
[image]
I think because of that loop closure doesnot work and the map is so sparse.

MichaelGrupp, 2020-10-23:
Closing for inactivity. Unfortunately we can't help with tuning for this custom sensor setup.

closed 2020-10-23 by MichaelGrupp