cartographer-project/cartographer_ros #1102: Help Start working with cartographer and RealSense pointcloud

doronhi, 2018-11-18:
Hello,
I am trying out the new Intel's RealSense D435i camera which produces both depth image and IMU measurements.
I am trying to activate cartographer_ros with it. My Goal is to see smooth movement of the position of my camera on the RViz viewer.

I tried running:
`roslaunch cartographer_ros my_robot.launch bag_filename:=src/ros_realsense/bagfiles/2018-11-15-05-29-04.bag`
but nothing appears on RViz. Tthe cartographer logs is attached here:
https://gist.github.com/doronhi/89623f4293d3829493700cc96b176410#file-cartographer_rosbag_validate-log

I tried running:
`roslaunch cartographer_ros offline_my_robot.launch bag_filenames:=src/ros_realsense/bagfiles/2018-11-15-05-29-04.bag` but it seems like it could not find my bagfile. See log in:
https://gist.github.com/doronhi/89623f4293d3829493700cc96b176410#file-cartographer_ros_log2-log

I tried playing the bagfile from a remote computer using:
`rosbag play ~/bagfiles/2018-11-15-05-29-04.bag`
with cartographer running on my computer using:
`roslaunch cartographer_ros my_robot.launch`
Now I saw very little movement in the Odom frame. It seemed to me like it was updated every couple of seconds.

I run the validation app:
`cartographer_rosbag_validate -bag_filename src/ros_realsense/bagfiles/2018-11-15-05-29-04.bag`
output in:
https://gist.github.com/doronhi/89623f4293d3829493700cc96b176410#file-cartographer_rosbag_validate-log

I saw in the results what seemed to me like minor time differences (between what exactly? How to correct?) and notifications of duplicating images, no doubt because of the synchronization mechanism inside librealsense. Is it important?

I put my launch files my_robot.launch, offline_my_robot.launch and lua file, my_robot.lua in my branch at:
https://github.com/doronhi/cartographer_ros.git

and the bag file in:
https://drive.google.com/open?id=1EypZubyaceVNElOdsE4AZZVgeBdttRAu

Is any of the problems found in the validation tool the root of my problem?
Could you help me figure out whats wrong with my launch files, regarding the offline runs I failed to do?
Is the extremely slow update rate of the Odom Frame caused by my configuration or is the machine simply not strong enough?

Thanks,
Doron

doronhi, 2018-11-27:
Hello,
I fixed some bugs and got a new bag files.
Still, all I can get is the TF for camera_link to jump randomly around while the camera stand still.
Is it because of noise in the IMU?
Is the pointcloud being used?

My goal is to have the camera_link TF track the motion of the camera. Is this goal unsuited for this data and the cartographer?

I post the new bag files:
https://drive.google.com/file/d/1xyPwnyrzsrSvAAz22Qk1Fq-YNF0sKr-2/view?usp=sharing
and
https://drive.google.com/file/d/10mHMwVUPcP1-pw4y2rlMRQuZCi-_nrej/view?usp=sharing

Also, my_robot.lua and my_robot.launch can be seen on:
https://github.com/doronhi/cartographer_ros.git

Thanks again,
Doron.

MichaelGrupp, 2020-10-21:
RGB-D cameras are not well supported from what I have heard or would require a lot of experimentation with custom tuning. Closing as inactive.

closed 2020-10-21 by MichaelGrupp

ShrutheeshIR, 2022-04-15:
Hello @doronhi , I'm also looking to use a realsense D435i for this task, I was wondering if you made any progress with it? Thanks.