cartographer-project/cartographer_ros #1106: tunung in 3D slam with vlp16, imu

jo679513, 2018-11-28:
Thank you for filing an issue!

It would help us tremendously if you run through the following checklist. This
ensures that we have the most information to quickly understand, analyze,
reproduce and eventually resolve your issue.

Please

1. run `rosbag_validate` which does some checks on your sensor data. This
   tool often finds issues that can explain poor performance and must be fixed
   at recording time. Please post the full output of the tool into a
   GitHub Gist at https://gist.github.com/, then link it in the issue even if
   it does not report anything. You can run the tool like this:

   rosrun cartographer_ros cartographer_rosbag_validate -bag_filename <bag filename>

2. post a link to a Git repository containing a branch of
   `cartographer_ros` containing all the configuration, launch, and URDF files
   required to reproduce your issue.
3. post a link to a bag file we can use to reproduce your issue. Put it on
   Google Drive, Dropbox, any webserver or wherever it is publicly
   downloadable.
4. remove this boilerplate text before submitting your issue.

We will likely be unable to help you without all of these points addressed.

Entropy512, 2018-11-29:
I don't see any files in your cartographer_ros directory that seem to be unique to your platform, and since you uploaded without commit history, it's impossible to see what you changed.

What I do know from experience with the VLP-16 is that by default, the ROS node for the lidar will internally accumulate scans and then deliver either a full rotation in single return mode or half rotation in dual return mode.

There's a variable (npackets?  My memory is fuzzy, it has been a while) that can be set which will provide 1 PointCloud message per packet.  What I recall is that I had to manually change the VLP-16's launch files when starting the node because the parameter can't be passed as an option to the default launch files.  (Sorry I cannot provide more detail, the vehicle I was working with has been in storage for over siz months.)  The Cartographer configuration at https://github.com/jo679513/cartographer_ros/blob/master/cartographer_ros/configuration_files/backpack_3d.lua#L45 relies on npackets = 1 and not whatever the VLP-16 node uses as default.

jo679513, 2018-12-04:
Thanks for your help, I will try and post the result.

Duke-Allen, 2019-02-22:
@jo679513  I am tuning in 3D slam with imu and rs_16 lidar. But it no map output.
It displays "No tf data, Actual error: [map] does not exist" and the robotmodel' status is error.
my files:
demo_rslidar_3d.launch:(https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
rslidar_3d.launch:(https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
mydemo_3d.rviz:(https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
rslidar_scan_3d.lua:(https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
rslidar_2d.urdf:(https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
I don't know how to solve it. Could you help me?

jo679513, 2019-02-24:
@Duke-Allen 
If you use an IMU this [line](https://github.com/Duke-Allen/cartographer-configuration/blob/afb1ae5befa17a6c58467bdb4a4c698a666f6443/rslidar_scan_3d.lua#L21) should use the frame name of imu.

Duke-Allen, 2019-02-24:
@jo679513 
Thanks for your advance. I changed it from `base_link` to `imu`(my frame name of imu), and the cartographer runs successfully.

closed 2019-02-25 by jo679513

Duke-Allen, 2019-02-28:
@jo679513 
I am sorry for disturb you again, but I really have some questions and do not know the solution.
I tun the parameters and run it with no error, the results are as follows:
[image]
[image]

There is a big gap with the real environment. I think the map is in a mess. What is the problem of this result? How can I solve it?
This is the validate result of my bag:

- E0227 22:30:57.092067 19056 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.6084 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
I0227 22:30:57.093359 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 37968  Min: 0.004133  Max: 0.016211  Mean: 0.010000
[0.004133, 0.005340)	                    	Count: 2 (0.005268%)	Total: 2 (0.005268%)
[0.005340, 0.006548)	                    	Count: 3 (0.007901%)	Total: 5 (0.013169%)
[0.006548, 0.007756)	                    	Count: 8 (0.021070%)	Total: 13 (0.034239%)
[0.007756, 0.008964)	                    	Count: 828 (2.180784%)	Total: 841 (2.215023%)
[0.008964, 0.010172)	 ###################	Count: 35958 (94.706070%)	Total: 36799 (96.921089%)
[0.010172, 0.011380)	                    	Count: 345 (0.908660%)	Total: 37144 (97.829750%)
[0.011380, 0.012588)	                    	Count: 820 (2.159714%)	Total: 37964 (99.989464%)
[0.012588, 0.013796)	                    	Count: 1 (0.002634%)	Total: 37965 (99.992096%)
[0.013796, 0.015004)	                    	Count: 0 (0.000000%)	Total: 37965 (99.992096%)
[0.015004, 0.016211]	                    	Count: 3 (0.007901%)	Total: 37968 (100.000000%)
E0227 22:30:57.094650 19056 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101409 s, recommended is [0.0005, 0.05] s with no jitter.
I0227 22:30:57.094745 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 3766  Min: 0.100213  Max: 0.101409  Mean: 0.100801
[0.100213, 0.100333)	                    	Count: 1 (0.026553%)	Total: 1 (0.026553%)
[0.100333, 0.100452)	                    	Count: 1 (0.026553%)	Total: 2 (0.053107%)
[0.100452, 0.100572)	                    	Count: 9 (0.238980%)	Total: 11 (0.292087%)
[0.100572, 0.100691)	                    	Count: 33 (0.876261%)	Total: 44 (1.168348%)
[0.100691, 0.100811)	      ##############	Count: 2717 (72.145515%)	Total: 2761 (73.313858%)
[0.100811, 0.100931)	               #####	Count: 972 (25.809877%)	Total: 3733 (99.123741%)
[0.100931, 0.101050)	                    	Count: 25 (0.663834%)	Total: 3758 (99.787575%)
[0.101050, 0.101170)	                    	Count: 6 (0.159320%)	Total: 3764 (99.946892%)
[0.101170, 0.101289)	                    	Count: 1 (0.026553%)	Total: 3765 (99.973450%)
[0.101289, 0.101409]	                    	Count: 1 (0.026553%)	Total: 3766 (100.000000%)

[image]

Here are my files:
[demo_rsldiar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
[my data bag](https://pan.baidu.com/s/1chlbMpfUJeY-JdXoPzxzPw)
Do you know how to solve it?