cartographer-project/cartographer_ros #1107: catkin_make_isolated --use-ninja error in abseil

ytgcljj, 2018-11-28:
CMake Error at CMakeLists.txt:49 (find_package):
  By not providing "FindAbseil.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Abseil", but
  CMake did not find one.

  Could not find a package configuration file provided by "Abseil" with any
  of the following names:

    AbseilConfig.cmake
    abseil-config.cmake

  Add the installation prefix of "Abseil" to CMAKE_PREFIX_PATH or set
  "Abseil_DIR" to a directory containing one of the above files.  If "Abseil"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/catkin_ws/build_isolated/cartographer_ros/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/catkin_ws/build_isolated/cartographer_ros/CMakeFiles/CMakeError.log".
<== Failed to process package 'cartographer_ros': 
  Command '['/home/nvidia/catkin_ws/devel_isolated/cartographer/env.sh', 'cmake', '/home/nvidia/catkin_ws/src/cartographer_ros/cartographer_ros', '-DCATKIN_DEVEL_PREFIX=/home/nvidia/catkin_ws/devel_isolated/cartographer_ros', '-DCMAKE_INSTALL_PREFIX=/home/nvidia/catkin_ws/install_isolated', '-G', 'Ninja']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/nvidia/catkin_ws/build_isolated/cartographer_ros && /home/nvidia/catkin_ws/devel_isolated/cartographer/env.sh cmake /home/nvidia/catkin_ws/src/cartographer_ros/cartographer_ros -DCATKIN_DEVEL_PREFIX=/home/nvidia/catkin_ws/devel_isolated/cartographer_ros -DCMAKE_INSTALL_PREFIX=/home/nvidia/catkin_ws/install_isolated -G Ninja

Command failed, exiting.

iDsy, 2018-12-11:
Hi, I have the same issue. Have you solved it?

jaredjohansen, 2019-01-08:
abseil is part of cartographer.  

It's likely that you have a new version of cartographer_ros and an older version of cartographer.  

By getting the latest cartographer and cartographer_ros, this problem went away for me.

jinfagang, 2019-01-12:
@jaredjohansen Already built and installed newest version cartographer.................. still can not find this module...................

jaredjohansen, 2019-01-14:
when I updated cartographer_ros (and not cartographer), I got this error.

When I updated cartographer (in addition to cartographer_ros), this package was found.

https://github.com/googlecartographer/cartographer/search?utf8=%E2%9C%93&q=abseil&type=

KangolHsu, 2020-05-07:
I use cartographer(0.3.0) +cartographer_ros (0.3.0) 
It succeed.

MichaelGrupp, 2020-10-23:
We are working on fixing the build instructions - see #1531. I will close this issue in favor of the new one.

closed 2020-10-23 by MichaelGrupp