cartographer-project/cartographer_ros #1108: ceres-solver error

ytgcljj, 2018-11-28:
==> ninja -j4 -l4 in '/home/nvidia/catkin_ws/build_isolated/ceres-solver/install'
[252/335] Building CXX object internal...lock_test.dir/residual_block_test.cc.o
FAILED: /usr/bin/c++   -DCERES_CXSPARSE_VERSION=\"3.1.4\" -DCERES_GFLAGS_NAMESPACE=google -DCERES_SUITESPARSE_VERSION=\"4.4.6\" -DEIGEN_MPL2_ONLY -Iconfig -I/home/nvidia/catkin_ws/src/ceres-solver/include -I/home/nvidia/catkin_ws/src/ceres-solver/internal -I/home/nvidia/catkin_ws/src/ceres-solver/internal/ceres -isystem /usr/include/eigen3 -I/usr/include/suitesparse -fopenmp -Wno-unknown-pragmas -Wno-sign-compare -Wno-unused-parameter -Wno-missing-field-initializers -O3 -DNDEBUG -MMD -MT internal/ceres/CMakeFiles/residual_block_test.dir/residual_block_test.cc.o -MF internal/ceres/CMakeFiles/residual_block_test.dir/residual_block_test.cc.o.d -o internal/ceres/CMakeFiles/residual_block_test.dir/residual_block_test.cc.o -c /home/nvidia/catkin_ws/src/ceres-solver/internal/ceres/residual_block_test.cc
In file included from /home/nvidia/catkin_ws/src/ceres-solver/internal/ceres/residual_block_test.cc:33:0:
/home/nvidia/catkin_ws/src/ceres-solver/internal/ceres/gtest/gtest.h: In instantiation of ‘void testing::internal::DefaultPrintTo(testing::internal::IsContainer, testing::internal::false_type, const C&, std::ostream*) [with C = Eigen::Matrix<double, -1, -1, 1>; testing::internal::IsContainer = int; testing::internal::false_type = testing::internal::bool_constant<false>; std::ostream = std::basic_ostream<char>]’:
/home/nvidia/catkin_ws/src/ceres-solver/internal/ceres/gtest/gtest.h:10546:17:   required from ‘void testing::internal::PrintTo(const T&, std::ostream*) [with T = Eigen::Matrix<double, -1, -1, 1>; std::ostream = std::basic_ostream<char>]’
/home/nvidia/catkin_ws/src/ceres-solver/internal/ceres/gtest/gtest.h:10776:12:   required from ‘static void testing::internal::UniversalPrinter<T>::Print(const T&, std::ostream*) [with T = Eigen::Matrix<double, -1, -1, 1>; std::ostream = std::basic_ostream<char>]’
/home/nvidia/catkin_ws/src/ceres-solver/internal/ceres/gtest/gtest.h:10925:30:   required from ‘void testing::internal::UniversalPrint(const T&, std::ostream*) [with T = Eigen::Matrix<double, -1, -1, 1>; std::ostream = std::basic_ostream<char>]’
/home/nvidia/catkin_ws/src/ceres-solver/internal/ceres/gtest/gtest.h:10853:19:   required from ‘static void testing::internal::UniversalTersePrinter<T>::Print(const T&, std::ostream*) [with T = Eigen::Matrix<double, -1, -1, 1>; std::ostream = std::basic_ostream<char>]’
/home/nvidia/catkin_ws/src/ceres-solver/internal/ceres/gtest/gtest.h:11051:44:   required from ‘std::__cxx11::string testing::PrintToString(const T&) [with T = Eigen::Matrix<double, -1, -1, 1>; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/home/nvidia/catkin_ws/src/ceres-solver/internal/ceres/gtest/gtest.h:10345:36:   required from ‘static std::__cxx11::string testing::internal::FormatForComparison<ToPrint, OtherOperand>::Format(const ToPrint&) [with ToPrint = Eigen::Matrix<double, -1, -1, 1>; OtherOperand = Eigen::Matrix<double, -1, -1, 1>; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/home/nvidia/catkin_ws/src/ceres-solver/internal/ceres/gtest/gtest.h:10420:45:   required from ‘std::__cxx11::string testing::internal::FormatForComparisonFailureMessage(const T1&, const T2&) [with T1 = Eigen::Matrix<double, -1, -1, 1>; T2 = Eigen::Matrix<double, -1, -1, 1>; std::__cxx11::string = std::__cxx11::basic_string<char>]’
/home/nvidia/catkin_ws/src/ceres-solver/internal/ceres/gtest/gtest.h:19989:53:   required from ‘testing::AssertionResult testing::internal::CmpHelperEQFailure(const char*, const char*, const T1&, const T2&) [with T1 = Eigen::Matrix<double, -1, -1, 1>; T2 = Eigen::Matrix<double, -1, -1, 1>]’
/home/nvidia/catkin_ws/src/ceres-solver/internal/ceres/gtest/gtest.h:20006:28:   required from ‘testing::AssertionResult testing::internal::CmpHelperEQ(const char*, const char*, const T1&, const T2&) [with T1 = Eigen::Matrix<double, -1, -1, 1>; T2 = Eigen::Matrix<double, -1, -1, 1>]’
/home/nvidia/catkin_ws/src/ceres-solver/internal/ceres/gtest/gtest.h:20030:23:   required from ‘static testing::AssertionResult testing::internal::EqHelper<lhs_is_null_literal>::Compare(const char*, const char*, const T1&, const T2&) [with T1 = Eigen::Matrix<double, -1, -1, 1>; T2 = Eigen::Matrix<double, -1, -1, 1>; bool lhs_is_null_literal = false]’
/home/nvidia/catkin_ws/src/ceres-solver/internal/ceres/residual_block_test.cc:297:3:   required from here
/home/nvidia/catkin_ws/src/ceres-solver/internal/ceres/gtest/gtest.h:10445:35: error: variable or field ‘it’ declared void
   for (typename C::const_iterator it = container.begin();
                                   ^
[252/335] Building CXX object internal...x_test.dir/invert_psd_matrix_test.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'ceres-solver': 
  Command '['/home/nvidia/catkin_ws/install_isolated/env.sh', 'ninja', '-j4', '-l4']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/nvidia/catkin_ws/build_isolated/ceres-solver && /home/nvidia/catkin_ws/install_isolated/env.sh ninja -j4 -l4

Command failed, exiting.

hehern, 2019-03-26:
I have met the same problem.Have you found the solution?

cesarhcq, 2019-06-21:
Try to install the following dependencies listed below independent

First, you need to install ceres-solver in your computer, [see](http://ceres-solver.org/installation.html#linux)
Second, you need to update your Eigen to 3.3.7 -> [see](http://eigen.tuxfamily.org/index.php?title=Main_Page)

After this two steps, try to install cartographer again and it will work fine!

marcelino-pensa, 2019-07-30:
The solution above does not work for me... I did install ceres and ran the example:
```
iter      cost      cost_change  |gradient|   |step|    tr_ratio  tr_radius  ls_iter  iter_time  total_time
   0  4.185660e+06    0.00e+00    1.09e+08   0.00e+00   0.00e+00  1.00e+04        0    6.50e-02    1.63e-01
   1  1.062590e+05    4.08e+06    8.99e+06   5.36e+02   9.82e-01  3.00e+04        1    1.35e-01    2.98e-01
   2  4.992817e+04    5.63e+04    8.32e+06   3.19e+02   6.52e-01  3.09e+04        1    1.25e-01    4.23e-01
   3  1.899774e+04    3.09e+04    1.60e+06   1.24e+02   9.77e-01  9.26e+04        1    1.26e-01    5.49e-01
   4  1.808729e+04    9.10e+02    3.97e+05   6.39e+01   9.51e-01  2.78e+05        1    1.24e-01    6.73e-01
   5  1.803399e+04    5.33e+01    1.48e+04   1.23e+01   9.99e-01  8.33e+05        1    1.24e-01    7.98e-01
   6  1.803390e+04    9.02e-02    6.35e+01   8.00e-01   1.00e+00  2.50e+06        1    1.25e-01    9.23e-01

Solver Summary (v 1.14.0-eigen-(3.3.7)-lapack-suitesparse-(4.4.6)-cxsparse-(3.1.4)-eigensparse-openmp-no_tbb)

                                     Original                  Reduced
Parameter blocks                        22122                    22122
Parameters                              66462                    66462
Residual blocks                         83718                    83718
Residuals                              167436                   167436

Minimizer                        TRUST_REGION

Dense linear algebra library            EIGEN
Trust region strategy     LEVENBERG_MARQUARDT

                                        Given                     Used
Linear solver                     DENSE_SCHUR              DENSE_SCHUR
Threads                                     1                        1
Linear solver ordering              AUTOMATIC                 22106,16
Schur structure                         2,3,9                    2,3,9

Cost:
Initial                          4.185660e+06
Final                            1.803390e+04
Change                           4.167626e+06

Minimizer iterations                        7
Successful steps                            7
Unsuccessful steps                          0

Time (in seconds):
Preprocessor                         0.098426

  Residual only evaluation           0.092055 (7)
  Jacobian & residual evaluation     0.362736 (7)
  Linear solver                      0.365441 (7)
Minimizer                            0.895651

Postprocessor                        0.003135
Total                                0.997213

Termination:                      CONVERGENCE (Function tolerance reached. |cost_change|/cost: 1.769761e-09 <= 1.000000e-06)
```

Also, Eigen is 3.3.7... the verbose above confirms that, and I also get it through:
```
pkg-config --modversion eigen3
3.3.7
```

MichaelGrupp, 2020-10-27:
I highly recommend to use normal Ubuntu / ROS packages on newer distributions (Ubuntu 18 or higher). I verified that these instructions in our documentation work on a clean Ubuntu /  ROS installation: https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html

closed 2020-10-27 by MichaelGrupp