cartographer-project/cartographer_ros #1109: How can I know the Cartographer covariance of the pose?

alfred31, 2018-11-28:
Hello,

First of all thanks for the amazing job you've done.

Now that I've tuned cartographer and it is working incredibly precise, I want to get a pose topic to get the localization of the robot. I have already made a ros node that publishes a geometry_msgs::PoseStamped message, but what I really want to get is a geometry_msgs::PoseWithCovariance.

The point is that I don't know how I can defined this covariance, Is there any way to get this covariance in the cartographer, or at least a way to approximate it?

Thank you very much.

Kind Regards

johuber, 2018-11-28:
In PR #1099 I suggested a change that will make cartographer directly publish a PoseStamped.

Finding a meaningful covariance for an optimization based localization is difficult, at least if you want a mathematically correct result. For many applications it is sufficient to just tune it with trial and error to values that work for you.

alfred31, 2018-11-28:
Hi,

Thanks for your answer. I will try to find a solution to this issue and post the outcomes in case it turns out useful for other people.

Best Regards.

closed 2020-10-22 by MichaelGrupp

Vineet-Pandey, 2021-06-08:
@alfred31 - Were you able to determine a deterministic method to estimate the covariance or should it be done through trial and error?