cartographer-project/cartographer_ros #1111: assets_writer_backpack_3d.launch could not generate the PLY file

XuanboWang, 2018-11-29:
Hello, everybody. I am using cartographer for 3D modeling. First, I use b3-2016-01-19-13-50-11.bag to generate the corresponding pbstream file. Then I failed to generate the ply file using bag and pbstream. I only got some png. Here is my message output:

> wangboxuan@wangboxuan-pc:~/Desktop$ roslaunch cartographer_ros assets_writer_backpack_3d.launch bag_filenames:=/media/wangboxuan/data/bag/cartographer/b3-2016-01-19-13-50-11.bag pose_graph_filename:=/media/wangboxuan/data/bag/cartographer/b3-2016-01-19-13-50-11.bag.pbstream 
... logging to /home/wangboxuan/.ros/log/57a4dd42-f3c9-11e8-88f3-0c9d92bde351/roslaunch-wangboxuan-pc-19571.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://wangboxuan-pc:37413/
> 
> SUMMARY
========
> 
> PARAMETERS
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.14
> 
> NODES
  /
    cartographer_assets_writer (cartographer_ros/cartographer_assets_writer)
> 
> auto-starting new master
process[master]: started with pid [19581]
ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to 57a4dd42-f3c9-11e8-88f3-0c9d92bde351
process[rosout-1]: started with pid [19594]
started core service [/rosout]
process[cartographer_assets_writer-2]: started with pid [19603]
I1129 19:24:32.605816 19603 configuration_file_resolver.cc:41] Found '/home/wangboxuan/catkin_ws/src/cartographer_ros/cartographer_ros/configuration_files/assets_writer_backpack_3d.lua' for 'assets_writer_backpack_3d.lua'.
I1129 19:24:32.606235 19603 configuration_file_resolver.cc:41] Found '/home/wangboxuan/catkin_ws/src/cartographer_ros/cartographer_ros/configuration_files/transform.lua' for 'transform.lua'.
I1129 19:24:32.606402 19603 configuration_file_resolver.cc:41] Found '/home/wangboxuan/catkin_ws/src/cartographer_ros/cartographer_ros/configuration_files/transform.lua' for 'transform.lua'.
I1129 19:24:32.610152 19603 assets_writer.cc:188] Processing /media/wangboxuan/data/bag/cartographer/b3-2016-01-19-13-50-11.bag...
I1129 19:24:33.137176 19603 assets_writer.cc:254] Processed 0 of 318.777 bag time seconds...
I1129 19:24:35.284519 19603 assets_writer.cc:254] Processed 3.06347 of 318.777 bag time seconds...
I1129 19:24:38.909720 19603 assets_writer.cc:254] Processed 6.12692 of 318.777 bag time seconds...
I1129 19:24:42.677558 19603 assets_writer.cc:254] Processed 9.1907 of 318.777 bag time seconds...
I1129 19:24:46.402613 19603 assets_writer.cc:254] Processed 12.2544 of 318.777 bag time seconds...
I1129 19:24:49.824177 19603 assets_writer.cc:254] Processed 15.3179 of 318.777 bag time seconds...
I1129 19:24:53.010408 19603 assets_writer.cc:254] Processed 18.3816 of 318.777 bag time seconds...
I1129 19:24:56.212211 19603 assets_writer.cc:254] Processed 21.4453 of 318.777 bag time seconds...
I1129 19:24:59.343277 19603 assets_writer.cc:254] Processed 24.5092 of 318.777 bag time seconds...
I1129 19:25:03.100255 19603 assets_writer.cc:254] Processed 27.5725 of 318.777 bag time seconds...
I1129 19:25:07.419483 19603 assets_writer.cc:254] Processed 30.6362 of 318.777 bag time seconds...
I1129 19:25:12.594413 19603 assets_writer.cc:254] Processed 33.7001 of 318.777 bag time seconds...
I1129 19:25:18.044780 19603 assets_writer.cc:254] Processed 36.7634 of 318.777 bag time seconds...
I1129 19:25:23.364634 19603 assets_writer.cc:254] Processed 39.8273 of 318.777 bag time seconds...
I1129 19:25:28.397992 19603 assets_writer.cc:254] Processed 42.8905 of 318.777 bag time seconds...
I1129 19:25:33.655019 19603 assets_writer.cc:254] Processed 45.9543 of 318.777 bag time seconds...
I1129 19:25:38.843272 19603 assets_writer.cc:254] Processed 49.0182 of 318.777 bag time seconds...
I1129 19:25:43.980425 19603 assets_writer.cc:254] Processed 52.0815 of 318.777 bag time seconds...
I1129 19:25:49.149274 19603 assets_writer.cc:254] Processed 55.1455 of 318.777 bag time seconds...
I1129 19:25:54.588270 19603 assets_writer.cc:254] Processed 58.2092 of 318.777 bag time seconds...
I1129 19:26:00.211326 19603 assets_writer.cc:254] Processed 61.2724 of 318.777 bag time seconds...
I1129 19:26:06.027845 19603 assets_writer.cc:254] Processed 64.3364 of 318.777 bag time seconds...
I1129 19:26:11.847126 19603 assets_writer.cc:254] Processed 67.3999 of 318.777 bag time seconds...
I1129 19:26:18.040110 19603 assets_writer.cc:254] Processed 70.4638 of 318.777 bag time seconds...
I1129 19:26:24.606871 19603 assets_writer.cc:254] Processed 73.5273 of 318.777 bag time seconds...
I1129 19:26:30.942266 19603 assets_writer.cc:254] Processed 76.5908 of 318.777 bag time seconds...
I1129 19:26:36.746826 19603 assets_writer.cc:254] Processed 79.6548 of 318.777 bag time seconds...
I1129 19:26:42.986706 19603 assets_writer.cc:254] Processed 82.7181 of 318.777 bag time seconds...
I1129 19:26:49.429412 19603 assets_writer.cc:254] Processed 85.782 of 318.777 bag time seconds...
I1129 19:26:55.815291 19603 assets_writer.cc:254] Processed 88.8455 of 318.777 bag time seconds...
I1129 19:27:01.941136 19603 assets_writer.cc:254] Processed 91.909 of 318.777 bag time seconds...
I1129 19:27:07.477023 19603 assets_writer.cc:254] Processed 94.9729 of 318.777 bag time seconds...
I1129 19:27:12.807477 19603 assets_writer.cc:254] Processed 98.0364 of 318.777 bag time seconds...
I1129 19:27:17.929802 19603 assets_writer.cc:254] Processed 101.1 of 318.777 bag time seconds...
I1129 19:27:22.964658 19603 assets_writer.cc:254] Processed 104.164 of 318.777 bag time seconds...
I1129 19:27:28.063108 19603 assets_writer.cc:254] Processed 107.227 of 318.777 bag time seconds...
I1129 19:27:33.073154 19603 assets_writer.cc:254] Processed 110.291 of 318.777 bag time seconds...
[... 57 lines cut ...]
I1129 19:31:43.548763 19603 assets_writer.cc:254] Processed 287.983 of 318.777 bag time seconds...
I1129 19:31:47.411000 19603 assets_writer.cc:254] Processed 291.047 of 318.777 bag time seconds...
I1129 19:31:51.451323 19603 assets_writer.cc:254] Processed 294.111 of 318.777 bag time seconds...
I1129 19:31:55.213274 19603 assets_writer.cc:254] Processed 297.174 of 318.777 bag time seconds...
I1129 19:31:59.204304 19603 assets_writer.cc:254] Processed 300.238 of 318.777 bag time seconds...
I1129 19:32:03.219044 19603 assets_writer.cc:254] Processed 303.301 of 318.777 bag time seconds...
I1129 19:32:07.235956 19603 assets_writer.cc:254] Processed 306.365 of 318.777 bag time seconds...
I1129 19:32:11.543886 19603 assets_writer.cc:254] Processed 309.429 of 318.777 bag time seconds...
I1129 19:32:15.633553 19603 assets_writer.cc:254] Processed 312.492 of 318.777 bag time seconds...
I1129 19:32:19.654641 19603 assets_writer.cc:254] Processed 315.556 of 318.777 bag time seconds...
I1129 19:32:23.646626 19603 assets_writer.cc:254] Processed 318.619 of 318.777 bag time seconds...
I1129 19:32:33.871654 19603 counting_points_processor.cc:43] Processed 344146792 and finishing.
================================================================================REQUIRED process [cartographer_assets_writer-2] has died!
process has finished cleanly
log file: /home/wangboxuan/.ros/log/57a4dd42-f3c9-11e8-88f3-0c9d92bde351/cartographer_assets_writer-2*.log
Initiating shutdown!
================================================================================
[cartographer_assets_writer-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Please help me see what the problem is and what should I do?

XuanboWang, 2018-11-30:
I have solved this problem. The reason for the mistake is that I do not require the output of PCD or ply in the Lua configuration file. Just like https://github.com/googlecartographer/cartographer/issues/1447, I should add the following configuration in the corresponding lua:

>     {
>       action = "write_ply",
>       filename = "points.ply",
>     },
>     {
>       action = "write_pcd",
>       filename = "points.pcd",
>     },

closed 2018-11-30 by XuanboWang

codeBeefFly, 2021-08-17:
hi, I also encountered such problem but was trying to generate a map.pgm file using a demo 2d rosbag. 

```
I0817 18:24:19.258977 30025 assets_writer.cc:254] Processed 93.2024 of 360.611 bag time seconds...
I0817 18:24:29.984477 30025 assets_writer.cc:254] Processed 124.264 of 360.611 bag time seconds...
I0817 18:24:41.377332 30025 assets_writer.cc:254] Processed 155.434 of 360.611 bag time seconds...
I0817 18:25:01.028228 30025 assets_writer.cc:254] Processed 186.442 of 360.611 bag time seconds...
I0817 18:25:19.001085 30025 assets_writer.cc:254] Processed 217.277 of 360.611 bag time seconds...
I0817 18:25:32.988505 30025 assets_writer.cc:254] Processed 248.321 of 360.611 bag time seconds...
I0817 18:25:41.931903 30025 assets_writer.cc:254] Processed 279.383 of 360.611 bag time seconds...
I0817 18:25:52.892603 30025 assets_writer.cc:254] Processed 310.487 of 360.611 bag time seconds...
I0817 18:26:06.210233 30025 assets_writer.cc:254] Processed 341.633 of 360.611 bag time seconds...
================================================================================REQUIRED process [cartographer_assets_writer-2] has died!
process has finished cleanly
log file: /home/ds16v2/.ros/log/33c91c28-ff45-11eb-aff9-a0c589ac1e85/cartographer_assets_writer-2*.log
Initiating shutdown!
================================================================================
[cartographer_assets_writer-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

do you know where the issue is and how to solve it, thanks.